MarlinFirmware/Marlin #6324: Wrong Toolchange Z Lift

Haffwa, 2017-04-12:
I am trying to equip my i3 clone with  dual e3d hotends. I changed everything in Marlin and now i can control both hotends. Both hotends are set to the same height and the x/y offset is correct too. I can even print with hotend 1 or 2 independently but when I want to use both I get following error:

Printer starts with Hotend 1 and when it switches to Hotend 2, it parks beside the object and it raises the Z axis a little. After that, Hotend 2 is heated to desired temperature but when the print continues (with 2nd hotend), it stays at the wrong Z height and doesnt move down to the correct height.
Does anyone have an idea of how to fix this?

I am using a normal ramps 1.4 with Marlin 1.1.0

Bob-the-Kuhn, 2017-04-12:
Are you using 1.1.0 RC8 or 1.1.0 RCBugFix?

I expect that you're using RC8.  If that's the case then there's an excellent chance that this has been fixed in RCBugFix.

You can get RCBugFix [here](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

The downside is you'll need to transfer all your machine specific settings to the new config files.

Haffwa, 2017-04-12:
Yes i got RC8 and not RCBugFix! I will try that out, thanks a lot!

Haffwa, 2017-04-12:
You Sir deserve a medal! The new version totally fixed my issue! Thanks!!!!

closed 2017-04-12 by Haffwa