MarlinFirmware/Marlin #6326: PID Tuning with AutoFan enabled caps out at 150C

Jo0, 2017-04-13:
I've been trying to run a PIDTune with AutoFan enabled. I set the Min AutoFan to 200C and when I run the PID Tune I never see the temperatures go above 150C.

I looked at temperature.cpp and I don't see how the temperature gets capped out at 150C. Any ideas on where to go from here?

Bob-the-Kuhn, 2017-04-13:
Is it just the M303 that has issues when Min AutoFan is set to 200C or does M104 also have the same problem?

If it's just M303 then I think there was a missing parameter when the command was issued.  Here's the M303 parameters.
```
/**
 * M303: PID relay autotune
 *
 *       S<temperature> sets the target temperature. (default 150C)
 *       E<extruder> (-1 for the bed) (default 0)
 *       C<cycles>
 *       U<bool> with a non-zero value will apply the result to current settings
 */
```

Jo0, 2017-04-13:
This is just with M303

I just tried it with the correct parameter order and it still won't go past 150C. 

With Autofan off it works according to the parameters I enter.

`Send: M303 c10 e0 s205
Recv: PID Autotune start
Recv:  T:21.5 /0.0 B:21.6 /0.0 @:127 B@:0
Recv:  T:21.8 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:22.4 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:23.5 /0.0 B:21.5 /0.0 @:127 B@:0
Recv:  T:24.6 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:27.1 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:29.1 /0.0 B:21.5 /0.0 @:127 B@:0
Recv:  T:31.4 /0.0 B:21.0 /0.0 @:127 B@:0
Recv:  T:33.9 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:36.4 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:39.0 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:41.9 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:44.6 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:47.4 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:50.3 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:52.9 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:55.7 /0.0 B:21.1 /0.0 @:127 B@:0
Recv:  T:58.5 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:61.2 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:63.9 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:66.6 /0.0 B:21.1 /0.0 @:127 B@:0
Recv:  T:69.2 /0.0 B:21.0 /0.0 @:127 B@:0
Recv:  T:72.1 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:74.9 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:77.3 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:80.0 /0.0 B:20.9 /0.0 @:127 B@:0
Recv:  T:82.5 /0.0 B:21.5 /0.0 @:127 B@:0
Recv:  T:85.1 /0.0 B:21.8 /0.0 @:127 B@:0
Recv:  T:87.8 /0.0 B:22.3 /0.0 @:127 B@:0
Recv:  T:90.2 /0.0 B:21.5 /0.0 @:127 B@:0
Recv:  T:92.6 /0.0 B:21.5 /0.0 @:127 B@:0
Recv:  T:95.1 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:97.3 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:99.7 /0.0 B:21.5 /0.0 @:127 B@:0
Recv:  T:102.2 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:104.4 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:106.7 /0.0 B:21.7 /0.0 @:127 B@:0
Recv:  T:108.9 /0.0 B:21.1 /0.0 @:127 B@:0
Recv:  T:111.2 /0.0 B:21.3 /0.0 @:127 B@:0
Recv:  T:113.4 /0.0 B:21.1 /0.0 @:127 B@:0
Recv:  T:115.7 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:117.9 /0.0 B:21.6 /0.0 @:127 B@:0
Recv:  T:120.0 /0.0 B:20.9 /0.0 @:127 B@:0
Recv:  T:122.1 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:124.2 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:126.3 /0.0 B:21.6 /0.0 @:127 B@:0
Recv:  T:128.6 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:130.4 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:132.4 /0.0 B:22.0 /0.0 @:127 B@:0
Recv:  T:134.3 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:136.5 /0.0 B:22.0 /0.0 @:127 B@:0
Recv:  T:138.7 /0.0 B:20.9 /0.0 @:127 B@:0
Recv:  T:140.7 /0.0 B:21.0 /0.0 @:127 B@:0
Recv:  T:142.7 /0.0 B:20.7 /0.0 @:127 B@:0
Recv:  T:144.1 /0.0 B:21.2 /0.0 @:127 B@:0
Recv:  T:145.9 /0.0 B:20.8 /0.0 @:127 B@:0
Recv:  T:147.7 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:149.9 /0.0 B:21.0 /0.0 @:127 B@:0
Recv:  T:151.7 /0.0 B:21.1 /0.0 @:0 B@:0
Recv:  T:153.3 /0.0 B:21.5 /0.0 @:0 B@:0
Recv:  T:154.5 /0.0 B:21.2 /0.0 @:0 B@:0
Recv:  T:155.3 /0.0 B:21.6 /0.0 @:0 B@:0
Recv:  T:155.6 /0.0 B:21.3 /0.0 @:0 B@:0
Recv:  T:155.6 /0.0 B:21.8 /0.0 @:0 B@:0
Recv:  T:155.3 /0.0 B:21.8 /0.0 @:0 B@:0
Recv:  T:154.8 /0.0 B:21.5 /0.0 @:0 B@:0
Recv:  T:154.4 /0.0 B:21.6 /0.0 @:0 B@:0
Recv:  T:153.6 /0.0 B:21.4 /0.0 @:0 B@:0
Recv:  T:152.8 /0.0 B:21.4 /0.0 @:0 B@:0
Recv:  T:151.9 /0.0 B:21.9 /0.0 @:0 B@:0
Recv:  T:151.0 /0.0 B:21.8 /0.0 @:0 B@:0
Recv:  T:150.0 /0.0 B:21.5 /0.0 @:0 B@:0
Recv:  T:149.1 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:148.3 /0.0 B:21.5 /0.0 @:127 B@:0
Recv:  T:147.8 /0.0 B:21.4 /0.0 @:127 B@:0
Recv:  T:147.9 /0.0 B:22.2 /0.0 @:127 B@:0
Recv:  T:148.3 /0.0 B:22.1 /0.0 @:127 B@:0
Recv:  T:149.2 /0.0 B:21.6 /0.0 @:127 B@:0
Recv:  T:150.2 /0.0 B:21.1 /0.0 @:0 B@:0
Recv:  T:151.3 /0.0 B:21.6 /0.0 @:0 B@:0
Recv:  T:152.5 /0.0 B:21.2 /0.0 @:0 B@:0
Recv:  T:153.0 /0.0 B:21.3 /0.0 @:0 B@:0
Recv:  T:153.3 /0.0 B:21.3 /0.0 @:0 B@:0
Recv:  T:153.3 /0.0 B:21.5 /0.0 @:0 B@:0
Recv:  T:152.9 /0.0 B:21.6 /0.0 @:0 B@:0
Recv:  T:152.5 /0.0 B:21.6 /0.0 @:0 B@:0
Recv:  T:151.8 /0.0 B:22.7 /0.0 @:0 B@:0
Recv:  T:151.1 /0.0 B:21.4 /0.0 @:0 B@:0
Recv:  T:150.4 /0.0 B:21.2 /0.0 @:0 B@:0
Recv:  bias: 98 d: 98 min: 147.81 max: 153.38 T:149.5 /0.0 B:21.2 /0.0 @:98 B@:0`

kcheeeung, 2017-04-16:
The duty cycle may be limiting the max temp your hotends when doing a PID tune. I see you have BANG_MAX as 127. Try increasing that to 191 (75%) or 255 (100%).

Also if the duty cycle is too high, it'll heat too fast and you won't be able to proceed with a PID tune.

Jo0, 2017-04-16:
My BANG_MAX is 255 in configuration.h

Not sure why it's limiting itself to 127

Tailhappy, 2017-04-21:
Have you tried specifying the E value first in M303?, i.e. M303 E0 C6 S200

thinkyhead, 2017-04-26:
> Have you tried specifying the E value first in M303?

The positions of the parameters are irrelevant. They are fetched in the same order regardless of their position in the command line.

Tailhappy, 2017-04-26:
At this point Jo0 seems MIA so it may be irrelevant, but in my case I get differing results depending on the entry order of the parameters. Feel free to stand firm on your theoretical fetching assertion, but I can demonstrate it easily on my Reprap printer.

thinkyhead, 2017-04-26:
> I get differing results depending on the entry order of the parameters.

Color me skeptical. Here is the actual code:

```cpp
/**
 * M303: PID relay autotune
 *
 *       S<temperature> sets the target temperature. (default 150C)
 *       E<extruder> (-1 for the bed) (default 0)
 *       C<cycles>
 *       U<bool> with a non-zero value will apply the result to current settings
 */
inline void gcode_M303() {
  #if HAS_PID_HEATING
    int e = code_seen('E') ? code_value_int() : 0;
    int c = code_seen('C') ? code_value_int() : 5;
    bool u = code_seen('U') && code_value_bool();
    float temp = code_seen('S') ? code_value_temp_abs() : (e < 0 ? 70.0 : 150.0);
    . . .
}
```

Jo0, 2017-04-26:
So for some reason my PID tuning has been working as intended. I'm unsure what caused it to cap out at 150C, but I haven't experienced it again in awhile.

closed 2017-11-03 by thinkyhead