MarlinFirmware/Marlin #6331: UBL Issue

bgort, 2017-04-13:
Finally have time to check out UBL, and am having a problem:

Using the latest RCBugFix (02fca54), I'm seeing an error after `G29 P1`:
```
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
Send: M105
Recv: ok T:82.7 /0.0 B:41.7 /0.0 T0:82.7 /0.0 T1:24.0 /0.0 @:0 B@:0 @0:0 @1:0
Send: M500
Recv: echo:Settings Stored (451 bytes)
Recv: ok
Send: G28
Recv: echo:Active Extruder: 0
Recv: echo:busy: processing
Recv: echo:Active Extruder: 0
Recv: X:128.00 Y:134.00 Z:6.20 E:0.00 Count X:12800 Y:13400 Z:4960
Recv: ok
Send: G29 P1
Recv: ubl.eeprom_start=64
Recv: Mesh invalidated. Probing mesh.
Recv: 
Recv: Error:Heating failed, system stopped! Heater_ID: bed
Changing monitoring state from 'Operational' to 'Error: Heating failed, system stopped! Heater_ID: bed
'
Recv: Error:Heating failed, system stopped! Heater_ID: bed
Recv: Error:Printer halted. kill() called!
```
Heating the bed and extruder 1 seems to be fine -- tested this because the problem seems to be related.

I also tried 1b12fea and it seems to crash with no error or any indication there's a problem:

```
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
Send: M500
Recv: echo:Settings Stored (451 bytes)
Recv: ok
Send: G28
Recv: echo:Active Extruder: 0
Recv: echo:busy: processing
Recv: echo:Active Extruder: 0
Recv: X:128.00 Y:134.00 Z:6.20 E:0.00 Count X:12800 Y:13400 Z:4960
Recv: ok
Send: G29 P1
Recv: ubl.eeprom_start=64
Recv: Mesh invalidated. Probing mesh.
Recv: 

**** Apparently we hard crash here?  The printer does nothing, and ****
**** serial output simply continues with the 'start' below. ****

Recv: start
Recv: echo:Marlin 1.1.0-RCBugFix
Recv: 
Recv: echo: Last Updated: 2016-12-06 12:00 | Author: BG
Recv: Compiled: Apr 13 2017
Recv: echo: Free Memory: 1255  PlannerBufferBytes: 1296
Recv: echo:V34 stored settings retrieved (451 bytes)
Recv:  UBL not active!
Recv: 
Recv: Initializing Bed Leveling State to current firmware settings.
Recv: 
Recv: UBL System reset()
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z800.00 E99.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z15.00 E30.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X1100 Y1100 Z100 E10000
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P1100.00 R1100.00 T3000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
Recv: echo:Home offset (mm)
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Hotend offsets (mm)
Recv: echo:  M218 T1 X-32.65 Y-0.45
Recv: Unified Bed Leveling:
Recv: echo:  M420 S0
Recv: 
Recv: UBL is Inactive
Recv: 
Recv: Active Mesh Slot: -1
Recv: z_offset: 0.000000
Recv: EEPROM can hold 4 meshes.
Recv: 
Recv: GRID_MAX_POINTS_X  15
Recv: GRID_MAX_POINTS_Y  15
Recv: UBL_MESH_MIN_X     (0 + 20)
Recv: UBL_MESH_MIN_Y     (35 + 20)
Recv: UBL_MESH_MAX_X     (250 - (20))
Recv: UBL_MESH_MAX_Y     (300 - (20))
Recv: MESH_X_DIST        (float((250 - (20)) - ((0 + 20))) / float(15 - 1))
Recv: MESH_Y_DIST        (float((300 - (20)) - ((35 + 20))) / float(15 - 1))
Recv: 
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H204 B45 F0
Recv:   M145 S1 H230 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P47.58 I7.45 D76.00
Recv: echo:LCD Contrast:
Recv: echo:  M250 C40
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 T1 D1.75
Recv: echo:  M200 D0
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.20
Recv: Error:volume.init failed
```
My config files:
- [Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/920468/Configuration.h.txt)
- [Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/920469/Configuration_adv.h.txt)

I believe I'm configured correctly (comment UBL and uncomment bilinear or mesh and everything works fine [with 1b12fea; haven't tried the latest yet]) and plan to go debugging shortly, but just wanted to post here to see if anyone had any thoughts?  @Roxy-3D?

Roxy-3D, 2017-04-13:
Hey @bgort !!!   I've been chasing this problem for 5 or 6 days.   If you want to jump in and help it would be very much appreciated.    Please turn on your M100_FREE_MEMORY_WATCHER option.  That will let you verify this is happening.  (It will give you access to the top_of_stack() function)

In your ubl_G29.cpp file, Add these lines of code right at the start of the function:
```cpp
char* top_of_stack();
  void gcode_G29() {
  SERIAL_PROTOCOLLNPAIR("Start of G29", (unsigned int) (top_of_stack() );
    if (ubl.eeprom_start < 0) {
      SERIAL_PROTOCOLLNPGM("?You need to enable your EEPROM and initialize it");
      SERIAL_PROTOCOLLNPGM("with M502, M500, M501 in that order.\n");
      return;
    }
  SERIAL_PROTOCOLLNPAIR("After accessing UBL object: ", (unsigned int) (top_of_stack() );
```
What I'm seeing is the stack pointer is in a normal position before UBL object is referenced.  The stack pointer will be somewhere close to 0x2180.     But after accessing the UBL object, the stack pointer has been corrupted and (in my case) is somewhere around 0x1cc0.      Obviously, this  is causing all kinds of problem.   Important system values are stored there.

Can you see and/or verify if you see the stack pointer jump ?

bgort, 2017-04-13:
Hey Roxy!  Sure!

Here's what I have doing that:
```
*** snip ***
Recv: ok
Send: G29 P1
Recv: Start of G29; stack @ 8588
Recv: ubl.eeprom_start=64
Recv: After accessing UBL object; stack @ 8506
Recv: Mesh invalidated. Probing mesh.
Recv: 
Recv: d from 3rd point: -0.000000  c: -0.000000
*** (snipped a bunch of newlines here) ***
start
Recv: echo:Marlin 1.1.0-RCBugFix
Recv: 
Recv: echo: Last Updated: 2016-12-06 12:00 | Author: BG
Recv: Compiled: Apr 13 2017
Recv: echo: Free Memory: 1204  PlannerBufferBytes: 1296
Recv: echo:V34 stored settings retrieved (451 bytes)
Recv:  UBL not active!
*** snip ***
```
Stack is different.

I'll dig in a bit right after I get back from FedEx.  Need to drop off a package.

Roxy-3D, 2017-04-13:
What ever is wrong...  It is something very small and simple.   But right now, I can't figure it out!!!

bgort, 2017-04-13:
So hard to find sometimes... the little things.

Starting on it now.  Hopefully something stands out - fresh eyes and all of that.

Roxy-3D, 2017-04-13:
> So hard to find sometimes... the little things.
> Starting on it now. Hopefully something stands out - fresh eyes and all of that.

I've got some updates to M100 that I'm going to get cleaned up and put back up in the main branch.  It is those that helped me narrow the scope to the very start of the G29 routine.    I'll probably have that in place to assist you within an hour.

bgort, 2017-04-13:
Ah, great!

bgort, 2017-04-13:
Still not sure what's going on, but it has nothing to do with G29, I don't think.

After looking around for a bit, I decided to try this just to see what I could see:

``` 
  char *top_of_stack();
  extern char __bss_end;
  extern char __heap_start;
  extern void* __brkval;

  void gcode_G29() {
    int cnt=0;

    SERIAL_PROTOCOLLNPAIR("Start of G29; stack @ ", (unsigned int)top_of_stack());

    while(1) {
      SERIAL_PROTOCOLLNPAIR("G29 Loop ", ++cnt);
      SERIAL_PROTOCOLLNPAIR("Stack @ ", (unsigned int)top_of_stack());

      int free_memory;
      if ((int)__brkval == 0) free_memory = ((int)&free_memory) - ((int)&__bss_end);
      else free_memory = ((int)&free_memory) - ((int)__brkval);

      SERIAL_PROTOCOLLNPAIR("Free memory ", free_memory);

      delay(200);
    }
```

Here's the output:
```
Send: G29 P1
Recv: Start of G29; stack @ 8588
Recv: G29 Loop 1
Recv: Stack @ 8586
Recv: Free memory 1534
Recv: G29 Loop 2
Recv: Stack @ 8586
Recv: Free memory 1534
*** snip - all identical ***
Recv: G29 Loop 22
Recv: Stack @ 8586
Recv: Free memory 1534
Recv: G29 Loop 23
Recv: Stack @ 8586
Recv: Free memory 1534
*** crash after loop 23 ***
Recv: start
Recv: echo:Marlin 1.1.0-RCBugFix
Recv: 
```

I've tried this numerous times -- always crashes at the same place - loop 23.  As you can see, we're not even touching the UBL object/structure, so it's not likely anything to do with that.  I don't see why this loop should ever crash, unless an interrupt fires and it crashes elsewhere?

Will continue to look, but thought I'd check in with that, at least.

Roxy-3D, 2017-04-13:
@bgort and @thinkyhead     Please take a look at:  https://github.com/MarlinFirmware/Marlin/pull/6334

And if you add the lines of code shown 
```cpp
void check_for_free_memory_corruption(char *);

void gcode_G29() {
check_for_free_memory_corruption("G29 starting... ");
SERIAL_PROTOCOLLNPAIR("ubl.eeprom_start=", ubl.eeprom_start);
if (ubl.eeprom_start < 0) {
SERIAL_PROTOCOLLNPGM("?You need to enable your EEPROM and initialize it");
SERIAL_PROTOCOLLNPGM("with M502, M500, M501 in that order.\n");
return;
}
```
I believe you will get a lot of good information about what is happening.   You should probably do a 
- M100 I
- M100 F
- G28
- M100 F
- G29
And at that point...  The stack pointer is messed up!

bgort, 2017-04-13:
Will check that out now.  

Also, added my loop to G28 and it crashes in loop 23 there, too.  So, nothing to do with G29.  Unless I'm just barking up the wrong tree entirely.

Roxy-3D, 2017-04-13:
Well...   We are dealing with a memory corruption problem.   So there is no telling what is at root cause.  But having a way to reliably cause the problem is a big help.    I've been over and over the UBL object class definition and I don't see a problem.   I've changed all the linkage from static to 'normal'.   I've moved stuff out of the class.   Etc.   Etc.     The problem still keeps happening...

So if you are seeing it happen in G28, that is actually a good thing.  It means that is getting impacted also.   There is a chance this has nothing to do with UBL and UBL being sick is just a symptom.

bgort, 2017-04-13:
> So if you are seeing it happen in G28, that is actually a good thing. It means that is getting impacted also. There is a chance this has nothing to do with UBL and UBL being sick is just a symptom.

That's my opinion/belief, right now.  It's not UBL -- UBL puking is just a symptom.  Might be wrong, but ...

bgort, 2017-04-13:
> I've been over and over the UBL object class definition and I don't see a problem.

FWIW, I don't see a problem there, either.

Roxy-3D, 2017-04-13:
I just compared our Configuratioh.h and Configuration_adv.h files.    About the only thing we have similar is we both have BL-Touch probes.  Other than that...  most stuff is different.   LCD Panels, End Stop Interrupts, Z-BabyStepping, Filament_Change_Sensor, etc...   Our machines seem to only share the BL-Touch configuration.

I wonder if we should be look through any all BL-Touch changes over the last 6 weeks????

bgort, 2017-04-14:
> I wonder if we should be look through any all BL-Touch changes over the last 6 weeks????

Makes sense to me...

> End Stop Interrupts,

I didn't realize endstop interrupts were in RCBugFix; I didn't get all the way through the new Configuration.h, pretty much just merged the new into the old, fixed the conflicts, and compiled.  Neat!  Looking forward to using that once this issue is sorted.

Bob-the-Kuhn, 2017-04-14:
The major change because of the BLTouch was the "guaranteed detection" changes.

With endstop interrupts off then the stepper ISR is run at least every 2 mS.  I'm running 200 steps per unit so if the feedrate is above 150mm/minute then the stepper ISR is running faster than 2mS.

If the endstop interrupts are on then the stepper ISR runs at it's normal rate and one extra ISR is run when an endstop is hit.

The other change was to call the stop() function when the BLTouch wouldn't come out of an error state.  As best I can tell the stop function doesn't actually do anything until all the stepper movements have finished.  There's some added long delays when doing the retries and trying to get the error message out.

I forced a WDT timeout but Marlin doesn't report why it went through reset.

bgort, 2017-04-14:
Yeah, resetting the watchdog enables the loop I was testing with to run forever, so it seems it was just the wdt doing a reset.  Sorry for not catching that sooner and for any sidetracking.  Thanks, BtK.

Back to looking for the other problem...

Tannoo, 2017-04-14:
FYI: I don't have an issue with doing a G29 P1 on this branch:
https://github.com/Tannoo/Marlin/tree/RGB_Printer_Events

There are now 7 merges that I have not assimilated into my branch. Maybe this will help you guys.

bgort, 2017-04-14:
I have UBL working now.  Reducing the UBL grid points to 11 or fewer 'fixes' the crash/reset problem for me.  

check_for_free_memory_corruption() still reports corrupted memory, but it seems to do that regardless of whether the ubl object is accessed.  I took a look at the positions of __heap_start, __brkval, the stack pointer, available memory, etc., and everything seems okay, generally.

Will go through check_for_free_memory_corruption() in a bit.  Maybe something is off there?

Roxy-3D, 2017-04-14:
@Tannoo   I was looking for all those screen captures you posted.  But I can't find them.  Do you know where those got posted???

Roxy-3D, 2017-04-14:
> I have UBL working now. Reducing the UBL grid points to 11 or fewer 'fixes' the crash/reset problem for me.

I think I can confirm...   I started with Tannoo's branch just because he said his was working and I started crossing over my settings.   With a 10x10 grid it worked.  When I bumped it up to 15x15 things got very sick.   There should be enough memory for 15x15.    

But one thing that did change is the name 
```cpp
  #define GRID_MAX_POINTS_X 15      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y 15      // Don't use more than 15 points per axis, implementation limited.
```
That was done to use common names across the various flavors of bed leveling.  There might be some array definitions or something else that is doing the wrong thing with that change.

> check_for_free_memory_corruption() still reports corrupted memory, but it seems to do that regardless of whether the ubl object is accessed/on the stack. I took a look at the positions of __heap_start, __brkval, the stack pointer, available memory, etc., and everything seems okay, generally.

Is the free area ending up as multiple blocks with 'noise' in the middle of it?   If not, it probably is OK.

> Will go through check_for_free_memory_corruption() in a bit. Maybe something is off there?

The check routine is not super fancy.   Mostly, it is just doing two things.  It is seeing if the free area is one big chunk and nothing has 'corrupted' the middle of it.  The other thing it does is make sure the stack pointer is bigger than the end of the heap.      

What exactly is the check complaining about on your system?

Tannoo, 2017-04-15:
>I was looking for all those screen captures you posted. But I can't find them. Do you know where those got posted??? (I'm trying to create a few distinctly different branches!)

For the bed topography?

Roxy-3D, 2017-04-15:
> Or the Git desktop operations? Can't find the post, but here are the captures:

YES!!!  The Git DeskTop steps to create a branch!!!   I kind of have it doing the right thing.  But it seems like any time I try to create a new branch it kills (re-uses) the directory the previous branch was in.

Tannoo, 2017-04-15:
Yes, it does that. If I want to save that branch, I just copy that directory somewhere else before I switch the branch.

Roxy-3D, 2017-04-15:
> Yes, it does that. If I want to save that branch, I just copy that directory somewhere else before I switch the branch.

Question:  If I decide to 'commit' all my changes to the branch, do I need to copy that directory off somewhere?   The next time I switch to that branch, it will hammer everything in that directory back to what I committed, right?

Tannoo, 2017-04-15:
After committing to a branch....click the sync button to "upload" them to your Git repo.

You can then switch to another branch. When you switch back to the previous branch, your commits that were sync'ed, will be there.

Tannoo, 2017-04-15:
>do I need to copy that directory off somewhere?

Only if you want to keep a local copy.

Tannoo, 2017-04-15:
> The next time I switch to that branch, it will hammer everything in that directory back to what I committed, right?

Only if you had SYNC'ed first.

Submitting a commit only commits it to the local copy of that branch. Sync it before you switch to another branch. Or those local commits will be lost.

Tannoo, 2017-04-15:
@Roxy-3D, what was that G29 command to get the least square fit?

Roxy-3D, 2017-04-15:
> @Roxy-3D, what was that G29 command to get the least square fit?

It was G29 G 3 to do a 3 x 3 Grid probe...    But there is too much push back on using a 'Magic Letter' like G for Grid.  So now the command is 'J'.    But it isn't done yet.   I got side tracked with that data corruption problem.   If that is behind me...   In the next couple of days I'll get the least square fit of the grid to tilt your mesh.

Tannoo, 2017-04-15:
Ok.

bgort, 2017-04-15:
@Roxy-3D UBL is really incredible, Roxy.  Finally had time to finish setting it up, and I'm very impressed.

Right out of the gate (first test):
[image]
(ignore the missing circle in the corner -- I did that after it was finished -- printed perfectly)

To say that I'm thrilled is an understatement.... Great, great work.

Will take a look at the M100-related routines here shortly.

bgort, 2017-04-15:
> @bgort Hey!!! With the latest RCBugFix and the attached Configuration.h files... I'm running just fine with a 15x15 grid.

Awesome!  I'll try it in a few minutes!

Roxy-3D, 2017-04-15:
>> @bgort Hey!!! With the latest RCBugFix and the attached Configuration.h files... I'm running just fine with a 15x15 grid.
> Awesome! I'll try it in a few minutes!

Actually...  That was with yesterday's files...  Today, the data corruption is back.  I would appreciate any help you can offer to find the problem!

But Thank You for the comments!   :)

bgort, 2017-04-15:
>Actually... That was with yesterday's files... Today, the data corruption is back. I would appreciate any help you can offer to find the problem!

I was just looking through your configs and the recent changes to see what might be responsible, but wasn't coming up with anything.  FWIW, I do think the idea to remove the z_values from the ubl object is a good one.

Rhonal89, 2017-04-15:
I found this thread looking on google, As Im also having the same problem as the OP. It will do G28 and G29 will be skipped. Doing so in Pronterface will read out this error 
ubl.eeprom_start=-1
?You need to enable your EEPROM and initialize it
with M502, M500, M501 in that order.

EEPROM is enable.  Also my latest RC8 fix I download it yesterday late at 11pm.  I also running a BlTouch.

Roxy-3D, 2017-04-15:
That problem is because UBL is detecting the state structure is incoherent.  If you do the M502, M500, M501 and reset...   It should go away.

bgort, 2017-04-15:
The latest is actually working w/a 15x15 here.  Populating the z_values matrix now.  I haven't yet applied the ultra_lcd changes you suggested above.

If you haven't already started, I can move z_values out of the ubl object?  Looks pretty straight forward.

Roxy-3D, 2017-04-15:
@bgort 

> UBL is really incredible, Roxy. Finally had time to finish setting it up, and I'm very impressed.
> Right out of the gate (first test):
> (ignore the missing circle in the corner -- I did that after it [...]

You needed to use the Interactive Mesh Editor to get the validation pattern to print that nice, right?
Can you be talked into posting your mesh?  I'm curious what types of imperfections people are fighting.   I went to the Dollar store to get the absolute worst piece of glass I could find just so the system would be stressed.   

> Will take a look at the M100-related routines here shortly.

I'm gearing up to take another run at the data corruption.   Like I say...  Any help would be hugely appreciated!

bgort, 2017-04-15:
>You needed to use the Interactive Mesh Editor to get the validation pattern to print that nice, right?

No, this is directly after G29 P1, then G29 P2 B to fill in gaps (only had 11 it couldn't get).

> Can you be talked into posting your mesh? I'm curious what types of imperfections people are fighting. I went to the Dollar store to get the absolute worst piece of glass I could find just so the system would be stressed.

I'm pretty flat, so there wasn't much stress.

Here's my mesh:

```
Recv: Bed Topography Report:
Recv: 
Recv: (0,10)                                                                                                                (10,10)
Recv: (0,0)                                                                                                            (237,308)
Recv:  -0.091   -0.087   -0.113   -0.086   -0.153   -0.174   -0.156   -0.175   -0.164   -0.199   -0.210
Recv: 
Recv:  -0.069   -0.056   -0.061   -0.093   -0.113   -0.084   -0.095   -0.140   -0.143   -0.163   -0.160
Recv: 
Recv:  -0.035   -0.045    0.011   -0.054   -0.039   -0.049   -0.075   -0.069   -0.099   -0.099   -0.089
Recv: 
Recv:   0.006   -0.003   -0.011    0.011   -0.003   -0.036   -0.038   -0.040   -0.065   -0.060   -0.053
Recv: 
Recv:   0.022    0.021    0.019    0.012    0.022   -0.009   -0.004   -0.005   -0.006   -0.030   -0.013
Recv: 
Recv:   0.017    0.044    0.011    0.024    0.000    0.001    0.011    0.002   -0.018   -0.028   -0.024
Recv: 
Recv:   0.045    0.015    0.029   -0.006    0.039    0.035    0.024    0.026    0.035   -0.008   -0.009
Recv: 
Recv:   0.026    0.055    0.019   -0.015    0.022    0.004    0.029    0.011    0.004    0.009    0.017
Recv: 
Recv:  -0.021   -0.035   -0.029   -0.031   -0.041   -0.036   -0.053   -0.024   -0.029   -0.024    0.001
Recv: 
Recv:  -0.034   -0.014   -0.059   -0.059   -0.058   -0.048   -0.060   -0.044   -0.051   -0.044   -0.023
Recv: 
Recv: [-0.090]  -0.050   -0.040   -0.050   -0.060   -0.020   -0.050   -0.050   -0.040   -0.020    0.010
Recv: (0,0)                                                                                                                (237,0)
Recv: (0,0)                                                                                                                   (10,0)
Recv: ok
```

Roxy-3D, 2017-04-15:
Wow!   That is very good!

bgort, 2017-04-15:
I'm pretty happy with it.  It's flattened K100S (cast/tooling aluminum) with a sheet of PEI.  Not quite the flattening I paid for (supposed to be +/- 0.001" - 25.4um), but so much better than what I had before.

bgort, 2017-04-15:
15x15 seems to be working fine here now:

```
Recv: Bed Topography Report:
Recv: 
Recv: (0,14)                                                                                                                                                                (14,14)
Recv: (0,0)                                                                                                                                                            (237,308)
Recv: [-0.021]  -0.023   -0.020   -0.024   -0.033   -0.104   -0.111   -0.086   -0.105   -0.141   -0.153   -0.163   -0.179   -0.170   -0.166
Recv: 
Recv:  -0.024    0.002   -0.026   -0.039   -0.019   -0.048   -0.076   -0.075   -0.093   -0.095   -0.121   -0.139   -0.129   -0.115   -0.124
Recv: 
Recv:   0.034    0.015    0.017    0.007   -0.034    0.025   -0.045   -0.074   -0.036   -0.066   -0.077   -0.031   -0.075   -0.064   -0.081
Recv: 
Recv:   0.031    0.048    0.040    0.045    0.046   -0.004   -0.024   -0.035    0.026   -0.061   -0.071   -0.055   -0.046   -0.048   -0.020
Recv: 
Recv:   0.101    0.089    0.060    0.029    0.071    0.043    0.024    0.014    0.019    0.017    0.005   -0.003   -0.013   -0.043   -0.001
Recv: 
Recv:   0.092    0.092    0.059    0.095    0.044    0.072    0.039    0.030    0.085    0.076    0.077    0.021    0.027    0.009    0.005
Recv: 
Recv:   0.086    0.056    0.086    0.040    0.076    0.104    0.065   -0.001    0.056    0.054   -0.008    0.072    0.065    0.020    0.027
Recv: 
Recv:   0.087    0.061    0.074    0.087    0.051    0.086    0.063    0.064    0.063    0.029    0.037    0.020   -0.004    0.071    0.036
Recv: 
Recv:   0.045    0.069    0.032    0.079    0.059    0.029    0.079    0.094    0.061    0.082    0.045    0.058    0.039    0.087    0.060
Recv: 
Recv:   0.061    0.089    0.035    0.101    0.069    0.053    0.079    0.096    0.087    0.102    0.060    0.060    0.075    0.075    0.094
Recv: 
Recv:   0.044    0.067    0.061    0.095    0.034    0.075    0.037    0.051    0.040    0.048    0.049    0.051    0.051    0.074    0.080
Recv: 
Recv:   0.009    0.025    0.027   -0.008    0.044    0.012    0.050    0.017   -0.006    0.014    0.021    0.021    0.020    0.049    0.056
Recv: 
Recv:   0.020    0.004    0.019    0.009   -0.021    0.058    0.000   -0.040    0.019    0.050    0.061    0.009    0.020    0.019    0.026
Recv: 
Recv:  -0.040   -0.010   -0.040   -0.020   -0.020   -0.030   -0.010    0.000   -0.010   -0.010   -0.020    0.020   -0.010   -0.010    0.030
Recv: 
Recv:  -0.070   -0.070   -0.020   -0.040   -0.010   -0.050   -0.010   -0.020   -0.020   -0.020    0.010   -0.010    0.000    0.020    0.020
Recv: (0,0)                                                                                                                                                                (237,0)
Recv: (0,0)                                                                                                                                                                   (14,0)      
```

[image]

Only a few little bad spots that I need to sort out how to fix. (I ruined the corner removing the prime blob, again.)

Roxy-3D, 2017-04-15:
> Only a few little bad spots that I need to sort out how to fix. 

Just move the nozzle over the area that needs adjustment, and give it a G29 P4 R     And when you have edited all the problem mesh points...  Press and hold the Encoder Wheel and move the nozzle to the next area.    

Don't forget to save the updated mesh!  And you can save it in a new location to keep the old one as a backup.

Rhonal89, 2017-04-16:
Hey Roxy, Thanks for the help earlier. Now I'm wondering how you activate the mesh after G28 so it can be use.

Roxy-3D, 2017-04-16:
Assuming you have the mesh loaded you want to use, you just do a G29 A
And you can combine commands.   You can load and activate a mesh with G29 L1 A
And if you have a mesh you like to use, you can activate it, and then save it with G29 S 1
The system will automatically load and restore the state of the last saved mesh.

bgort, 2017-04-16:
Still not finding anything wrt the memory issue. 

Did find a small issue with ubl_state padding, however:  https://github.com/MarlinFirmware/Marlin/pull/6361

Rhonal89, 2017-04-16:
I have encounter a problem. I have it set to 15 points and do get the low memory available, stability problem may occur. So something happen when doing the mesh leveling.  I have hotend and bed cooled, and I notice I was getting error:printer halted,  kill () called! . So switch back to 10 points and I was able to complete the mesh probing. So I try 15 points again and was getting the same error. But I did not know what was causing it, so this time I tried it again and pay attention.  So I was looking at the LCD and the hotend got set automatically to heat up to 500c and bed to 80c . I shut it down and soon as I notice the temp was around 225 it was hot. I had not told the printer to heat, it was just doing mesh leveling cold.

Roxy-3D, 2017-04-16:
Yeah...  I've seen that too.  The temperature object got its data corrupted.   It was just trying to do what it remembered it was supposed to do.

Rhonal89, 2017-04-16:
So is not my printer acting up. I have checked wiring and everything looks great. Only time I seen the error is switching to 15 points from 10. Don't know if is because 15 point probing takes longer than 10 points probing that it starts heating up and kills the heating to prevent over temp.

Rhonal89, 2017-04-16:
So I found lots of thread about this problem and to correct my values, I had no be temp set and Hotend started to heat to 500c and bed was actually -80c automatically in the middle of 15 point mesh leveling. I have not notice this with 10 point with the time I ran it today. Every time I run 15 point mesh it kills the action.

Roxy-3D, 2017-04-16:
> Only time I seen the error is switching to 15 points from 10. Don't know if is because 15 point probing takes longer than 10 points probing that it starts heating up and kills the heating to prevent over temp.

No.   15x15 is a much bigger memory footprint than 10x10.   It is moving some other critical data (like what the hotends and bed temperatures are supposed to be) into the target area for what ever is causing this data corruption.   

My bet is if you increase the Bi-Linear mesh size to 15x15 and add in some of the BILINEAR_SUBDIVISIONS,  You are going to see other random things happen.

Tannoo, 2017-04-16:
Is the 15x15 ubl stepping on the memory locations of the tempuratures?

Tannoo, 2017-04-16:
Have you tried a 14x14?

Rhonal89, 2017-04-16:
Will try in the AM. Hoping we can figure this out. Love the work everyone does. Especially Roxy that has been helping  me since 2014.

Roxy-3D, 2017-04-16:
> Is the 15x15 ubl stepping on the memory locations of the tempuratures?

I don't think so.   But 15 x 15 is big enough it is pushing critical pieces of data (like hotend target temperatures) in the blast zone.   I think it is random what gets hit with corruption.   I've seen the serial communication buffers get hit with 'noise'.

bgort, 2017-04-18:
Random thought here:  If we stored the z_values as microns of deviation from zero as signed 16-bit integers we'd save roughly half the memory vs. floats.  I'm not sure yet how the mesh/z correction is calculated or applied, but suspect there might be a few cycles worth of extra math if we did this?

Anyway, not sure if this idea is practical or not (or if it's already been considered), but would probably solve the memory issue by simply reducing usage.

Roxy-3D, 2017-04-18:
> Anyway, not sure if this idea is practical or not (or if it's already been considered), but would probably solve the memory issue by simply reducing usage.

Yes.  I agree.   But finding the reason memory is getting corrupted is the 'Best Answer' from a Marlin Stability perspective.   I am getting more and more convinced UBL isn't doing this.  But it enables what ever is causing the corruption.   I guess we won't know that is true until we find the root cause.

But back to your idea.    I'm not sure we even need 2 byte integers.   A signed char would give us +/- 1.28 mm of displacement from a 'Perfect 0.000 mm' mesh.   And having .01 mm of granularity (or accuracy) on each mesh point is probably fine for today's printers.

The extra math is not that bad.   Depending on how the Z-Correction factor stuff is coded, it may be we can even do the Bi-Linear interpolation using integer math.   But at worst case, using brute force, we would cast the four corners of the mesh cell to (float), do the math, and then multiply by .01 at the end (in the signed char case) to get rid of the extra two digits.

bgort, 2017-04-18:
> But finding the reason memory is getting corrupted is the 'Best Answer' from a Marlin Stability perspective. I am getting more and more convinced UBL isn't doing this. But it enables what ever is causing the corruption. I guess we won't know that is true until we find the root cause.

Oh, I agree.  Best to find and fix whatever the real issue is, if at all possible.  Mostly just thinking out loud about ways to save memory and work around the problem, if it ends up being necessary.

I'm still not coming up with anything.  I've looked at the stack, heap, and all of the other relevant addresses; inside G29 there -shouldn't- be an issue.  There's plenty of space between the heap and stack for the ubl object.

FWIW, I still don't think there's anything wrong with UBL; seems this is just a memory-related issue.  Doesn't seem there's exhaustion or a stack-heap collision, but I don't know what the other possibilities would be.  The compiler is happy with everything, visual inspection yields nothing, etc.  That leaves a runtime issue, and the only runtime issue that makes any sense at all is 'inappropriate' allocation of memory.

What's really weird is that it seems to occur semi-randomly.  I know there's a debugger for Arduino out there, though I've never tried it; have you?  Not sure if it'd work with Marlin or even could help with a problem like this.

> But back to your idea. I'm not sure we even need 2 byte integers. A signed char would give us +/- 1.28 mm of displacement from a 'Perfect 0.000 mm' mesh. And having .01 mm of granularity (or accuracy) on each mesh point is probably fine for today's printers.

I agree.  I was thinking micron precision would be ideal and so the 16-bit int was best, but you're right that it really isn't necessary; 8 bits would be enough for most use cases.

Roxy-3D, 2017-04-18:
> I'm still not coming up with anything. I've looked at the stack, heap, and all of the other relevant addresses; inside G29 there -shouldn't- be an issue. There's plenty of space between the heap and stack for the ubl object.

@thinkyhead showed me how to get both the symbol table and a disassembly of all the code.   Did you know about this:  https://github.com/MarlinFirmware/Marlin/pull/6354#issuecomment-294700312

With that...  We can get the exact location of everything in the firmware image.  And see the disassembled code.   So now, if I can get another bogus stack frame to show up in the middle of the free memory area or in the communication buffers...   It might be possible to unwind the stack and see what was going on...

> FWIW, I still don't think there's anything wrong with UBL; seems this is just a memory-related issue. Doesn't seem there's exhaustion or a stack-heap collision, but I don't know what the other [...]

Actually...   Looking at the link map and symbol table...  I'm wondering if it is something else.   At the front of the link map you will see:
```cpp
SYMBOL TABLE:
00800200 l    d  .data	00000000 .data
00000000 l    d  .text	00000000 .text
0080036c l    d  .bss	00000000 .bss
00000000 l    d  .comment	00000000 .comment
00000000 l    d  .note.gnu.avr.deviceinfo	00000000 .note.gnu.avr.deviceinfo
00000000 l    d  .debug_aranges	00000000 .debug_aranges
00000000 l    d  .debug_info	00000000 .debug_info
00000000 l    d  .debug_abbrev	00000000 .debug_abbrev
00000000 l    d  .debug_line	00000000 .debug_line
00000000 l    d  .debug_frame	00000000 .debug_frame
00000000 l    d  .debug_str	00000000 .debug_str
00000000 l    d  .debug_loc	00000000 .debug_loc
00000000 l    d  .debug_ranges	00000000 .debug_ranges
00000000 l    df *ABS*	00000000 
0000003e l       *ABS*	00000000 __SP_H__
0000003d l       *ABS*	00000000 __SP_L__
0000003f l       *ABS*	00000000 __SREG__
0000003b l       *ABS*	00000000 __RAMPZ__
00000000 l       *ABS*	00000000 __tmp_reg__
00000001 l       *ABS*	00000000 __zero_reg__
00800204 l     O .data	00000001 soft_endstops_enabled
000001f8 l     O .text	0000001b _ZZ27prepare_move_to_destinationvE3__c
000001d9 l     O .text	0000001f _ZZ27prepare_move_to_destinationvE3__c_0
```
I know for a fact, the Stack Pointer is getting changed.   How can that happen?  Well, look at address 0x3e and 0x3d of the link map.    Is the stack pointer not really an internal register on the AVR?  Is it stored in memory?   If so...   A bad pointer might be writing over stuff near NULL (0x00000) and hammering the Stack Pointer.   And that would explain everything else we have been seeing.

> What's really weird is that it seems to occur semi-randomly. I know there's a debugger for Arduino out there, though I've never tried it; have you? Not sure if it'd work with Marlin or even could help with a problem like this.

I would love to have an In Circuit Emulator.   I know Atmel sells one.   And you can buy a little USB gizmo on eBay to use with it.   http://www.ebay.com/itm/AVR-JTAG-AVR-USB-Emulator-Simulator-Debugger-Programmer-JTAG-ICE-for-Atmel/272191095130?_trksid=p2047675.c100005.m1851&_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D2%26asc%3D41811%26meid%3D0547ee6a520c4943ae7a0e40a0604534%26pid%3D100005%26rk%3D3%26rkt%3D6%26sd%3D371919571425   I'm not sure what has to be done to use one on an Atmega board, but it sure would be nice to have a hardware debugger running!

bgort, 2017-04-18:
> @thinkyhead showed me how to get both the symbol table and a disassembly of all the code. Did you know about this: #6354 (comment)

No, I hadn't seen that and didn't know about it.  Will review now.  Looks really cool.  Thanks!

> I would love to have an In Circuit Emulator. I know Atmel sells one. And you can buy a little USB gizmo on eBay to use with it. [...]

>It shows a CD also for $8!  Here is one for $3.00 but no CD: http://www.ebay.com/itm/AVR-JTAG-AVR-USB-Emulator-Simulator-Debugger-Programmer-JTAG-ICE-for-Atmel-/371919571425?hash=item56982249e1:g:gFEAAOSwmgJY7~UI  

>I wonder if these have everything to do In Circuit Debugging? I'm not sure what has to be done to use one on an Atmega board, but it sure would be nice to have a hardware debugger running!

I've just spent the last 20 minutes looking at the various ICE/debuggers and may very well pick up the JTAG ICE -- I need to be able to debug/program Cortex MCUs for an upcoming project, anyway, and it seems to do almost everything.

[104 more replies]

closed 2017-04-26 by thinkyhead