MarlinFirmware/Marlin #6336: steps_to_mm was not declared in this scope - Recent babystepping changes

emartinez167, 2017-04-14:
When trying to compile the tatest RCBugFix, I get two errors:

Arduino: 1.6.13 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch/ultralcd.cpp: In function 'void lcd_babystep_zoffset()':
ultralcd.cpp:876: error: 'steps_to_mm' was not declared in this scope
           const float new_zoffset = zprobe_zoffset + steps_to_mm[Z_AXIS] * babystep_increment;
                                                      ^
ultralcd.cpp:879: error: 'class Planner' has no member named 'abl_enabled'
             if (planner.abl_enabled)
                         ^
exit status 1
'steps_to_mm' was not declared in this scope

-----
I believe it should have been:
 const float new_zoffset = zprobe_zoffset + planner.steps_to_mm[Z_AXIS] * babystep_increment

Not sure about planner.abl_enabled.

teemuatlut, 2017-04-14:
Looks like there's also a bunch of other compile errors if you just enable
```
FIX_MOUNTED_PROBE
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
BABYSTEPPING
```
Poor Travis never stood a chance. [All the changes](https://github.com/MarlinFirmware/Marlin/commit/361655828f627896ee2fee42f996294da88574d6#diff-34f7cdd618621ae7d8c113cf0b86a2f4) are behind ENABLED(BABYSTEPPING) and HAS_BED_PROBE and he never has the time to look into both at the same time.

benlye, 2017-04-14:
Created PR #6341.  There are other issues as well - I'll work through as many of them as I can.

emartinez167, 2017-04-15:
I'll close this one as at least the steps_to_mm issue has been resolved. Still need to fix planner.abl_enabled issue, but I guess @benlye has it covered?

closed 2017-04-15 by emartinez167