MarlinFirmware/Marlin #6338: Filament change feature, heatup endless loop RCbugfix

gsart84, 2017-04-14:
Hello,
M600 Does not work on my Dual_X-Printer in the actual RCbugfix

i do M600, filament came out, put new in, push button and then it heats up the nozzle. end never ends.
any ideas?

Roxy-3D, 2017-04-14:
Can you tell us what the temperature's being displayed on the LCD Panel are?   Do the number look reasonable?     Once the nozzle is at temperature, you need to click the encoder wheel to say prime (and some times more than once) and then tell it to continue...

Is that what you are doing?

gsart84, 2017-04-14:
The temperature is going up to 220/220 and then nothing is happen. If i click the encoder, nothings happen, same if i chlick 3 times or wait 5 min

gsart84, 2017-04-14:
it stays on this screen:
nozzle is heating up.
Please wait.....

Nozzle E1 220/220

if i hit the encoder, nothin is happen

gsart84, 2017-04-18:
@Roxy-3D @thinkyhead 

i found the problem in RcBugFix
Filament_change_feature does only work if both toolheats are on the same temperature (220/220)
if i heat only one extruder, M600 Does not work. if i heat both T0 and T1 M600 work on T0.

in RC8 there was no problem.

closed 2017-11-01 by thinkyhead