MarlinFirmware/Marlin #6339: [REQUEST] Guidance for adding a new printer geometry

pwbecker, 2017-04-14:
Hi boffins, I wonder if someone can point me in the right direction in the code?

I have made a 2D printer mechanism that uses polar coordinates, and would like to modify this software to accept X,Y coordinates, and convert them into angle/radius polar coords before the instructions are sent to the steppers.

I have written Arduino code to drive this printer, and a python script to read a file and feed it to an Uno with a custom shield, but it is limited in that for e.g. it doesn't do look ahead, nor does it calculate accelerations etc.  It would be a lot of work to add all that, never mind all the other cool features that Marlin/Repetier Host have, like LCD interfacing, 3D view, pause, etc.

I was hoping to find something in the Marlin code like convert_geometry() (which would be used for delta printers for e.g.) where I could add a config option to neatly convert the gcode X,Y numbers into polar coords, and then allow the existing code to calculate accelerations, implement look-ahead etc.  I have had a look through the move planning code, but have not seen anything obvious.

It would be really great if you help me by making a suggestion as to the best way to implement this!

thinkyhead, 2017-11-01:
Sorry no one was around to help you in April. Have you since figured this out?

pwbecker, 2017-11-01:
Hello!  No I haven't.  Best way forward I can see is to pre-process the GCode to convert the geometry.  Several downsides to that, like maybe confusing the segmentation code, and also a very strange preview in Repetier Host.

It strikes me a layer of abstraction is needed to make a 'plugin' option for new geometry - i.e. one function which is called before anything else, that could be modded for any new geometries that arrive.

vanbins, 2018-03-01:
Hi! I currently got the same issue, got a polar printer but Marlin doesn't support it. I sort of know how to convert normal coordinates to polar ones, but not the extrusion speed and feedrate. Would be really nice if that was done by the firmware.

thinkyhead, 2018-03-01:
Conversion of feedrate is easier to manage now than ever before because of changes to the planner code. You now have the option to pass the mm linear distance of segments to `buffer_segment` and other planner buffering methods, and this length will be used in figuring the proportion of the feedrate to the block's Bresenham operation. I was able to remove code that did pre-scaling of SCARA distances (from mm/s to degrees/s) thanks to this change.

boelle, 2019-02-18:
@pwbecker since this sounds more like a question:

Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/ 
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

boelle, 2019-03-12:
@thinkyhead i think we can close this one, no activity from OP

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sjasonsmith, 2020-11-11:
This was a question from a particular user and doesn't seem to actually be a feature request.

closed 2020-11-11 by sjasonsmith