MarlinFirmware/Marlin #6340: [QUESTION] Where does the 32-bit version stand?

mperdue, 2017-04-14:
I'm wondering about the 32-bit version because I have a controller that has a defect in the SD card handling and would rather not buy an 8-bit board when my plan is to go 32-bit anyway. Is the current code working at all? What controller(s) is(are) supported? I prefer to stay with Marlin because of the active group here.

teemuatlut, 2017-04-14:
The Teensy HAL should have working basics but is largely untested and lacks hardware support to make things convenient. I'm not familiar with the other platforms.

JustAnother1, 2017-04-14:
There have been some forks for different controllers. But none of them got merged and they are all based on rather old Marlin versions.

Just recently @bobc started the "official" move towards 32bit. You can see it when switching to the "32-Bit-RcBugFix*" branch. The Idea is to support all ARM Cortex-M controllers.

As far as I know the branch is in an early stage. If you need to get printing fast then you need to find something else. Right now more testers and especially more implementers are needed to move this forward. Do you know someone who can code using the Arduino IDE ?

mperdue, 2017-04-15:
It looks as if the 32-Bit-RcBugFix-new version is the most current. I guess I'll need to download that and start taking a look at the code involved. I haven't done any coding since I retired three years ago but I make be able to revive some brain cells.

JustAnother1, 2017-04-15:
I would be happy to help, once you can get it into a state where the makefile works. (I'm incompatible to the Arduino IDE). I have done some stuff for STM32,..

dp90pusika, 2017-04-17:
Hi, I have a dual x carriage machine with an arduino due in it, and want to mowe to the newest 32bit version.
I tried to test the newest "32-Bit-RcBugFix-new" that is currently up, but it's busted. Does not compile because of some problem with the serial.cpp

I would be happy to help out with testing, there are a lot of new exciting features in the script. But I'm not much use as a programer. Let me know If I can help somehow.

Her is the compiling error I get:

```
In file included from sketch\serial.cpp:23:0:

serial.h:41: error: expected initializer before 'PROGMEM'

 extern const char echomagic[] PROGMEM;

                               ^

serial.h:42: error: expected initializer before 'PROGMEM'

 extern const char errormagic[] PROGMEM;

                                ^

serial.h:79: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, uint8_t v) { serial_echopair_P(s_P, (int)v); }

 ^

serial.h:80: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, uint16_t v) { serial_echopair_P(s_P, (int)v); }

 ^

serial.h:81: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, bool v) { serial_echopair_P(s_P, (int)v); }

 ^

serial.h:82: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, void *v) { serial_echopair_P(s_P, (unsigned long)v); }

 ^

serial.h:87: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serialprintPGM(const char* str) {

 ^

serial.cpp:25: error: expected initializer before 'PROGMEM'

 const char errormagic[] PROGMEM = "Error:";

                         ^

serial.cpp:26: error: expected initializer before 'PROGMEM'

 const char echomagic[] PROGMEM = "echo:";

                        ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, const char*)':

serial.cpp:28: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, const char *v)   { serialprintPGM(s_P); SERIAL_ECHO(v); }

[... 162 lines cut ...]
                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:48: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

sketch\serial.h:57:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

sketch\serial.cpp:34:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, unsigned long v) { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

exit status 1
expected initializer before 'PROGMEM'
```

thinkyhead, 2017-04-17:
Compile error fixed. I hope to rebase the branch and get it into shape after the 1.1.0 release. The hard deadline for the release is April 30.

Roxy-3D, 2017-04-17:
> The hard deadline for the release is April 30.

Is that for sure?   It would help people to know that!

JustAnother1, 2017-04-17:
> The hard deadline for the release is April 30.

 “I love deadlines. I love the whooshing noise they make as they go by.”

― Douglas Adams, The Salmon of Doubt

closed 2017-05-26 by mperdue