MarlinFirmware/Marlin #6343: Compiler error with BABYSTEPPING enabled and BABYSTEP_ZPROBE_OFFSET disabled

benlye, 2017-04-14:
Using the latest RCBugFix, if I try to compile with these options:
```
#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define BABYSTEP_XY              // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false  // Change if Z babysteps should go the other way
  #define BABYSTEP_MULTIPLICATOR 1 // Babysteps are very small. Increase for faster motion.
  //#define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping
  #define DOUBLECLICK_FOR_Z_BABYSTEPPING // Double-click on the Status Screen for Z Babystepping.
  #define DOUBLECLICK_MAX_INTERVAL 1250 // Maximum interval between clicks, in milliseconds.
                                        // Note: Extra time may be added to mitigate controller latency.
#endif
```
I get this error (Arduino IDE 1.8.1, Windows 10):
```

sketch\ultralcd.cpp: In function 'void lcd_control_motion_menu()':

ultralcd.cpp:319: error: invalid conversion from 'void (*)(bool)' to 'screenFunc_t {aka void (*)()}' [-fpermissive]

         menu_action_ ## TYPE(__VA_ARGS__); \

                                         ^

sketch\ultralcd.cpp:328:7: note: in expansion of macro '_MENU_ITEM_PART_2'

       _MENU_ITEM_PART_2(TYPE, ## __VA_ARGS__); \

       ^

sketch\ultralcd.cpp:373:53: note: in expansion of macro 'MENU_ITEM'

   #define MENU_ITEM_EDIT_CALLBACK(type, label, ...) MENU_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)

                                                     ^

sketch\ultralcd.cpp:2413:7: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'

       MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, refresh_zprobe_zoffset);

       ^

sketch\ultralcd.cpp:210:8: note: initializing argument 5 of 'void menu_action_setting_edit_callback_float32(const char*, float*, float, float, screenFunc_t)'

   void menu_action_setting_edit_callback_float32(const char* pstr, float* ptr, float minValue, float maxValue, screenFunc_t callbackFunc);

        ^

exit status 1
invalid conversion from 'void (*)(bool)' to 'screenFunc_t {aka void (*)()}' [-fpermissive]
```

I've tried to figure out what to change to resolve this one but it's beyond me at this point.

benlye, 2017-04-14:
If I comment out [ultralcd.cpp L2415](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/ultralcd.cpp#L2415):
```
MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, refresh_zprobe_zoffset);
```
I get a new set of errors:
```
sketch\ultralcd.cpp.o: In function `lcd_goto_screen(void (*)(), unsigned long)':

sketch/ultralcd.cpp:434: undefined reference to `lcd_babystep_z()'

sketch/ultralcd.cpp:434: undefined reference to `lcd_babystep_z()'

sketch\ultralcd.cpp.o: In function `lcd_tune_menu()':

sketch/ultralcd.cpp:1090: undefined reference to `lcd_babystep_z()'

sketch/ultralcd.cpp:1090: undefined reference to `lcd_babystep_z()'

collect2.exe: error: ld returned 1 exit status

exit status 1
```
These I can fix by changing this:
```
    #else // !HAS_BED_PROBE

      void _lcd_babystep_z() { _lcd_babystep(Z_AXIS, PSTR(MSG_BABYSTEPPING_Z)); }
      void lcd_babystep_z() { lcd_goto_screen(_lcd_babystep_z); babysteps_done = 0; defer_return_to_status = true; }

    #endif // HAS_BED_PROBE
```
To this:
```
    #endif // HAS_BED_PROBE

    void _lcd_babystep_z() { _lcd_babystep(Z_AXIS, PSTR(MSG_BABYSTEPPING_Z)); }
    void lcd_babystep_z() { lcd_goto_screen(_lcd_babystep_z); babysteps_done = 0; defer_return_to_status = true; }
```

Again, I'm not sure if this is the right way to fix it.

mtowara, 2017-04-14:
I think thats not the correct fix. I think lcd_goto_screen needs to figure out if _z or _zoffset needs to be called.

I'm also playing around with those features right now and getting errors and strange behaviours left and right, i don't think its quite finished yet.

gcormier, 2017-04-14:
Yeah, current rcbugfix does not compile with babystepping. Same errors here.

Roxy-3D, 2017-04-15:
As a temporary fix...  You can try:  https://github.com/MarlinFirmware/Marlin/issues/6331#issuecomment-294245107

Tannoo, 2017-04-15:
I ran into the same:

In `ultralcd.cpp`:
```
  #if ENABLED(BABYSTEPPING)

    void _lcd_babystep(const AxisEnum axis, const char* msg) {
      if (lcd_clicked) { defer_return_to_status = false; return lcd_goto_previous_menu(); }
      ENCODER_DIRECTION_NORMAL();
      if (encoderPosition) {
        const int babystep_increment = (int32_t)encoderPosition * (BABYSTEP_MULTIPLICATOR);
        encoderPosition = 0;
        lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
        thermalManager.babystep_axis(axis, babystep_increment);
        babysteps_done += babystep_increment;
      }
      if (lcdDrawUpdate)
        lcd_implementation_drawedit(msg, ftostr43sign(planner.steps_to_mm[axis] * babysteps_done));
    }

    #if ENABLED(BABYSTEP_XY)
      void _lcd_babystep_x() { _lcd_babystep(X_AXIS, PSTR(MSG_BABYSTEPPING_X)); }
      void _lcd_babystep_y() { _lcd_babystep(Y_AXIS, PSTR(MSG_BABYSTEPPING_Y)); }
      void lcd_babystep_x() { lcd_goto_screen(_lcd_babystep_x); babysteps_done = 0; defer_return_to_status = true; }
      void lcd_babystep_y() { lcd_goto_screen(_lcd_babystep_y); babysteps_done = 0; defer_return_to_status = true; }
    #endif

    #if HAS_BED_PROBE

      void lcd_babystep_zoffset() {
        if (lcd_clicked) { defer_return_to_status = false; return lcd_goto_previous_menu(); }
        defer_return_to_status = true;
        ENCODER_DIRECTION_NORMAL();
        if (encoderPosition) {
          const int babystep_increment = (int32_t)encoderPosition * (BABYSTEP_MULTIPLICATOR);
          encoderPosition = 0;

          const float new_zoffset = zprobe_zoffset + planner.steps_to_mm[Z_AXIS] * babystep_increment;
          if (WITHIN(new_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX)) {

            if (planner.abl_enabled)
              thermalManager.babystep_axis(Z_AXIS, babystep_increment);

            zprobe_zoffset = new_zoffset;
            refresh_zprobe_zoffset(true);
            lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
          }
        }
        if (lcdDrawUpdate)
          lcd_implementation_drawedit(PSTR(MSG_ZPROBE_ZOFFSET), ftostr43sign(zprobe_zoffset));
      }

    #else // !HAS_BED_PROBE

      void _lcd_babystep_z() { _lcd_babystep(Z_AXIS, PSTR(MSG_BABYSTEPPING_Z)); }
      void lcd_babystep_z() { lcd_goto_screen(_lcd_babystep_z); babysteps_done = 0; defer_return_to_status = true; }

    #endif // HAS_BED_PROBE

  #endif //BABYSTEPPING
```
I changed it to this:
```
  #if ENABLED(BABYSTEPPING)

    void _lcd_babystep(const AxisEnum axis, const char* msg) {
      if (lcd_clicked) { defer_return_to_status = false; return lcd_goto_previous_menu(); }
      ENCODER_DIRECTION_NORMAL();
      if (encoderPosition) {
        const int babystep_increment = (int32_t)encoderPosition * (BABYSTEP_MULTIPLICATOR);
        encoderPosition = 0;
        lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
        thermalManager.babystep_axis(axis, babystep_increment);
        babysteps_done += babystep_increment;
      }
      if (lcdDrawUpdate)
        lcd_implementation_drawedit(msg, ftostr43sign(planner.steps_to_mm[axis] * babysteps_done));
    }

    #if ENABLED(BABYSTEP_XY)
      void _lcd_babystep_x() { _lcd_babystep(X_AXIS, PSTR(MSG_BABYSTEPPING_X)); }
      void _lcd_babystep_y() { _lcd_babystep(Y_AXIS, PSTR(MSG_BABYSTEPPING_Y)); }
      void lcd_babystep_x() { lcd_goto_screen(_lcd_babystep_x); babysteps_done = 0; defer_return_to_status = true; }
      void lcd_babystep_y() { lcd_goto_screen(_lcd_babystep_y); babysteps_done = 0; defer_return_to_status = true; }
    #endif
    void _lcd_babystep_z() { _lcd_babystep(Z_AXIS, PSTR(MSG_BABYSTEPPING_Z)); }
    void lcd_babystep_z() { lcd_goto_screen(_lcd_babystep_z); babysteps_done = 0; defer_return_to_status = true; }

    #if HAS_BED_PROBE

      void lcd_babystep_zoffset() {
        if (lcd_clicked) { defer_return_to_status = false; return lcd_goto_previous_menu(); }
        defer_return_to_status = true;
        ENCODER_DIRECTION_NORMAL();
        if (encoderPosition) {
          const int babystep_increment = (int32_t)encoderPosition * (BABYSTEP_MULTIPLICATOR);
          encoderPosition = 0;

          const float new_zoffset = zprobe_zoffset + planner.steps_to_mm[Z_AXIS] * babystep_increment;
          if (WITHIN(new_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX)) {

            if (planner.abl_enabled)
              thermalManager.babystep_axis(Z_AXIS, babystep_increment);

            zprobe_zoffset = new_zoffset;
            refresh_zprobe_zoffset(true);
            lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
          }
        }
        if (lcdDrawUpdate)
          lcd_implementation_drawedit(PSTR(MSG_ZPROBE_ZOFFSET), ftostr43sign(zprobe_zoffset));
      }

    #endif // HAS_BED_PROBE

  #endif //BABYSTEPPING
```
It really is no different that @benlye's fix, but back to what was in my previous compiling code. Functionality is the same (other than compiling issues).

benlye, 2017-04-15:
#6354 seems to fix this problem when applied to the latest ultralcd.cpp.

gcormier, 2017-04-15:
Can't compile with BABYSTEP_ZPROBE_OFFSET  enabled either.

benlye, 2017-04-16:
All set now.

closed 2017-04-16 by benlye