MarlinFirmware/Marlin #6347: Dual stepper driver feature request

outbound-pt, 2017-04-15:
Greetings,

Is there anyway to try and configure the [code]//#define Y_DUAL_STEPPER_DRIVERS[/code] feature in the [i]Configuration_adv.h[/i] file but without sacrificing the second extruder, instead remapping it to the third available extruder driver on the Megatronics v3.1? Would it be possible to do this by mapping the Y axis to more than one set of pins?

Bob-the-Kuhn, 2017-04-15:
Marlin always wants to map the first dual stepper to the first available extruder slot.

The easiest way is to lie to Marlin and swap the E2 & E3 pin mappings.  That way you can connect your second Y stepper to the physical E3 socket but Marlin now thinks it is the logical E2 slot.

If you ever get a second extruder then you'll need to swap them back but at least you don't have to mess with the Y2 cabling.

The only downside is you need to drag the modified pins_YOUR_BOARD.h file with you whenever you change to a newer version of software.

outbound-pt, 2017-04-15:
But that's the thing, I already have a dual extruder setup.

thinkyhead, 2017-04-15:
Marlin automatically uses the next unused E stepper. So all you have to do is set `EXTRUDERS` to **2**, then enable `Y_DUAL_STEPPER_DRIVERS`. Marlin will automatically use **E0** for your first extruder, **E1** for your second extruder, and **E2** for your **Y2** axis.

outbound-pt, 2017-04-15:
Ah, awesome!

I thought it would only go to E1 as the comments on the code seem to suggest.

outbound-pt, 2017-04-15:
Actually, I'm getting an error when trying to compile:

> #error  EXTRUDERS must be 1 with Y_DUAL_STEPPER_DRIVERS

I'm using Marlin 1.1.0 RC6 for Megatronics v3.1.

Bob-the-Kuhn, 2017-04-15:
Any chance of you going to RC8 or RCBugFix?  They don't have that issue.

You'll need to copy over your machine specific items into the new config files and change the motherboard to  BOARD_MEGATRONICS_31

outbound-pt, 2017-04-16:
Well, if that's the solution then I guess so. I've done it countless times before. This machine is originally a Leapfrog Creatr Dual. I switched the electronics so I've done the copy/paste of the main settings quite a few times already.

Thanks for the help.

closed 2017-04-17 by outbound-pt