MarlinFirmware/Marlin #6356: Feature request

batata004, 2017-04-15:
When I select PRE HEAT ABS my printer starts preheating the bed and the nozzle at the same time. The nozzle of course reaches the end rempetarute muuuch faster and it has to wait frequenty 5 to 10 minutes to the bed gets hot. Is there a way to make the nozzle only start heating after bed finished heating?

Roxy-3D, 2017-04-16:
That can be done...   But I'm busy with a few other problems.   If nobody else helps you...  In a few weeks remind me about this.

batata004, 2017-04-16:
@Roxy-3D  no hurry! I think it's not too hard to implement cause when I try to print from and SD card AND I SET the temperatute of the bed and the nozzle in the slicer, then Marlin with first heat up the bed and only after that the nozzle! This behaviour already happens when I use a SD card, I am just asking if it would be possible to implement some #define heat_bed_before_nozzle_when_heating 

It's just a feature request, nothing really necessary but would save some energy cause my heat bed takes almost 10 minutos to heat and while that the nozzle keeps the 240 celsius temperature doing"nothing" but waiting.

Roxy-3D, 2017-04-16:
> nothing really necessary but would save some energy cause my heat bed takes almost 10 minutos to heat and while that the nozzle keeps the 240 celsius temperature doing"nothing" but waiting.

Yeah...  I don't really care about the energy.  I care that the nozzle is cooking the filament.  First, it ooozes all over the place.  But some filament can get cooked too much and clog the nozzle.   If I were to do this, I might try to start heating the nozzle when the bed is 90% of the way warmed up.

teemuatlut, 2017-04-16:
> I think it's not too hard to implement cause when I try to print from and SD card AND I SET the temperatute of the bed and the nozzle in the slicer, then Marlin with first heat up the bed and only [...]

That's because the slicer will issue a [M109 Wait for hotend temperature](http://marlinfw.org/docs/gcode/M109.html) while using the LCD will do somethings equivalent of M104 and just sets the target temperature.

batata004, 2017-04-16:
@teemuatlut  yeap I can confirm that, the slicer is using the M109 command.

@Roxy-3D  that would be nice to have an option to set the % of the bed temperature to start heating the nozzle! 90% is good, but some people may wanna try 100% (like me) cause my printer really takes a looong time to heat the last degrees! The very last one I think take almost 1 minute!

Tannoo, 2017-04-17:
>The very last one I think take almost 1 minute!

Increasing the P on your pid will reduce that.

JohnOCFII, 2017-04-17:
It would seem the slicer GCODE pre-print settings would be a better place to handle this.  Here is my start code:

```
M140 S60 ; Set bed temperature (to 60C)
M190 S60 ; Wait for bed temperature to heat to temperature (60C)
M104 S230 T0 ; Set hot-end (tool 0) temperature (to 230C)
M109 S230 T0 ;Wait for bed temperature to heat to temperature (230C)
```

Also - here is the GCODE reference: http://reprap.org/wiki/G-code

Doing it this way keeps the firmware more streamlined.

bgort, 2017-04-21:
@JohnOCFII:  I agree -- GCODE is likely the best place for this.

Also, FYI, both the M140 and M104 are superfluous if you issue M190 and M109 immediately afterwards, as they both set temperature if it's not already set, then wait.  No need to set twice.  Not sure if you're using both, but thought I'd mention.

---

For whatever it's worth, this has worked best for me:
 - set extruder to 150C using M104 so it can preheat to slightly below the temp. where oozing starts
 - wait on the bed to reach the desired temp with M190 (if it isn't already there -- lately I've just been preheating and then holding the bed at 45C except when removing the previous print)
 - do whatever other pre-print things you might want to do, like homing, load mesh, activate leveling system, or whatever
 - finish heating the extruder to extrusion temp with M109 just before beginning the print

JohnOCFII, 2017-04-21:
@bgort - Yes, when I was posting that example, I noticed the the initial commands seemed superfluous.  It's the way Simplify3D 3.1.1 generates the code, though.  Not sure why.  I've been meaning to try to do some additional start code tweaks too, such as the two-stage pre-heat.

Thanks

bgort, 2017-04-21:
No problem.  

Yeah - I use S3D, too, and it does some weird things.  I think I remember seeing that or something similar when I initially set it up, too, actually.

batata004, 2017-04-21:
@bgort  your suggestion is really really really amazing! It will make me save lots of time initializing my prints!

Your idea of pre heating the be and nozzle and whilte that doing routines like homing, auto leveling... and only after that waiting to reach final temperature is PERFECT! So far I had first to wait the temperaures reach the ideal condition  and after that homing, auto leveling... GREAT IDEA MAN!!!

bgort, 2017-04-21:
Glad it helped, @batata004 !

closed 2017-04-21 by batata004