MarlinFirmware/Marlin #6364: Volumetric Extrusion with Filament Width Sensor

WrQth, 2017-04-16:
Just installed a filament width sensor and enabled in firmware. With sensor disabled (M406) printed a 20mm cube and came out fine but when I enabled sensor it over-extruded significantly. Can these two features not work with each other? Was hoping to have the sensor accomplish automatically what the M200 Dx.xx command does when volumetric extrusion is enabled.

Tannoo, 2017-04-16:
I have not had good luck with mine being reliable. So, I don't use it anymore.

Scratchbuild52, 2017-04-17:
I do not use volumetric Extrusion with the filament width sensor and get very good results. I am not sure if the volumetric extrusion is compatible with it. After all the filament width sensor just controls the percentage of the extrusion at a given point and should also apply to the volumetric extrusion. Check your other settings on your slicer, I suddenly had over extrusion when I started using the filament width sensor and realised it was my settings on the slicer that was over compensating for the filament width when I did not had it, now that the width is measured the settings had to be tuned in again.

WrQth, 2017-04-17:
I would think there could be a check while compiling if both enabled the functions related to reporting the filament width sensor measurement would set the variables that volumetric extrusion use such as the one related to the M200 command. Sounds like I have a reason to learn more about Arduino programming and the Marlin code project.

I use Simplify3D which does not natively support volumetric extrusion but was working fine before adding the sensor. The over-extrusion after added the sensor was significant, I turned down the extrusion multiplier while printing from 100% to 40% and it still extruded so much that the nozzle would visually deflect from its path. I have yet to print something with just the filament width sensor enabled but I guess that is my next step.

landodragon141, 2017-04-17:
Do you have filament width in Simplify3D set to: 1.128379? That's the value you should use if you're doing volumetric extrusion.

WrQth, 2017-04-17:
Yes I do have that set as the filament diameter in Simplify3D. I have had volumetric extrusion setup and printing well for a while now (months). I changed nothing in the slicer when adding the filament width sensor and enabled it in firmware. On the LCD I can see the measurement and it matches what I measure with my calipers. I'm thinking a variable being set by the sensor functions is overriding a variable used by volumetric which when only one or the other is enabled would be fine but I have both and I would think they'd be able to play nice with each other.

landodragon141, 2017-04-17:
Hmm sounds like a bug then.

WrQth, 2017-04-18:
I'd be happy to help debug but don't know where to start

WrQth, 2017-04-21:
I'm not familiar with how to debug if even possible to see what values are being set with functions for Volumetric and Width Sensor, anyone know if that can be done?

thinkyhead, 2017-04-26:
This is handled in `planner.cpp` in the vicinity of line 1045-1077. Logging can be added to that region of code, printing out the values being processed, and from this the problem should become obvious.

WrQth, 2017-04-26:
@thinkyhead what would be the corresponding lines in planner.cpp for the volumetric extrusion?

WrQth, 2017-04-26:
Would the value output through serial differ from the measured diameter displayed on screen when LCD options is enabled?

WrQth, 2017-05-05:
@thinkyhead should the word "value" be replaced with a variable name for which I want to output the value of? If that is the case what is the variable name I should be using?

WrQth, 2017-05-05:
@thinkyhead I don't know what values to have it display using SERIAL_ECHOLNPAIR("value:", value); I would think it would be a value used by volumetric calculation and is set by filament width sensor but I can't seem to find the correlation. Some more guidance would be helpful.

closed 2017-11-01 by thinkyhead