MarlinFirmware/Marlin #6366: Marlin restart when using Z

max246, 2017-04-16:
I have am using RC8 with Rumba to a custom 3d printer frame.

I have made a simple GCODE that draw 3 letters and I lift the Z axe to avoid to mess around the printing. Doing this operation few times, it triggers the software to restart without no reason.

This is the code:

```
G28 Z
G28
G1 Z50 E0.2
G1 X0.0 Y0.0
;U
G1 X120.0 Y0.0 E1.0
G1 X140.0 Y20.0 E1.2
G1 X140.0 Y40.0 E1.4
G1 X120.0 Y60.0 E1.6
G1 X0.0 Y60.0 E2.6
;N
G1 Z48
G1 X140.0 Y80.0
G1 Z50.0 E3.0
G1 X0.0 Y80.0 E4.0
G1 X140.0 Y100.0 E5.0
G1 X0.0 Y100.0 E6.0
;I
G1 Z48.0
G1 X140.0 Y120.0
G1 Z50.0 E7.0
G1 X0.0 Y121.0 E8.0
```

Is there something wrong on my side that makes the marlin software crash?

Bob-the-Kuhn, 2017-04-17:
RCBugFix has added some code to try to print out the reason for a crash.

Try reducing your feedrate by half.  Too high a feedrate can cause the software to crash.

Tannoo, 2017-04-17:
I tried your code after I deleted the E movements and have no issues. (I didn't want to extrude 50mm above the bed). I also changed the Z height, as I didn't need to move to 50.

Here's what I modified to run:
```
G28 Z       ; <-- This is not needed
G28          ; <-- Because of this.
G1 Z50
G1 X0.0 Y0.0
;U
G1 X120.0 Y0.0
G1 X140.0 Y20.0
G1 X140.0 Y40.0
G1 X120.0 Y60.0
G1 X0.0 Y60.0
;N
G1 Z48
G1 X140.0 Y80.0
G1 Z50.0
G1 X0.0 Y80.0
G1 X140.0 Y100.0
G1 X0.0 Y100.0
;I
G1 Z48.0
G1 X140.0 Y120.0
G1 Z50.0
G1 X0.0 Y121.0
```
This ran slow, but ran fine. So....I sped it up:
```
G28 Z       ; <-- This is not needed
G28          ; <-- Because of this.
G1 Z50
G1 X0.0 Y0.0 F8000
;U
G1 X120.0 Y0.0 F8000
G1 X140.0 Y20.0 F8000
G1 X140.0 Y40.0 F8000
G1 X120.0 Y60.0 F8000
G1 X0.0 Y60.0 F8000
;N
G1 Z48
G1 X140.0 Y80.0 F8000
G1 Z50.0
G1 X0.0 Y80.0 F8000
G1 X140.0 Y100.0 F8000
G1 X0.0 Y100.0 F8000
;I
G1 Z48.0
G1 X140.0 Y120.0 F8000
G1 Z50.0
G1 X0.0 Y121.0 F8000
```
This ran much faster and still ran fine.

Possible issues with your extruder? Because I didn't test that.

Tannoo, 2017-04-17:
You should whip up your letters in a CAD program, export them as STL, then run them through a slicer program, then print.
See what happens then.

max246, 2017-04-17:
I am using a modify version of printer and because i didn't find the flag
to remove raising the when X goes home, i had to add first Z home. This is
due i am using a cnc frame. I could potentially reverse the direction.

I have running the same rc8 on a prusa i3, not even an issue. I am creating
a custom printer and try to create the gcode for some samples with a custom
software.
I was wondering if what i created made the machine crash or is something
that has to be fixed.

I tested the whole thing without z moving and worked fine.

One thing that i didn't get is

Gcode x0 y10 e1.0
Gcode x0 y20 e2.0

This code will make my x to go negative after the second line. Do marlin
get confused if the previous x or y is sent the same?

max246, 2017-04-17:
Also,  is this issue present on the previous version , no RC? Because i
just need to move and extrude. I am not using other feature that a 3d
printer needs.

Basically i am using a syringe as extrude. I am using the model provide by
the reprap website. But i am using rumba as board for 24v compatibility and
having 3 extra extruder driver.

Bob-the-Kuhn, 2017-04-17:
Why don't you want Z to raise when homing?

It sounds like you've added some custom code.  What does the custom code do?

max246, 2017-04-17:
The problem is  I am using the OX CNC frame, and my Z zero is fully raised.
When the Z is lowered to 50 ( almost fully expended ), the printer moves
even further down before move the X axe. This will crash my syringe to the
bed.
As said before, probably the solution is just to invert the direction of
the motor and use negative value to lower it.

My custom code just convert black pixel from a picture to gcode to print
with the syringe. I tried to use some software but has to be command line
or a library in python.
The bigger issue I found is that current software use CNC style but I want
to use E to extrude the liquid. If you have any suggestion, I would like to
hear about it.

Roxy-3D, 2017-04-17:
> It sounds like you've added some custom code. What does the custom code do?

Have you made changes to the firmware?   Can we see what you did?    Because if you have not made changes to the firmware, it should not be resetting like you describe.

max246, 2017-04-17:
I have removed different safety as I am not using an hot end and dimension of the plate.
Here is my config : https://pastebin.com/aSA16MEa

But again, will I have less issue with the previous version? I am not sure why doing
GCODE x0.0 y10.0
GCODE x0.0 y20.0 
will make the X motion crazy.

Bob-the-Kuhn, 2017-04-17:
You probably have Z_HOMING_HEIGHT disabled and  Z_CLEARANCE_BETWEEN_PROBES enabled & set to 5 (the defaults).  You can kill the Z raise during G28 by setting Z_HOMING_HEIGHT to zero or by disabling Z_CLEARANCE_BETWEEN_PROBES .

The next thing that should be done is to automatically get the tip of the syringe to the desired printing height .  After G28 issue G0 Z0, lower the syringe to the correct height and then issue M114.  If, for example, M114 says Z is -389.675 then set Z_MIN_POS to +389.675.  Once that's been setup, after G28, a G0 Z0 command will set the syringe to the correct height,

How critical is the distance between the bed and the tip of the syringe?  Most beds are not perfectly flat and have some tilt to them.  If you need to keep the nozzle less than a couple om mm from the bed then you'll probably need to use one of the bed leveling options.  We can attack that after we get things running correctly.

max246, 2017-04-17:
the nozzle of the syringe is not critical as it is liquid and is pretty sticky. No need to worry about bed levelling.
Thanks for your suggestion, I will check tomorrow.

What about the Gcode generation, there is any guide lines that I should follow? Due to my issue above that it reset the software, how can I solve this? Just go back to the normal stable version?
Would be good to have the Z motion between letters so it will avoid to drag any excess liquid on the surface.

Bob-the-Kuhn, 2017-04-17:
Do you really need a temperature sensor?  If not then change TEMP_SENSOR_0 from 1 to 998 and disable PREVENT_COLD_EXTRUSION.

ENDSTOPPULLUPS is almost always enabled.  Is there a reason it's commented out?

To prevent depositing liquid during travel the usual methods are to raise Z and to retract the syringe..

Another way to get around the Z homing issue is to home in the +Z direction.  That requires a Z_MAX endstop.  Homing this way means you don't have to worry about how high the Z axis is when doing G28.

Have you modified the Marlin code (files other than Configuration.h and Configuration_adv.h)?  If yes then we'll want to see the changes and get an explanation as to what it's expected to do.  Re-reading the comment I expect that the custom code is to generate the G-codes.

You've been able to convert the picture to CNC style G-codes.  If you knew some type of macro/scripting/basic language then you could write a program that converted the negative Z movements to e0 movements.

You might also look at converting the picture to vectors first and then to 3D printing.

You're running a straight Cartesian printer so doing a Y only move shouldn't ever result in an X movement.  

Exactly what do you mean by the X going crazy?  Is it banging up against the X min physical stop, is it going in the + direction?  Does it misbehave when doing something like G0 X50 Y50 then G0 X50 Y 100?

max246, 2017-04-17:
I tried to put TEMP_SENSOR_0  -1 to disable but apparently I was wrong because it didnt let me compile it, I see now you are suggesting 998 to disable it, I will give a try. I have disable the protection for cold extrusion.

I havent touch that, I have just configured the endstop that are inverted ( just few lines below ), they work, when I press home all the axes go to the endstop and stop.

I will do the Z tricky as you suggested before then lets see if it will be solved.

I have no modified any other files, just the configuration.

How did you conver the picture to CNC? Using some sort of software? Would be good I get something that works under linux and is command line. I know Python and different other languages so for this project I am sticking to Python but I could use multiples.

For the gcode issue, I am saying that doing 
G1 X0.0 Y10.0 E1.0  worked fine
Then by telling to do:
G1 X0.0 Y20.0 E2.0 this means it will keep the same X but moving by 10 in Y axe.
The issue I am facing is that the printer try to move to Y20 but also going to negative X which cant do because it reached the endstop and the motor keeps forcing it. If I remove X0.0 on the second line it works fine

Bad gcode
G1 X0.0 Y10.0 E1.0 
G1 X0.0 Y20.0 E2.0

Good gcode
G1 X0.0 Y10.0 E1.0 
G1 Y20.0 E2.0

I am not sure if there is a guide line for making sure what I make as gcode is correct because looking at the one made by simplify3d, each line has a different X Y so they dont face this issue.

I am sorry, I might sound so stupid but it is hard to understand if the issue is due to a RC version or I should go back to stable version.

max246, 2017-04-17:
In case you were wondering how was my setup: https://s17.postimg.org/h28d2p0hb/IMG_20170413_204436.jpg still WIP so stuff are going to improve

Bob-the-Kuhn, 2017-04-17:
Cable ties?  Personally I prefer cable ties & duct tape.

Your machine is a lot cleaner than mine.  I got it 99% functional and then stopped.  Wires everywhere.

You've just modified the two configuration files.  That's good news.

I'm having a **REALLY** hard time imagining why you're getting X movement when doing G1 X0 Y10 & then G1 X0 Y20 but not with  G1 X0 Y10 & then G1 Y20.  

Do you get X movement with this sequence?
G1 X10 Y10.0 E1.0
G1 X010 Y20.0 E2.0

Please do the following and then post the results.
G28
M114
G1 X0.0 Y10.0 E1.0
M114
G1 X0.0 Y20.0 E2.0
M114

Sanity check.  Is X0 when the head is at the far left of the travel and Y0 when the gantry is as far forward as possible?

As to which flavor of Marlin to use ... I seriously doubt that the X problem is due to the flavor of Marlin.  If we don't get to the bottom of the X problem today I'd like you to move to [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix) just because it has a tool I'd like you to use.  

I'm wondering if there is a second function assigned to the X_STEP pin.  Once you've downloaded RCBugFix and transferred your machine specific items over to the new config files then enable PINS_DEBUGGING in Configuration_adv.h and issue M43 I to the printer.

M43 I will print out all the pins and all the signals/functions assigned to them.  See if the X_STEP pin has other signal(s) assigned to it.

max246, 2017-04-17:
Thanks, I have tried to keep it clean as possible, I have cleaned up my Prusa i3 few times so I am trying to be more precise.

I cant understand the motion of the X either, not even why by doing the first code at the first post that will restart marlin. 

Tomorrow I will try to do all the tests and can post it here. Are you based in USA or Europe? I could do tests over IRC chat if you prefer.

The only thing that comes to mind is that I am not using RepRap but Rumba, but it should be the same thing + an extra driver.

Bob-the-Kuhn, 2017-04-17:
I’m in the USA.

 

Tomorrow is pretty busy with DR. appointments.

 

  Bob

max246, 2017-04-18:
Ok, well I am in Europe so hopefully my test are going to be your late afternoon. Lets continue then.

max246, 2017-04-18:
Update:
- Update the temp sensor to 998 worked
- Disabled Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES removed the lower of the Z when X going home
- Set the right Z_MIN_POS

I am stilling facing the issue of the printer going a bit crazy after doing 
G1 Z-1 E11.0 
;Draw letter
G1 Z0 E11.0 

After the second letter it just send the printer to the outside limit of Y ( not X this time ) and I have to unplug the power. It seems like it goes into an infinite loop because if I send another command while is doing this motion, the serial port doesnt respond anymore and I have to physically remove the usb.

The only solution is to run my code without the Z motion and the gcode works.

If you got any enlightenment about this, would be great.

Tannoo, 2017-04-18:
Z steps/mm incorrect?
Bad wiring on Z?
Bad stepstick for Z?

max246, 2017-04-18:
The Z axe works totally fine, it is the marlin software that freak out if do:
G1 Z-1 E11.0
;Draw letter
G1 Z0 E11.0

If I remove the G1 Z operations, printer does the whole printing fine, I am not sure if what the printer goes mad if I use so many times the G1 Z codes.

Tannoo, 2017-04-18:
Mattercontrol has a software based bed leveling system built into the slicer. I ran that once and it sends Z movements all the time.

Printer printed just fine, but Octoprint just wigged out thinking that it was changing layers all the time.

Your issue is definately a strange one.

max246, 2017-04-18:
Yep, I am thinking that I am doing something wrong on the code, do you have an example of your GCODE?

cormier8, 2017-04-19:
Hi,

May be it can help you :

I got the same problem with a K8400 
- X and Y movements worked fine with Gcode,
- but most of the time, when trying to move Z with Gcode, the controller restart.

Finaly, I set Z DEFAULT_MAX_FEEDRATE from 10 mm/s to 8 mm/s and everything works fine now.

Bob-the-Kuhn, 2017-04-20:
6000 steps per unit on your extruder is pretty high.

Do you have problems if you remove the extruder portion of the command?

Usually if the feedrate and steps per unit are too high Marlin will be so busy that the WDT times out.   

I just used your settings & had no issue.

I like the idea of dropping your max feed rate down.  In your case I suggest dividing it by 10.

thinkyhead, 2017-04-26:
> 6000 steps per unit on your extruder is pretty high.

And that's the problem. If you try to do too many steps-per-second you have no CPU left to do anything else, so the watchdog timer gets invoked and resets the board.

I suggest reducing your microsteps by half, if you can't get the speeds you want on Z.

max246, 2017-04-26:
Apparently lowering the feedrate in the config, solved the issue, now I can do different Z motion within the printer.

I will close the ticket in 5 days if nothing happens. Thanks for you help!!

thinkyhead, 2017-04-29:
Reopen this issue if the problem recurs.

closed 2017-04-29 by thinkyhead