MarlinFirmware/Marlin #6369: BABYSTEPPING doesn't move as LCD shows

Tannoo, 2017-04-17:
With `DOUBLECLICK_FOR_Z_BABYSTEPPING` on, the LCD shows movement that is not happening.

This is when the printer is not printing and doesn't matter if it is homed or not.

closed 2017-04-17 by Tannoo

Kaibob2, 2017-04-17:
@Tannoo I cannot confirm this. On [RCBugFix from 12.4.2017](https://github.com/MarlinFirmware/Marlin/commit/6d5400dc318d77502aaacced64db62fac5501cd1) Z-Babystepping via Doubleclick works, no matter if homed or not, like intended. When i double click the Encoder and turn it afterwards the Z-steppers get powered and move.

Kaibob2, 2017-04-17:
Oops, I should've typed faster :)

Tannoo, 2017-04-17:
The babystepping mulplicator isn't as "sensitive" as it once was. And a recent change also made babystepping much more silent. I did find that it works just fine.