MarlinFirmware/Marlin #6372: Dual-X-Carriage Acceleration not Working on Both ToolHeads

gsart84, 2017-04-17:
Hello,
i have Problem with acceleration (Dual-X-Carriage with 2 independend stepper motors and stepsticks.)
if i change the acceleration values extreme, like from 4500 to 200 i can see that the acceleration only changes to T1 in a dual color print. The acceleration seems to me is not changing to T0. (rattering,vibrations on T0 and smooth moves on T1. if i print with T0 only or T1 only, it seems de accelerations are correct.

any ideas? anyone similar problem?

dp90pusika, 2017-04-17:
Same thing here, I use a RADDS board with arduino due, so I use the 32bit version.
@gsart84 which control board are you using?

gsart84, 2017-04-17:
@dp90pusika 
i use the Rumba Board.
i have not see any possibility to assign a individual acceleration for second extruder.

dp90pusika, 2017-04-17:
It's probably gets the default settings, and after that doese't refreshes it when you change the walues with the M commands. You can probably hunt down the missing lines, by finding the corresponding M code script, and complement it with the missing code to change the walues for both tools.

gsart84, 2017-04-17:
@dp90pusika seems a little bit to complicated for me. i dont know where to find the  "corresponding M code script"

dp90pusika, 2017-04-17:
@gsart84 I looked inti it quickly, if you are using an LCD controler to change the acceleration than theres no option for that right now, but if you send custom M codes via USB communication, you can do it.
example: M201 X1000 Y1000 T0 (or T1 depending which toolhead you vant to modeify)

gsart84, 2017-04-17:
@dp90pusika
thanks  
My eeprom looks like this:

Recv: echo:V27 stored settings retrieved (410 bytes)
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y100.00 Z4000.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X150.00 Y150.00 Z3.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X700 Y700 Z100
Recv: echo:Accelerations: P=printing, R=retract and T=travel
Recv: echo:  M204 P200.00 R3000.00 T2000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X11.00 Y11.00 Z0.40 E5.00
Recv: echo:Home offset (mm)
Recv: echo:  M206 X0.00 Y-14.00 Z0.00
Recv: echo:Hotend offsets (mm)
Recv: echo:  M218 T1 X0.00 Y0.30 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H200 B50 F0
Recv:   M145 S1 H220 B80 F0
Recv: echo:PID settings:
Recv: echo:  M301 P26.95 I2.52 D72.19
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 T1 D3.00
Recv: echo:  M200 D0
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.45

if i send: M201 X700 Y700 Z100 T0 or M201 X700 Y700 Z100 T1
or i: M204 P200.00 R3000.00 T2000.00 T0 
or: M204 P200.00 R3000.00 T2000.00 T1

i cant see for example a new line with T0 or T1

does it not show in the M501?

dp90pusika, 2017-04-17:
Can't really tell howe should it work, sadly I can't test it right now because it's busted for me, can't compile it for 32 bit board.
I only referred to this section in Marlin:main.cpp :

```
/**
 * M201: Set max acceleration in units/s^2 for print moves (M201 X1000 Y1000)
 *
 *       With multiple extruders use T to specify which one.
 */
inline void gcode_M201() {

  GET_TARGET_EXTRUDER(201);

  LOOP_XYZE(i) {
    if (code_seen(axis_codes[i])) {
      const uint8_t a = i + (i == E_AXIS ? TARGET_EXTRUDER : 0);
      planner.max_acceleration_mm_per_s2[a] = code_value_axis_units(a);
    }
  }
  // steps per sq second need to be updated to agree with the units per sq second (as they are what is used in the planner)
  planner.reset_acceleration_rates();
}

#if 0 // Not used for Sprinter/grbl gen6
  inline void gcode_M202() {
    LOOP_XYZE(i) {
      if (code_seen(axis_codes[i])) axis_travel_steps_per_sqr_second[i] = code_value_axis_units(i) * planner.axis_steps_per_mm[i];
    }
  }
#endif

```

Bob-the-Kuhn, 2017-04-17:
By chance is DISTINCT_E_FACTORS enabled?  If it is enabled then you can have different settings for the two extruders.  If it's disabled then they will be treated the same.

If you do enable DISTINCT_E_FACTORS then you'll need to add another term to DEFAULT_AXIS_STEPS_PER_UNIT , DEFAULT_MAX_FEEDRATE & DEFAULT_MAX_ACCELERATION.  Right now they're of the form { a, b, c, d}.  They'll need to be  { a, b, c, d, e} where d & e are the same for your setup.

What version of Marlin are you running?  RC8 has some known issues with DISTINCT_E_FACTORS.  They are fixed in [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).  If you switch to RCBugFix then you'll manually need to move your machine specific settings over to the new config files.

gsart84, 2017-04-17:
wow, thanks @Bob-the-Kuhn it seems exactly what im searching for.
first i will try  with my actual rc8. the rcbugfix has an another yet not solved problem about the M600, i would like to try the rc8 first.

now my config is like this:
/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 700, 700, 100, 10000 }

i use X, Y, Z, E0, E1 and my X2 is connectet on the E3 Stepstick,

now it should looklike this?

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 700, 700, 100, 10000, 10000, 700 }

Thanks

Bob-the-Kuhn, 2017-04-17:
Marlin assumes that X2 uses the same settings as X which means there isn't an X2 term in the  {xx, yy, ...} setting.  That means, with DISTINCT_E_FACTORS disabled, { 700, 700, 100, 10000 }  should be correct.  With DISTINCT_E_FACTORS enabled, { 700, 700, 100, 10000 , 10000} would be correct.

gsart84, 2017-04-18:
@Bob-the-Kuhn 
distinct_e_factors are now enabled.

now im on RCBugFix (i found the problem about the M600)

ok, now i do it like this:
-i adjust the accelleration settingsfor x axis in configuration.h.
-then i clear the eeprom.
-now the acceleration for X, X2 and Y works perfect.
if i change the acceleration in LCD, only T1 changes. T0 stays like Firmware settings.

Bob-the-Kuhn, 2017-04-18:
I'm not familiar with what's available in the LCD system.  Can you switch to T0 and do the update?

Is the change done by the LCD saved to EEPROM or will the old value come back after a power cycle?

BlocksTechnology, 2018-01-19:
@gsart84 i'm also trying to use dual x carriage with a rumba board but im having problems in my configuration since my endstops and motors are not working properly. could you help by putting here your "configuration.h" and "configuration_adv.h"? 

My X2 motors is not working at all and I also connected to the E3 position.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-18:
@gsart84 was the problem fixed with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-17 by Roxy-3D