MarlinFirmware/Marlin #6374: 32-Bit-RCBugFix-new Compiling error

dp90pusika, 2017-04-17:
Tried to test this new 32-Bit-RCBugFix-new branch but it cant be compiled in the current state.
There's some problems since the: (Indent MarlinSerial code) commit:  eaa66f3c464f2e09bb093e5f0285ced971b48724

I use an Arduino Due with RADDS board, in a dual x carriage setup.

compiling error:

```
In file included from sketch\serial.cpp:23:0:

serial.h:41: error: expected initializer before 'PROGMEM'

 extern const char echomagic[] PROGMEM;

                               ^

serial.h:42: error: expected initializer before 'PROGMEM'

 extern const char errormagic[] PROGMEM;

                                ^

serial.h:79: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, uint8_t v) { serial_echopair_P(s_P, (int)v); }

 ^

serial.h:80: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, uint16_t v) { serial_echopair_P(s_P, (int)v); }

 ^

serial.h:81: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, bool v) { serial_echopair_P(s_P, (int)v); }

 ^

serial.h:82: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serial_echopair_P(const char* s_P, void *v) { serial_echopair_P(s_P, (unsigned long)v); }

 ^

serial.h:87: error: 'FORCE_INLINE' does not name a type

 FORCE_INLINE void serialprintPGM(const char* str) {

 ^

serial.cpp:25: error: expected initializer before 'PROGMEM'

 const char errormagic[] PROGMEM = "Error:";

                         ^

serial.cpp:26: error: expected initializer before 'PROGMEM'

 const char echomagic[] PROGMEM = "echo:";

                        ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, const char*)':

serial.cpp:28: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, const char *v)   { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:48: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

sketch\serial.h:57:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

sketch\serial.cpp:28:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, const char *v)   { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, char)':

serial.cpp:29: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, char v)          { serialprintPGM(s_P); SERIAL_CHAR(v); }

                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:44: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_CHAR(x) (MYSERIAL.write(x))

                         ^

sketch\serial.cpp:29:81: note: in expansion of macro 'SERIAL_CHAR'

 void serial_echopair_P(const char* s_P, char v)          { serialprintPGM(s_P); SERIAL_CHAR(v); }

                                                                                 ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, int)':

serial.cpp:30: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, int v)           { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:48: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

sketch\serial.h:57:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

sketch\serial.cpp:30:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, int v)           { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, long int)':

serial.cpp:31: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, long v)          { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:48: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

sketch\serial.h:57:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

sketch\serial.cpp:31:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, long v)          { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, float)':

serial.cpp:32: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, float v)         { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:48: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

sketch\serial.h:57:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

sketch\serial.cpp:32:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, float v)         { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, double)':

serial.cpp:33: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, double v)        { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:48: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

sketch\serial.h:57:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

sketch\serial.cpp:33:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, double v)        { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

sketch\serial.cpp: In function 'void serial_echopair_P(const char*, long unsigned int)':

serial.cpp:34: error: 'serialprintPGM' was not declared in this scope

 void serial_echopair_P(const char* s_P, unsigned long v) { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                              ^

In file included from sketch\serial.cpp:23:0:

serial.h:48: error: 'MYSERIAL' was not declared in this scope

 #define SERIAL_PROTOCOL(x)                  (MYSERIAL.print(x))

                                              ^

sketch\serial.h:57:40: note: in expansion of macro 'SERIAL_PROTOCOL'

 #define SERIAL_ECHO(x)                 SERIAL_PROTOCOL(x)

                                        ^

sketch\serial.cpp:34:81: note: in expansion of macro 'SERIAL_ECHO'

 void serial_echopair_P(const char* s_P, unsigned long v) { serialprintPGM(s_P); SERIAL_ECHO(v); }

                                                                                 ^

exit status 1
expected initializer before 'PROGMEM'

```

bobc, 2017-04-17:
serial.h is missing an include \:

`#include "src/HAL/HAL.h"`

dp90pusika, 2017-04-17:
Thank you @bobc this fixed it.

thinkyhead, 2017-04-17:
Now patched in the branch also.

closed 2017-04-17 by thinkyhead