MarlinFirmware/Marlin #6377: Issuers with 32-Bit-RCBugFix-new when compiling with LCD enabled

dp90pusika, 2017-04-17:
I tried to activate the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
and compile the script and I got the following error:

```
sketch\ultralcd.cpp: In function 'void lcd_sdcard_pause()':

ultralcd.cpp:660: error: expected ';' before '}' token

     }

     ^

sketch\ultralcd.cpp: In function 'void lcd_sdcard_resume()':

ultralcd.cpp:669: error: expected ';' before '}' token

     }

     ^

sketch\ultralcd.cpp: In function 'void lcd_status_printf_P(uint8_t, const char*, ...)':

ultralcd.cpp:3573: error: 'vsnprintf_P' was not declared in this scope

   vsnprintf_P(lcd_status_message, 3 * (LCD_WIDTH), fmt, args);

                                                             ^
exit status 1
expected ';' before '}' token

```
This is easyli fixed by adding two ':' to the right lines:

```
 #if ENABLED(SDSUPPORT)

    void lcd_sdcard_pause() {
      card.pauseSDPrint();
      print_job_timer.pause();
      #if ENABLED(PARK_HEAD_ON_PAUSE)
        enqueue_and_echo_commands_P(PSTR("M125"));
      #endif
    }

    void lcd_sdcard_resume() {
      #if ENABLED(PARK_HEAD_ON_PAUSE)
        enqueue_and_echo_commands_P(PSTR("M24"));
      #else
        card.startFileprint();
        print_job_timer.start();
      #endif
    }

```
But after this, another problem arises, and I dont know what to do here:

```
sketch\ultralcd.cpp: In function 'void lcd_status_printf_P(uint8_t, const char*, ...)':

ultralcd.cpp:3573: error: 'vsnprintf_P' was not declared in this scope

   vsnprintf_P(lcd_status_message, 3 * (LCD_WIDTH), fmt, args);

                                                             ^

exit status 1
'vsnprintf_P' was not declared in this scope

```

p3p, 2017-04-18:
I think vsnprintf_P is AVR specific, put the following in HAL_due.h after the headers and see if that helps
```
#ifndef vsnprintf_P
  #define vsnprintf_P vsnprintf
#endif
```
You are running a very experimental branch here, not sure anyone had tested this before you started posting issues ^^

dp90pusika, 2017-04-18:
Thank you @p3p this fixed the problem.
Sorry if I'm making extra problems, but I'm using a dual x carriage machine with an arduino Due controller in it.
And this branch looks really promising, all the latest features are in it, would reali like to implement them on my machine. 
Right now it is compiling al right, but I will only test it on the machine tomorow.

p3p, 2017-04-18:
It most definitely wasn't a complaint can't fix things if no one is using it, The problem with the 32bit branch is that I'm pretty sure none of the people working on it are using a Due, even though that is the only platform currently supported.

dp90pusika, 2017-04-18:
In that case I'm more than happy to test this 32bit branch 🙂
The 32bit ARM processors are the way to go forward but this is just my opinion, I would like to stick to at least 32 times microstepping and right now I have 7 stepper controllers, 2X 1Y 2Z 2E, so I need the extra speed

dp90pusika, 2017-04-18:
Ok, first test print is running from sd card and LCD screen. This is a normal print with onli T0 toolhead, everithing works fine sofar.

Babystepping can't be enabled, some compiling error comes up if I doo:
It misses som timing define, but cant find this TCNT0 in any of the HAL-scripts

sketch\stepper.cpp: In static member function 'static void Stepper::babystep(AxisEnum, bool)':

```
stepper.cpp:1255: error: 'TCNT0' was not declared in this scope

     #define _SAVE_START (pulse_start = TCNT0)

                                        ^

sketch\stepper.cpp:1264:7: note: in expansion of macro '_SAVE_START'

       _SAVE_START; \

       ^

sketch\stepper.cpp:1288:9: note: in expansion of macro 'START_BABYSTEP_AXIS'

         START_BABYSTEP_AXIS(X, false);

         ^

stepper.cpp:1256: error: 'TCNT0' was not declared in this scope

     #define _PULSE_WAIT while ((uint32_t)(TCNT0 - pulse_start) < STEP_PULSE_CYCLES - CYCLES_EATEN_BY_BABYSTEP) { /* nada */ }

                                           ^

sketch\stepper.cpp:1270:7: note: in expansion of macro '_PULSE_WAIT'

       _PULSE_WAIT; \

       ^

sketch\stepper.cpp:1289:9: note: in expansion of macro 'STOP_BABYSTEP_AXIS'

         STOP_BABYSTEP_AXIS(X);

         ^

stepper.cpp:1255: error: 'TCNT0' was not declared in this scope

     #define _SAVE_START (pulse_start = TCNT0)

                                        ^

sketch\stepper.cpp:1264:7: note: in expansion of macro '_SAVE_START'

       _SAVE_START; \

       ^

sketch\stepper.cpp:1294:9: note: in expansion of macro 'START_BABYSTEP_AXIS'

         START_BABYSTEP_AXIS(Y, false);

         ^

stepper.cpp:1256: error: 'TCNT0' was not declared in this scope

     #define _PULSE_WAIT while ((uint32_t)(TCNT0 - pulse_start) < STEP_PULSE_CYCLES - CYCLES_EATEN_BY_BABYSTEP) { /* nada */ }

                                           ^

sketch\stepper.cpp:1270:7: note: in expansion of macro '_PULSE_WAIT'

       _PULSE_WAIT; \

       ^

sketch\stepper.cpp:1295:9: note: in expansion of macro 'STOP_BABYSTEP_AXIS'

         STOP_BABYSTEP_AXIS(Y);

         ^

stepper.cpp:1255: error: 'TCNT0' was not declared in this scope

     #define _SAVE_START (pulse_start = TCNT0)

                                        ^

sketch\stepper.cpp:1264:7: note: in expansion of macro '_SAVE_START'

       _SAVE_START; \

       ^

sketch\stepper.cpp:1303:11: note: in expansion of macro 'START_BABYSTEP_AXIS'

           START_BABYSTEP_AXIS(Z, BABYSTEP_INVERT_Z);

           ^

stepper.cpp:1256: error: 'TCNT0' was not declared in this scope

     #define _PULSE_WAIT while ((uint32_t)(TCNT0 - pulse_start) < STEP_PULSE_CYCLES - CYCLES_EATEN_BY_BABYSTEP) { /* nada */ }

                                           ^

sketch\stepper.cpp:1270:7: note: in expansion of macro '_PULSE_WAIT'

       _PULSE_WAIT; \

       ^

sketch\stepper.cpp:1304:11: note: in expansion of macro 'STOP_BABYSTEP_AXIS'

           STOP_BABYSTEP_AXIS(Z);

           ^

exit status 1
'TCNT0' was not declared in this scope

```

teemuatlut, 2017-04-18:
TCNT0 is the 8bit AVR platform current timer count and is not present in the 32bit chips.

The DUE HAL is not quite there yet and will need some attention and preferably an active developer. I've ordered a DUE clone to see if I can help out but it'll be a while for it to ship from China.

p3p, 2017-04-18:
Looks like that hasn't been ported yet, or a rebase has reintroduced some AVR code. (TCNT0 is an AVR timer register). I'm afraid only a small subset of the config combinations have been tested and I've never used baby stepping.

dp90pusika, 2017-04-18:
@p3p you have a point, I also never used the baby stepping, just figured I would test it if it works.
Also the #define ADVANCE is also missing the same TCNT0
These are not esential for my machine, just wanted to try them out.

But there's a much more needed things for my dual x carriage machine, trough the LCD there is no option to move the second toolhead maualy, and the filament change script only applys for T0. 
The filament change and runout detection for both toolheds is a really powerfull and usefull tool to have when your machine has a 320X450x480mm a buildvolume :D

dp90pusika, 2017-04-18:
> Ok, first test print is running from sd card and LCD screen. This is a normal print with onli T0 toolhead, everithing works fine sofar.

Big mess up on my side, I wasn't actually using this version of the firmware
Actually the printer locks in a bootloop when uploaded, but I see someone already started am issue on this

p3p, 2017-04-18:
Sorry to hear that, although its not entirely unexpected I was surprised when you said you had it working,  There have been substantial changes to the main Interrupt handlers since the Due platform was written, hopefully @teemuatlut can help once he gets the hardware to test on. I intend to work on the 32bit platform over summer time permitting, I was hoping to get my hands on a Re-arm which is LPC1768, but I'l see about getting a Due as well. My printer currently uses an STM32M7 but I don't think there will ever be a commercial board that uses it so not the best mcu to target.

As a guess, you could try disabling the sei and cli calls in HAL_Due.cpp,  (just comment out the contents of the functions), though it's a dodgy workaround at best, in the unlikely event it boots it will probably be unstable.

Roxy-3D, 2017-04-18:
> Looks like that hasn't been ported yet, or a rebase has reintroduced some AVR code. (TCNT0 is an AVR timer register). I'm afraid only a small subset of the config combinations have been tested and I've never used baby stepping.

The code to do the BabyStepping is in the timer interrupt just because it needs to be done over time.   On the 32-Bit platforms, it would be helpful to get a few timer tasks running that can have logic that gets checked periodically.  If we can get an RTOS under Marlin on the 32-Bit platforms, those kind of things can be their own task that gets sent a message and just sleep until there is something to do.

> I intend to work on the 32bit platform over summer time permitting, I was hoping to get my hands on a Re-arm which is LPC1768, but I'l see about getting a Due as well. 

The Re-Arm board is in a number of people's hands here.    That would be a very good one to do your 32-Bit work on.

apballard, 2017-04-21:
@Roxy-3D BTW do you know if RE-ARM is commercially available yet? I would love to get one and Do some testing. Do we have any available code for it?

Roxy-3D, 2017-04-21:
> BTW do you know if RE-ARM is commercially available yet? I would love to get one and Do some testing. Do we have any available code for it?

I think it is only available through the Kick-Starter campaign right now.   But it is one of the agreed upon 32-bit boards that people are trying to get Marlin up and running on.

p3p, 2017-04-21:
@apballard I think [panucatt](http://www.panucatt.com/ProductDetails.asp?ProductCode=RA1768 ) is the only source for Re-Arm atm

TGMods, 2017-04-21:
The Re-ARM is available from Panucatt, they have them in stock right now.

closed 2017-11-01 by thinkyhead