MarlinFirmware/Marlin #6378: How to Disable Microstepping?/Motors make loud noise upon disabling

kcheeeung, 2017-04-18:
Hey guys, I have a problem with the firmware that you may be able to help me solve. On the Azteeg X3 Pro board, I have 1/16 stepping set with the settings found on my Git. https://github.com/kcheeeung/MarlinRC8git

When I disable all microstepping by removing all the jumpers SD6128 drivers (for 1 full step), the motors make a really loud noise. I have Nema 17 17HS16 2004S1 motors. Is there something in the firmware that I must change in order to have no microstepping on the extruders? I tried setting the jumpers on 1/2 steps and it also didn't work.

http://www.omc-stepperonline.com/3d-printer-nema-17-stepper-motor-2a-45ncm64ozin-17hs162004s-p-16.html

Roxy-3D, 2017-04-18:
> stepper-motor-2a-45ncm64ozin-17hs162004s-p-16.html

You probably need to change the number of steps per unit of movement.   But
really...  Micro-steps are a good thing.

kcheeeung, 2017-04-18:
@Roxy-3D For my purpose, I do **not** want microstepping. This is to increase the overall max torque. We are using to an extrusion system that requires high torque.

Roxy-3D, 2017-04-18:
OK!  Thank you for explaining!   That makes sense!

Back to your problem...  There is not much in the firmware concerning micro-stepping.  If your electronics has the stepper driver chips connected up to the AVR chip, it can control the micro-stepping.  But that is very unusual.   Usually there are jumpers on the controller board that control the number of micro steps.    Looking at:  http://files.panucatt.com/datasheets/x3pro_wiring.pdf  It would seem like your board is controlled with jumpers.

So if you tell it not to have any micro-steps, you probably need to divide the number of steps per unit of travel by 16 in your Configuration.h file.   But free advice is worth what you pay for it!

fiveangle, 2017-04-18:
> When I disable all microstepping by removing all the jumpers SD6128 drivers (for 1 full step), the motors make a really loud noise.

Correct ! :)

Bob-the-Kuhn, 2017-04-18:
With microstepping you trade off noise vs. power.

If you want low noise and power then you'll need to go to motors that have more power than you need and use microstepping.

What voltage are you using on your steppers?  If you're using 12V then I'd seriously look at going to 24V.  Your steppers and the controller can handle 24V.

I didn't find anything saying your 6128 drivers were anything special when it came to motor noise other than it had a true 128 microstep mode.  What little I could find on the decay mode was that the mixed mode seems to be preferred and the fast mode is noisiest.

kcheeeung, 2017-04-21:
Hey guys. Thanks for the replies. I figured out that reducing the acceleration helps a bit.

Also how would I calculate the steps/mm for a planetary gear box? I am confused about the proper way to do this. I have a 14:1 Nema 17 planetary gearbox.

https://www.amazon.com/gp/product/B00QA8XUGC/ref=ox_sc_act_title_1?ie=UTF8&psc=1&smid=AWQBCGWISS7BL

Bob-the-Kuhn, 2017-04-21:
The best/easiest way is to command a movement, measure the actual movement and use that ratio to adjust your steps per unit.  Start out with a small movement (one you know is safe) & repeat until you can accurately move 90% of the travel limit.

thinkyhead, 2017-04-26:
> Also how would I calculate the steps/mm for a planetary gear box?

Let the compiler do the work for you.
```
// Your extruder parts:
#define NEMA17_FULL_STEPS 200.0   // 1.8 degrees per whole step
#define NEMA17_MICROSTEPS 1.0     // Usually 16.0
#define GEAR_REDUCTION 14.0       // Planetary Gearbox
#define MK8_DIAMETER 12.0         // The wheel that pushes the filament

// Don't edit these:
#define MK8_CIRCUMFERENCE (M_PI * MK8_DIAMETER)
#define NEMA17_MOTOR_STEPS (NEMA17_FULL_STEPS * NEMA17_MICROSTEPS)
#define E_STEPS (NEMA17_MOTOR_STEPS * GEAR_REDUCTION / MK8_CIRCUMFERENCE)

// Use for E steps-per-unit
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, E_STEPS }
```

kcheeeung, 2017-04-28:
Thanks so much everyone! I figured it out :)

closed 2017-04-28 by kcheeeung