MarlinFirmware/Marlin #6379: Bootloop when SD card in NOT plugged in.

ragemellons, 2017-04-18:
My printer seems infatuated with my SD card. If I turn it on with out the SD card, it goes into an infinite bootloop until I put the card back into the controller. It also reboots whenever I plug it into the computer and stalls when I disconnect it from said computer. I've tried uploading a new firmware. I've also sent an M500 command. I have been and still printing just fine, as long as the card is in, but it is a little annoying. It also doesn't say that the card has been inserted or taken out but I do have access to it when it's in. 

More recently, now if I take the card out while the system is still on, to put on another gcode, it doesn't acknowledge that the card has been put back in. When going into "print from sd" just leaves me with a main> option only. I then have to reboot to get into the SD card. 

My apologies if this isn't the correct place to ask this question, but I'm at a lost and have failed to come up with google searches related to my problem. I do however appreciate your time and any help.

thinkyhead, 2017-04-18:
I've seen the reboot loop with a Viki 2 display when no SD card was plugged in. I don't recall ever figuring it out. That particular display had one or two flaky buttons, so I think it could have been hardware-related in that particular case.

Are you able to test another controller or see what happens when the controller isn't plugged into the board?

ragemellons, 2017-04-18:
I don't. But I guess I can try unplugging the controller before removing the card then reverse operation with a new file added and see if it picks up the new file? Seems like a strange way of troubleshooting it. Will have to wait for the morning as its off printing ATM.

benlye, 2017-04-18:
We had the bootloop-without-sdcard problem on Anet boards, but it was fixed back in January in #5681.  Are you running RCBugFix?

dp90pusika, 2017-04-18:
I have a similar bootloop problen with 32-Bit-RCBugFix-new RADDS board and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
for me it is not related to the sd cart, it starts when uploading the firmware.
I know that it is a highly experimental version, but it's a similar thing

ragemellons, 2017-04-18:
@benlye I read that and couldn't come up with an answer. Long story short I lost track of which firmware I was using when trying to troubleshoot a board prior to this one. And since I couldn't remember which it was I just downloaded the most recent RCBugFix and problem has been solved. Thank you.

closed 2017-04-18 by ragemellons