MarlinFirmware/Marlin #6380: Can't compile RCBugFix with LCD enabled

benlye, 2017-04-18:
It seems like some recent changes, possibly introduced in 1286d9603dd12e06dccbcdef062e7b9f0b45a8e4 (Clean up and organize ultralcd.cpp) and/or f8e77d65454722b843e53c6ac71bb25806bd4361 (Add Linear Advance values to EEPROM and LCD) have broken things.

When I try to compile I get two sets of errors, the first related to lines starting `MENU_ITEM_EDIT_CALLBACK(long5, ...` in the M201 Settings section:
```
ketch\ultralcd.cpp: In function 'void lcd_control_motion_menu()':

ultralcd.cpp:248: error: invalid conversion from 'uint32_t* {aka long unsigned int*}' to 'long int*' [-fpermissive]

         lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PSTR(LABEL), ## __VA_ARGS__); \

                                                                                                                 ^

sketch\ultralcd.cpp:260:7: note: in expansion of macro '_MENU_ITEM_PART_1'

       _MENU_ITEM_PART_1(TYPE, LABEL, ## __VA_ARGS__); \

       ^

sketch\ultralcd.cpp:302:53: note: in expansion of macro 'MENU_ITEM'

   #define MENU_ITEM_EDIT_CALLBACK(type, label, ...) MENU_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)

                                                     ^

sketch\ultralcd.cpp:2511:5: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'

     MENU_ITEM_EDIT_CALLBACK(long5, MSG_AMAX MSG_X, &planner.max_acceleration_mm_per_s2[X_AXIS], 100, 99000, _reset_acceleration_rates);

     ^

In file included from sketch\ultralcd.cpp:65:0:

sketch\ultralcd_impl_DOGM.h:778:17: note: initializing argument 5 of 'void lcd_implementation_drawmenu_setting_edit_callback_long5(bool, uint8_t, const char*, const char*, long int*, ...)'

     inline void lcd_implementation_drawmenu_setting_edit_callback_ ## _name (const bool sel, const uint8_t row, const char* pstr, const char* pstr2, _type * const data, ...) { \

                 ^

sketch\ultralcd_impl_DOGM.h:794:3: note: in expansion of macro 'DEFINE_LCD_IMPLEMENTATION_DRAWMENU_SETTING_EDIT_TYPE'

   DEFINE_LCD_IMPLEMENTATION_DRAWMENU_SETTING_EDIT_TYPE(long, long5, ftostr5rj);

   ^
```
(This repeats once for each acceleration setting)

Second, there is an error for each time the DEFINE_MENU_EDIT_TYPE macro is called:
```
sketch\ultralcd.cpp: In function 'void _menu_action_setting_edit_accessor_int3(const char*, int (*)(), void (*)(int), int, int)':

ultralcd.cpp:3196: error: invalid conversion from 'void (*)(int)' to 'void*' [-fpermissive]

       editSetter = pset; \

                  ^

sketch\ultralcd.cpp:3207:3: note: in expansion of macro 'DEFINE_MENU_EDIT_TYPE'

   DEFINE_MENU_EDIT_TYPE(int, int3, itostr3, 1);

   ^
```

I've tried to put my finger on the root cause but so far haven't been able to.  I have REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled.

Tannoo, 2017-04-20:
I have the same display and have no issues with the current RCBugFix. Give it a shot.

benlye, 2017-04-20:
@Tannoo, I am already using the current RCBugFix.  Which compiler/IDE version are you using?  

At some point the Arduion IDE dropped the -fpermissive flag by default, so errors like this started to show up.  I'm using Arduino IDE 1.8.1, and if I add '-fpermissive' to my compiler options the error does go away, but I don't believe that is the right solution.

benlye, 2017-04-20:
FWIW,I did find where the invalid conversions were and fix them in PR #6385.

benlye, 2017-04-20:
OK, I take some of that back - it turns out I'm using a recent compiler, but the hardware configuration has an old platforms.txt file.

In fact, recently (mid 2016) the Arduino IDE *added* the -fpermissive to the compiler options by default [(link)](https://github.com/arduino/Arduino/commit/ba0c41b).  This was seen as both [good and bad](https://github.com/arduino/Arduino/commit/ba0c41b).

My specific problem is that for my board I'm using a hardware package written for an older IDE, with a platforms.local file which does not include the -fpermissive flag in the compiler options.  For sanity I'll update that so I'm using the same options as everyone else.

That said, it seems like having -fpermissive on by default is risky.  The conversion errors here are relatively benign, but the flag can mask more serious problems.

closed 2017-04-21 by benlye