MarlinFirmware/Marlin #6384: some questionable #defines

ruggb, 2017-04-18:
Given that I am not a coder who is intimate with C, I may be missing something, but the following appears problematic to me.

ONE:
There is the following defines:
DEFAULT_MACHINE_NAME
CUSTOM_MACHINE_NAME
MACHINE_NAME

and:
DEFAULT_MACHINE_UUID
CUSTOM_MACHINE_UUID
MACHINE_UUID

In both cases, there is 3 definitions that eventually boil down to just one?

If MACHINE_NAME & MACHINE_UUID were defined in configuration.h the user could change them there, if desired, and they would be a custom name without calling it CUSTOM. If not changed, it is the DEFAULT name. Then all the gymnastics of testing names in language.h to eventually set the MACHINE_NAME and MACHINE_UUID is unnecessary.

Some pins files set DEFAULT_MACHINE_NAME but are inconsistent. Pins_3DRAG.h has a #ifndef DEFAULT_MACHINE_NAME then defines it. Other pin files just define it. Then language.h boils both default and custom down to MACHINE_NAME. Are these defines in the pin files necessary? By the order of the conversions in language.h any DEFAULT name would be negated by a CUSTOM name anyway.

TWO:
In language.h there is an #ifdef DEFAULT_SOURCE_CODE_URL then undef SOURCE_CODE_URL and set it  = to the default. BUT I can find no #define DEFAULT_SOURCE_CODE_URL anyplace.
I think the same is true for DEFAULT_WEBSITE_URL & WEBSITE_URL

THREE:
What is the diff between SHORT_BUILD_VERSION & DETAILED_BUILD_VERSION
if they are set = to each other in version.h?

It would appear to me that all this is artifact of previous incomplete changes.

thinkyhead, 2017-04-26:
> It would appear to me…

It is all very conscientious and absolutely intentional, by design, and necessary.

- `DEFAULT_MACHINE_NAME` is provided by the pins files.
- `CUSTOM_MACHINE_NAME` is set in `Configuration.h` to override `DEFAULT_MACHINE_NAME`. This is the user-facing setting for this value.
- `MACHINE_NAME` may be set in `Version.h` (optional), as a means for vendors to override the name provided by the pins file. This is useful to have the name of the printer appear instead of the name of its board.

The logic that chooses the final `MACHINE_NAME` for use in the code is in `language.h`…

```cpp
#ifdef CUSTOM_MACHINE_NAME
  #undef  MACHINE_NAME
  #define MACHINE_NAME CUSTOM_MACHINE_NAME
#else
  #ifdef DEFAULT_MACHINE_NAME
    #undef  MACHINE_NAME
    #define MACHINE_NAME DEFAULT_MACHINE_NAME
  #endif
#endif
```

thinkyhead, 2017-04-26:
> hat is the diff between SHORT_BUILD_VERSION & DETAILED_BUILD_VERSION

The `SHORT_BUILD_VERSION` should be set to a shorter version of the `DETAILED_BUILD_VERSION`, but this is only needed if the version string is too long to fit well on the LCD display.

thinkyhead, 2017-04-26:
> BUT I can find no #define DEFAULT_SOURCE_CODE_URL anyplace.

Do a global search with a smarter text editor. It's defined in the pins files.

ruggb, 2017-04-26:
@thinkyhead Thanks
Things would be simpler if you had the user uncomment MACHINE_NAME and set that instead of CUSTOM_. Then you would only have to test for one item - if MACHINE_NAME is not defined then set default = machine. - I think the UUID would work the same.
It would be more user friendly if you had all user definable variables in the file(s) that you tell them they are in (config files) instead of other files like VERSION.
But then, maybe there is something about C and the compile that prevents KISS.

closed 2017-11-01 by thinkyhead