MarlinFirmware/Marlin #6386: PULL UP resistors stay active

willemnitert, 2017-04-18:
I have the following observation defining a Configuration.h file for my HY_BASE_L 1.0 board (RAMPS 1.4 compatible):
ENDSTOPPULLUPS is disabled, and ENDSTOPPULLUP_ZMIN is disabled in the Configuration.h file. And yet the Z_MIN_PIN (for my board this is Arduino pin 18) is still pulled high. Where is the pin defined for INPUT or INPUT_PULLUP? 

NOTE, that in the Arduino DIgital Pins explanation it says: 
"
Prior to Arduino 1.0.1, it was possible to configure the internal pull-ups in the following manner:

pinMode(pin, INPUT);           // set pin to input
digitalWrite(pin, HIGH);       // turn on pullup resistors
"

Prior to Arduino 1.0.1 means that with the present Arduino IDE only 'pinMode (pin, INPUT_PULLUP)' may be used? How are the pins initialised, INPUT or INPUT_PULLUP?

teemuatlut, 2017-04-18:
Marlin doesn't seem to use the Arduino framework "pinMode" at all, but directly manipulates the port registers.

```cpp
endstops.cpp: 85
    #if ENABLED(ENDSTOPPULLUP_ZMIN)
      SET_INPUT_PULLUP(Z_MIN_PIN);
    #else
      SET_INPUT(Z_MIN_PIN);
    #endif

fastio.h: 72
    #define _SET_INPUT(IO) do {DIO ##  IO ## _DDR &= ~MASK(DIO ## IO ## _PIN); } while (0)

    #define SET_INPUT(IO)  _SET_INPUT(IO)
    #define SET_INPUT_PULLUP(IO) do{ _SET_INPUT(IO); _WRITE(IO, HIGH); }while(0)
```

Here's what the datasheet (p68) says about setting the pin pullups

> If PORTxn is written logic one when the pin is configured as an input pin, the pull-up resistor is activated. To switch the pull-up resistor off, PORTxn has to be written logic zero or the pin has to be configured as an output pin.

So bug maybe? Add the write low on SET_INPUT(IO)?

Bob-the-Kuhn, 2017-04-18:
Those are the only two places that enable/disable the internal pullups for the endstops.

Long shot - make sure the pullup for the Z_PROBE is also disabled.  If Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is enabled Marlin wants the defines/settings for Z_MIN and Z_MIN_PROBE the same in this section.

Another possibility is there's a hardware pullup on your controller.  If you can point me at the schematic I can determine if that's the case.

----

Why do you want to disable the pullup? Except for the inductive/capacitive probes, all the endstop schemes I've seen either depend on the pullup or doesn't care if it's present or not.

willemnitert, 2017-04-19:
@Bob-the-Kuhn : Thank you for your comment. 
1. I have FIX_MOUNTED_PROBE and Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled.
2. Disable PULLUP is used for a PNP Normally Open Inductive sensor. It delivers (after a resistor  Voltage divider) 0V if LOW and +4.5V if HIGH.
3. Unfortunately till now I could not find a schematic for my controller HY-BASE-L V1.0 . Perhaps someone in the audience? But I think there are no external pull-ups.
 But one of my remarks was: has the method of initiating the internal PULL UP resistors in the Arduino IDE changed after version 1.0.1 of the IDE? And has Marlin adapted its code? The description I quoted was from the official Arduino.cc page.

willemnitert, 2017-04-19:
@teemuatlut 
Thank you for your comment. 
You exactly point at the comment I was making. I quoted directly from the official Arduino.cc page, where is stated (Arduino.cc, section Digital Pins) that this construction did work up till version 1.0.1 of the Arduino IDE. Or do I interpret this wrong? If the statement on the official Arduino.cc page is correct, at least it should be checked if the MARLIN firmware correctly uses the digital pins  wrt. the internal pullups. It would not be noted directly, since most people would use the hardware endstops for which the pullups are usually necessary. 
I do not know how you address the development team, so I used this channel.

willemnitert, 2017-04-19:
As a further test I have disabled the X and Y endstop pullups as well, and measured the voltages directly at the X and Y endstop terminals: both stayed at 5 volts and were not floating. So I must have missed something, or my HY BASE_L V1.0 board fools me. I have failed to find a schematic for this board, so its difficult to be sure.
I think I will write a small Arduino program which disables the pullups and run it on the board. See what happens.

Roxy-3D, 2017-04-19:
>  It would not be noted directly, since most people would use the hardware endstops for which the pullups are usually necessary.  I do not know how you address the development team, so I used this channel.

I have debug switches wired directly to the Atmega-2560 pins.   I know for a fact I have to turn on the Pull Up Resistors or I get false triggers.  And I know the opposite.  If I turn on the Pull Up Resistors....  Everything works as it should    It is my belief the pin_mode()  works as it is supposed to.

Bob-the-Kuhn, 2017-04-19:
Marlin uses the pinMode function.

In Arduino 1.18.1 the function has the following definition.  It's in wiring_digital.c
```
void pinMode(uint8_t pin, uint8_t mode)
{
	uint8_t bit = digitalPinToBitMask(pin);
	uint8_t port = digitalPinToPort(pin);
	volatile uint8_t *reg, *out;

	if (port == NOT_A_PIN) return;

	// JWS: can I let the optimizer do this?
	reg = portModeRegister(port);
	out = portOutputRegister(port);

	if (mode == INPUT) { 
		uint8_t oldSREG = SREG;
                cli();
		*reg &= ~bit;
		*out &= ~bit;
		SREG = oldSREG;
	} else if (mode == INPUT_PULLUP) {
		uint8_t oldSREG = SREG;
                cli();
		*reg &= ~bit;
		*out |= bit;
		SREG = oldSREG;
	} else {
		uint8_t oldSREG = SREG;
                cli();
		*reg |= bit;
		SREG = oldSREG;
	}
}
```

Basically the options are INPUT & INPUT_PULLUP.  If neither are specified then it's an output.

Marlin also uses the following macros.  These macros write to the registers directly and are much faster than pinMode.
```
SET_INPUT(pin)
SET_INPUT_PULLUP(pin)
SET_OUTPUT(pin)
```

----

Agreed - you can't have a pullup on Z_MIN if the sensor is to be read properly.

I suspect your controller has external pullups on the endstops.  To confirm you can tie a 1k resistor from the input to ground and then measure the voltage.  Rpullup will be:
Rup = ( 1k * 5V) / Vmeasured - 1k

The internal pullup is in the 20k - 50k range

You may want to move the Z_MIN_PIN to another pin & see if that cures the problem.

willemnitert, 2017-04-19:
@Bob-the-Kuhn  thank you for the comment. 
I will do the test and let you know the result asap, just to know what the situation on my board is. Of course I could also invert the probe's logic using a transistor for instance.

willemnitert, 2017-04-19:
@Bob-the-Kuhn 

Tested with a small program in the Arduino of the HY BASE_L V1.0 board. 1K resistor from input to ground.

pinMode(18,INPUT); Vmeasured is 0.9V, so I calculate Rup to be about 4500 Ohms.  
pinMode(18,INPUT_PULLUP); Vmeasured is 1.0V Not much of a difference....

If the internal pullups are 20K my conclusion is: the board has external pullups, or the board is defective. Other possibilities??

I have no schematic, and it is not very easy to spot the resistors on the board. So I think I will reverse the hardware logic, so I can stay with the pullups. Till now I have not spotted others with  experience with this specific HY BASE_L V1.0 motherboard.

Roxy-3D, 2017-04-19:
> If the internal pullups are 20K my conclusion is: the board has external pullups, or the board is defective. Other possibilities??

I have an Atmega-2560 board that I use to try different things on.  It doesn't have a RAMPS board plugged into it.  I just use it to load simple programs that are going to simple things.   The reason that board has such a basic life is I blew out several of the GPIO pins.   The board pretty much does work.  But some of the pins that go to the Y-Axis step stick on a RAMPS board are dead.

It is possible to blow out a few pins on these AVR processors and still have the chip mostly functional.  You may want to try your end stop logic on a different set of pins and see if the behavior changes.

Bob-the-Kuhn, 2017-04-19:
There's a 4.7Kpullup on that line.  

I suggest you edit the pins_RAMPS.h file and change the Z_MIN_PIN to an unused pin.

Bob-the-Kuhn, 2017-04-19:
There's a utility in the latest RCBugFix that will toggle the pin.  That might help you identify the resistor if you're going to remove it.  

To use it enable PINS_DEBUGGING in Configuration_adv.h and then issue the following command:
M43 T S18 L18 I R50

 ```
 *  M43 T       - Toggle pin(s) and report which pin is being toggled
   *                  S<pin>  - Start Pin number.   If not given, will default to 0
   *                  L<pin>  - End Pin number.   If not given, will default to last pin defined for this board
   *                  I       - Flag to ignore Marlin's pin protection.   Use with caution!!!!
   *                  R       - Repeat pulses on each pin this number of times before continueing to next pin
   *                  W       - Wait time (in miliseconds) between pulses.  If not given will default to 500
```

willemnitert, 2017-04-20:
@Bob-the-Kuhn, @teemuatlut  and @Roxy-3D : thank you for your contributions so far. I will continue to quest for the schematics of the HY BASE_L V1.0 board, but if you know where I can find it, please let me know. The coming day I have a full schedule, so I will not be able to work on it.

willemnitert, 2017-04-22:
@Bob-the-Kuhn 
Have searched, but cannot find the board's schematic. So, I think I'll go for a transistor solution to reverse the inductive sensor's logic.

Bob-the-Kuhn, 2017-04-22:
Either that or use an unassigned pin for your input.  Much simpler.  Just go into the pins_YOUR_BOARD.h file and change Z_MIN_PIN to the other pin.

willemnitert, 2017-04-22:
Using an unassigned pin is more complicated than you think, because my board is not listed in the boards.h file, and there is not a pins_ file directly. I use the RAMPS 1.3 file, but I am not sure it matches completely.

Bob-the-Kuhn, 2017-04-22:
Here's a write up on the utilities that I use when I need to identify a pin.

```
SETUP

1. Enable PINS_DEBUGGING in Configuration_adv.h
2. Recompile and upload to your controller

FINDING THE LOGICAL PIN NUMBER AND SIGNAL NAME OF AN UNKNOWN PIN

Issue the command M43 W1 I1.  It sets all pins to inputs, turns on the internal pull up resistors and then reports the pin number & name if it sees any changes.

I take a 1k resistor, use a jumper to tie one end to ground and then use a jumper to connect the other to the unknown pins and watch the host display to see what pops up.  This'll identify about 80% of the pins.

DISPLAY LOGICAL PINS, THE SIGNAL NAME9S) ATTACHED TO THEM AND THEIR STATUS

Issue the M43 I1 command.

If a pin is listed as unused/unknown and it's listed as an output then it's definitely used but we don't know the name of the signal.
```

----

Here's the info we have on the M43 command & it's options.

```
  /**
   * M43: Pin debug - report pin state, watch pins, toggle pins and servo probe test/report
   *
   *  M43         - report name and state of pin(s)
   *                  P<pin>  Pin to read or watch. If omitted, reads all pins.
   *                  I       Flag to ignore Marlin's pin protection.
   *
   *  M43 W       - Watch pins -reporting changes- until reset, click, or M108.
   *                  P<pin>  Pin to read or watch. If omitted, read/watch all pins.
   *                  I       Flag to ignore Marlin's pin protection.
   *
   *  M43 E<bool> - Enable / disable background endstop monitoring
   *                  - Machine continues to operate
   *                  - Reports changes to endstops
   *                  - Toggles LED when an endstop changes
   *                  - Can not reliably catch the 5mS pulse from BLTouch type probes
   *
   *  M43 T       - Toggle pin(s) and report which pin is being toggled
   *                  S<pin>  - Start Pin number.   If not given, will default to 0
   *                  L<pin>  - End Pin number.   If not given, will default to last pin defined for this board
   *                  I       - Flag to ignore Marlin's pin protection.   Use with caution!!!!
   *                  R       - Repeat pulses on each pin this number of times before continueing to next pin
   *                  W       - Wait time (in miliseconds) between pulses.  If not given will default to 500
   *
   *  M43 S       - Servo probe test
   *                  P<index> - Probe index (optional - defaults to 0
   */
'''

willemnitert, 2017-04-22:
Have installed the transistor switch. Works fine. Now I go and calibrate the inductive sensor height.

Roxy-3D, 2017-04-22:
> Here's a write up on the utilities that I use when I need to identify a pin.

I go the opposite route.   I take an LED and ground its negative side.  And I put the positive side on the free pin that I don't know the Pin Number.    And I guess the command has morphed into M43 T R 10 W 100    Assuming I see it blink, I then start narrowing the scope of possible pin numbers with S and L.

Bob-the-Kuhn, 2017-04-22:
Got it working - GREAT.

willemnitert, 2017-04-23:
@Bob-the-Kuhn @teemuatlut @Roxy-3D Thank you all for your reactions. 
Conclusion of this all: the HY BASE_L V1.0 has some external pullups of 4.7K. These are undocumented, and therefore they can get you into problems, when making changes to the original 3d printer setup (chinese Infitary I3 in my case). Since it seems not to be possible to get the board's schematics,  this is something you should take into account when buying such a chinese 3d printer. However, it was fun putting it together, it was very complete and the quality was better than expected.

Again, thank you all.

willemnitert, 2017-04-24:
This is the schematic I used to connect the Inductive sensor LJ12A3-4-Z/BY:

[Zendstop.pdf](https://github.com/MarlinFirmware/Marlin/files/950854/Zendstop.pdf)

Roxy-3D, 2017-04-24:
@willemnitert   For $20 you can swap out your electronics with an Atmega-2560 and RAMPS board...  Probably 1/2 of the RepRap printers are using that setup now.   You won't have any support problems if you go that route.

willemnitert, 2017-04-25:
@Roxy-3D IThat's a good idea for the future. My present board has an Atmega-2560, but a RAMPS board has far more users and hence support. The schematics are available, so you can 'see what you do'. I have asked the chinese shop to send me the schematics of the HY-BASRE_L V1.0 board, but they sent me the installation manual again. I suppose they aquired the board somewhere else, and they do'nt have the schematics either. The board is OK and looks professional, but no schematics, thats a big drawback. I add a picture of my board, perhaps someone has the schematics.

[HY_BASE_L V1.0.pdf](https://github.com/MarlinFirmware/Marlin/files/954089/HY_BASE_L.V1.0.pdf)

lrpirlet, 2017-04-25:
@willemnitert 
My 2 cents here...
This site https://www.banggood.com/All-in-one-3D-Printer-Motherboard-MKS-Base_L-Compatible-Marlin-For-Beginners-p-1119110.html says BASE_L is just an MKS Base with some change...
Quote 
> The difference with MKS Base:
> 
> 1, USB drive using 16u2 instead of FT232
> 2, The power module using 2596, also supports 24V
> 3, The motor drive using HR4982
EndQuote

The Schematics for MKS Base seems to be available if I believe the pictures seen under this link: https://www.google.be/search?q=mks+base+v1.5+schematic&newwindow=1&client=ubuntu&hs=WpJ&channel=fs&biw=1366&bih=645&tbm=isch&tbo=u&source=univ&sa=X&sqi=2&ved=0ahUKEwjktsO_o7_TAhXD1BoKHfhdARsQ7AkIMQ

No, sorry I did not go further cause if your picture seems to match the first link, there are several revisions to the MKS Base and got no time to go further...

Anyway, enjoy :-)

closed 2017-11-01 by thinkyhead