MarlinFirmware/Marlin #6389: Reliable non-stepper output

dronus, 2017-04-19:
Is there any easy way to modify Marlin to get a reliable low-level axis position output via SPI or Serial interfaces?

Would be useful to add an external non-stepper motor control like servo-controlled DC or brushless motors. 

Receiving high frequency step/dir signals again for many axes simultanously is not that easy for an external controller...

bobc, 2017-04-19:
Do you have a specific hardware in mind, or is this one of those "wouldn't
it be nice if Marlin did everything on any platform" kind of things?

I suspect that controlling drivers via SPI/serial at the required rate
would be more than an AVR could handle, but without knowing the number of
axes, the bit rate, the protocol and the update frequency it's impossible
to be sure. Engineering can only work with numbers.

thinkyhead, 2017-04-26:
I've implemented a two-board printer that uses i2c to communicate from the main board (a Megatronics 3) to the secondary board (a RAMPS 1.4). The second board manages 4 Z axis steppers and 4 endstops so that it can home, move Z, and auto-level as commanded by the main board.

The main board's instance of Marlin merely sends very short string commands to the secondary board, and can query the second board for its "state" — which is simple: "B" for busy, "L" for leveled successfully, "H" for homed successfully, "F" for failed, etc.

It works well, in fact.

bobc, 2017-04-26:
Sure, you can do operations like bed levelling which are not time critical,
but what about controlling steppers during printing?

thinkyhead, 2017-04-26:
> but what about controlling steppers during printing?

For running a Z axis the serial speed is more than adequate.

bobc, 2017-04-26:
Yes, but while extruding?

bobc, 2017-04-26:
It would be great if this feature was in Marlin, I designed a 6 axis
extruder expansion board, but never got round to using it. I would like to
do a version with an I2C IO expander.

thinkyhead, 2017-04-26:
> Yes, but while extruding?

What can I say? It works well.

> It would be great if this feature was in Marlin

If you'd like to mess with it, here are the branches:
- https://github.com/thinkyhead/Marlin/tree/a3d_main_controller
- https://github.com/thinkyhead/Marlin/tree/a3d_zaxis_controller

dronus, 2017-11-13:
Well, I like to use Marlin as universal CNC controller as it provides quite well working lookahead. But for many uses, servo motors would be a better choice. So just outputting speed or position information via SPI would allow external controllers to be connected, without havin the Arduino handle servo control as well.

thinkyhead, 2018-02-09:
The demand for this kind of feature is extremely low and devs have a lot on their plates dealing with bugs and 32-bit migration so it's unlikely this will be implemented anytime soon. But, you may be able to hire a developer for $$$ if you need this feature very much.

boelle, 2019-02-18:
@dronus 
Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/ 
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

dronus, 2019-02-20:
While I used a question mark, this still is a feature request I thought...

boelle, 2019-03-12:
@thinkyhead this one should have the feature request label added and the question label removed

boelle, 2019-06-22:
wrong label

InsanityAutomation, 2019-06-22:
I would think this may be more feasible with drivers like the TMC5160 and 4361 more available now. @teemuatlut  may be the expert on the capability to do this with TMC drivers and if a 2.1 milestone is feasible.