MarlinFirmware/Marlin #6390: Unretract Feedrate changes in the menu are not respected

FHeilmann, 2017-04-19:
I am using FWRETRACT with my printer and, no matter what I set in the Menu during a print, the recover feedrate doesn't change.

The confusing part is that the retract feedrate works perfectly fine and changes in the menu are also respected by the printer. It is merely the recover feedrate that doesn't update.

I had a look at the code but I couldn't find a place where ``retract_recover_feedrate_mm_s`` is treated differently than ``retract_feedrate_mm_s`` so I am at a loss.

thinkyhead, 2017-04-26:
After editing on the LCD use `M503 S0` to view the current `M208` settings. The `M208 F` value should match your edited setting. (If you want to preserve your edit, don't forget to send `M500`.)

The code is using the recover feed-rate:
```cpp
void retract(const bool retracting, const bool swapping = false) {
  . . .
  const float old_feedrate_mm_s = feedrate_mm_s;
  . . .
  if (retracting) {
    feedrate_mm_s = retract_feedrate_mm_s;
    . . .
  }
  else {
    . . .
    feedrate_mm_s = retract_recover_feedrate_mm_s;
    . . .
  }
  . . .
  feedrate_mm_s = old_feedrate_mm_s;
  . . .
}
```

FHeilmann, 2017-04-26:
Funnily enough the Terminal does indeed show the correct value, however the recover feedrate is still not correct :/ I shot a video of the issue, 

you can watch it here:

https://www.youtube.com/watch?v=9aZN5tVKpjc

Here is the G-Code File that I used:

[top_holes.txt](https://github.com/MarlinFirmware/Marlin/files/958196/top_holes.txt)

FHeilmann, 2017-05-02:
Problem still persists with a freshly synced RCBugFix as of today. Also, setting the recover feedrate from Terminal using `M208 F600` updates the value shown in the menu, but the recovery still happens using the higher feedrate.

thinkyhead, 2017-05-03:
Maybe stick some `SERIAL_ECHO()` lines into the `retract()` function printing the values of its operating variables and see what's happening…?

FHeilmann, 2017-05-03:
I did as you instructed (see the rc_debug_recovery_mms branch in my repo [direct commit link here](https://github.com/FHeilmann/Marlin/commit/df38abd1c8846bf3a0b18c9138336fc53a436917))

According to the debug code, the retraction code behaves as expected, **however, my extruder motor recovery speed still doesn't change**. Maybe someone else who uses FWRetract can test if the recovery feedrate can be changed from the value in the configuration_adv.h ? Maybe it is just an issue with my board/fork (even though I haven't made any changes other than configuration)

Here is my terminal output with the changes from the above branch:

```
Send: M503 S0
Recv:   G21 ; Units in mm
Recv: 
Recv: 
Recv:   M200 D3.00
Recv:   M200 D0
Recv:   M92 X80.00 Y80.00 Z400.00 E608.00
Recv:   M203 X300.00 Y300.00 Z30.00 E70.00
Recv:   M201 X2500 Y2500 Z100 E10000
Recv:   M204 P3000.00 R2000.00 T2000.00
Recv:   M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
Recv:   M206 X0.00 Y0.00 Z0.00
Recv:   M420 S0 Z0.00
Recv:   M145 S0 H180 B70 F0
Recv:   M145 S1 H240 B110 F0
Recv:   M301 P23.02 I1.95 D67.89
Recv:   M304 P174.47 I26.49 D287.28
Recv:   M207 S5.00 F4200.00 Z0.20
Recv:   M208 S-0.10 F600.00
Recv:   M209 S0
Recv:   M851 Z0.40
[...]
Send: G10
Recv: >>> retract() feedrate:25.00 mm/s
Recv: >>> if(retracting) feedrate:25.00 mm/s
Recv: <<< if(retracting) feedrate:70.00 mm/s
Recv: <<< retract() feedrate:25.00 mm/s
[...]
Send: G11
Recv: >>> retract() feedrate:25.00 mm/s
Recv: >>> if(!retracting) feedrate:25.00 mm/s
Recv: >>> if(!retracting) pre_change feedrate:25.00 mm/s
Recv: >>> if(!retracting) post_change feedrate:10.00 mm/s
Recv: <<< if(!retracting feedrate:10.00 mm/s
Recv: <<< retract() feedrate:25.00 mm/s
```

FHeilmann, 2017-05-03:
I did some more digging:
On my machine, the `RETRACT_RECOVER_FEEDRATE` configuration parameter is also not respected. No matter what I set it to, the recovery always happens at the same speed fast. I played around with some values and found the following:

The recovery always uses the maximum extruder speed Vmax(E).

I hope this makes tracking down this issue easier.

FHeilmann, 2017-05-04:
@thinkyhead I managed to fix it. Please see Pull Request #6570 for my proposed change.

I don't know if this only masks an underlying bug, but it fixes the problem described in this issue, as the recovery rate configured via LCD or Terminal is now respected.

FHeilmann, 2017-05-04:
fixed by above mentioned PR

closed 2017-05-04 by FHeilmann