MarlinFirmware/Marlin #6391: Homing causes endstop crash when attempting to home with T1 as active tool

PeterGrace, 2017-04-19:
I have a reproducible problem where attempting to home the gantry with T1 as the active extruder causes grinding and thrashing.  If I switch to extruder 0 prior to homing, the gantry homes as expected.

I'm working off of source from 89731cac17545ebf3146573e44bf689f7d8e97ef.

Here's a gcode snippet that causes the head crash:

```
T0
M218 T1 X-18
G0 X100Y100 F6000
T1
G28 X Y
```

Here's a diff of my config vs. the above git rev:

```
#define STRING_CONFIG_H_AUTHOR "(Pete)" // Who made the changes.
#define MOTHERBOARD BOARD_RAMPS_14_EEB
#define CUSTOM_MACHINE_NAME "Corey"
#define MACHINE_UUID "deefc78c-3072-4d45-9c28-75ceee4e2b8c"
#define EXTRUDERS 2
#define TEMP_SENSOR_1 1
#define TEMP_SENSOR_BED 1
#define PID_AUTOTUNE_MENU // Add PID Autotune to the LCD "Temperature" menu to run M303 and apply the result.
#define PID_PARAMS_PER_HOTEND // Uses separate PID parameters for each extruder (useful for mismatched extruders)
#define  DEFAULT_Kp 21.37
#define  DEFAULT_Ki 1.47
#define  DEFAULT_Kd 77.73
#define BED_LIMIT_SWITCHING
#define EXTRUDE_MINTEMP 160
#define EXTRUDE_MAXLENGTH 1001
#define COREXY
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 642, 408 , 408}
#define DEFAULT_MAX_FEEDRATE          { 70, 70, 5, 25 }
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
#define HOMING_FEEDRATE_XY (25*60)
#define ULTRA_LCD   // Character based
#define ENCODER_PULSES_PER_STEP 1
#define ENCODER_STEPS_PER_MENU_ITEM 5
#define INDIVIDUAL_AXIS_HOMING_MENU
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define QUICK_HOME  //if this is defined, if both x and y are to be homed, a diagonal move will be performed initially.
#define HOME_Y_BEFORE_X
```

Bob-the-Kuhn, 2017-04-20:
What flavor of Marlin are you running?

thinkyhead, 2017-04-26:
1.0.2-2 probably has bugs in that area. I suggest upgrading to 1.1.0 (which will be released on Sunday).

closed 2017-11-01 by thinkyhead