MarlinFirmware/Marlin #6393: Print Aborted message stuck

landodragon141, 2017-04-19:
If I start a print from the SD (Graphical Display) and then stop the print. The message print aborted gets stuck on the front page. I can start a new print, preheat the hotend. The only way to get it to reset is to power off and back on.

Tannoo, 2017-04-20:
Which LCD to you have enabled in `configuration.h`?

landodragon141, 2017-04-20:
I'm on the latest as of a few days ago. This one: #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Tannoo, 2017-04-20:
Hmm.

psavva, 2017-04-20:
In order to set another message on the LCD, you would use the Gcode M117.

Unless you instruct the firmware to display  a different message, it will not update that kind of the display.

I suggest adding an M117 with a message to your start gcode.

Example:
M117 Printing...

landodragon141, 2017-04-20:
No it should change to show the Hot end heating message when I start a new print. I'm guessing there is something in the way that message is coded that doesn't allow it to be superseded.

psavva, 2017-04-20:
@Tannoo can you check this in G M104?

If M104 S parameter > Actual Hotend temperature, then display the heating message.
However, if it's all ready at the target temperature then the abort message would stay on screen.. 
Perhaps an else statement is needed for a heating completed message?
Marlin_Main function gcode_M104
if (code_value_temp_abs() > thermalManager.degHotend(target_extruder)) LCD_MESSAGEPGM(MSG_HEATING);

Tannoo, 2017-04-20:
Yes, I see that. `E1 Heating...` was displayed until I sent `M104 S0`.

psavva, 2017-04-20:
I think we have some room for improvement.
We should either show the welcome message, or heating message, but should not keep the last message displayed which was "Aborted" in this case.

Tannoo, 2017-04-20:
How about adding `lcd_setstatus(MSG_PRINTING);` in this function?
```
  /**
   * M24: Start or Resume SD Print
   */
  inline void gcode_M24() {
    #if ENABLED(PARK_HEAD_ON_PAUSE)
      move_back_on_resume();
    #endif

    card.startFileprint();
    print_job_timer.start();
    lcd_setstatus(MSG_PRINTING);
  }
```

trivalik, 2017-04-23:
BTW In the start code is in CURA:

M117 Printing...

but the message is not changing.

landodragon141, 2017-04-26:
Huzzah!

commit adb8e685 (2017-04-27): Fix stuck "Print Aborted" LCD message

Fixes #6393
closed 2017-04-26 by thinkyhead