MarlinFirmware/Marlin #6396: Error Compiling Using any Character Based LCD

Simulatedbog545, 2017-04-20:
So, I've been trying to upgrade the marlin firmware on my printrobt simple metal (I want the proper mesh leveling, not the 3 point version), and as long as I do not enable any character based LCD's, everything works just fine. But the second I try to enable them though, I cannot compile the firmware and get the error "invalid conversion from 'uint16_t* {aka unsigned int*}' to 'int16_t* {aka int*}' [-fpermissive]". It stems from the line "void dac_driver_commit() { dac_current_set_percents(driverPercent); }" in the ultralcd.cpp file. I've figured out some errors before, but I just can't figure out what's wrong here. Could someone give me a hand?

Tannoo, 2017-04-20:
RCBugFix compiles just fine for me with `REPRAP_DISCOUNT_SMART_CONTROLLER` enabled.

Is your LCD different?

Bob-the-Kuhn, 2017-04-20:
I just tried ULTIPANEL with no issues.

Which flavor of Marlin are you trying to upgrade to?  I hope it's RCBugFix.

If you ZIP your config files and drop it on your reply I'll try to reproduce the problem.

Simulatedbog545, 2017-04-20:
Hm.. How can I check the version of Marlin that I have? I have attatched a link to the zip below:
https://www.dropbox.com/s/v1xugh99ga200fi/Printrbot%20Marlin%20Newer.zip?dl=0

Simulatedbog545, 2017-04-20:
Yep, i'm officially an idot. I've been using the RC branch, not RCbugfix. So pro.... :/

Tannoo, 2017-04-20:
I wouldn't say that. RC is the default branch when to go to the Main Marlin git site.
You have to change to the RCBugFix branch to get that one.

Simulatedbog545, 2017-04-20:
Well, i moved I moved over to the RCBugFix version, but the instant I even try to do something it just dies. The only thing i've changed is the main board, and as a result, I just get the error 
'customizedSerial' was not declared in this scope. I just can't see it. I only change the board configuration, and its corresponding setting in the arduino IDE. Grrrrrrrrrr

Tannoo, 2017-04-20:
Please attatch your config files here.

benlye, 2017-04-20:
#6380 and #6385 could be related.

Roxy-3D, 2017-04-20:
> Well, i moved I moved over to the RCBugFix version, but the instant I even try to do something it just dies. 

You do need to pull your Configuration.h and Configuration_adv.h file into a visual diff program and map your settings to the new file.  You can not just drop your old files in place and expect things to work.

Simulatedbog545, 2017-04-20:
Benlye Thanks for that, i'm reading it right now. Seems like it may fix my problem.
Roxy-3D This is what i did unfortunately, I started completely from scratch, transferring over my settings by hand, I did not copy the configuration files over.

Simulatedbog545, 2017-04-20:
Here is my current current file. I figured out the 1st issue were it would not compile with the printrboard. I am now stuck again with the original issue: Keep in mind, I've only been able to reproduce this with the printrboard revision F selected, which requires teensyduino to be installed; Just something to notice if you try this out yourself.
Firmware Folder: https://www.dropbox.com/s/k78tn6c9l4rs9r9/Marlin-RCBugFix.zip?dl=0

benlye, 2017-04-20:
So you're still getting 'invalid conversion' errors?  That could be caused by your build environment missing the -fpermissive compiler switch I mentioned in #6380.  Want to share the full error?

Simulatedbog545, 2017-04-20:
benlye: Yeah, here is the full error:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Thomas\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Thomas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensypp2:usb=serial,speed=16,keys=en-us -ide-version=10802 -build-path C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372 -warnings=none -build-cache C:\Users\Thomas\AppData\Local\Temp\arduino_cache_388859 -verbose C:\Users\Thomas\Desktop\Latest Marlin\Marlin-RCBugFix\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Thomas\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Thomas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensypp2:usb=serial,speed=16,keys=en-us -ide-version=10802 -build-path C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372 -warnings=none -build-cache C:\Users\Thomas\AppData\Local\Temp\arduino_cache_388859 -verbose C:\Users\Thomas\Desktop\Latest Marlin\Marlin-RCBugFix\Marlin\Marlin.ino
Using board 'teensypp2' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Using core 'teensy' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/avr/bin/avr-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -fno-exceptions -felide-constructors -std=c++0x -mmcu=at90usb1286 -DTEENSYDUINO=136 -DARDUINO_ARCH_AVR -DARDUINO=10802 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy" "C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\Marlin.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/avr/bin/avr-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -fno-exceptions -felide-constructors -std=c++0x -mmcu=at90usb1286 -DTEENSYDUINO=136 -DARDUINO_ARCH_AVR -DARDUINO=10802 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy" "-IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src" "C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\Marlin.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\G26_Mesh_Validation_Tool.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\M100_Free_Mem_Chk.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\MarlinSerial.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\Marlin_main.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\Sd2Card.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\SdBaseFile.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\SdFatUtil.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\SdFile.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\SdVolume.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\blinkm.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\cardreader.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\configuration_store.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\dac_mcp4728.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\digipot_mcp4018.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\digipot_mcp4451.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\endstops.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\hex_print_routines.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\least_squares_fit.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\mesh_bed_leveling.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\nozzle.cpp
Using cached library dependencies for file: C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\planner.cpp
[... cut ...]
C:\Users\Thomas\AppData\Local\Temp\arduino_build_101372\sketch\ultralcd.cpp:3214:3: note: in expansion of macro 'DEFINE_MENU_EDIT_TYPE'

   DEFINE_MENU_EDIT_TYPE(unsigned long, long5, ftostr5rj, 0.01);

   ^

Using library LiquidCrystal at version 1.0.5 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 
invalid conversion from 'uint32_t* {aka long unsigned int*}' to 'long int*' [-fpermissive]

benlye, 2017-04-20:
Yep, seems like the same issue I ran into - you need to compile with the -fpermissive flag and then those errors become warnings, and you won't see them any more.  At some point that flag was added by default in the Arduino IDE, so most people aren't seeing these errors.  But, people like you and me, with older hardware support files that don't include the flag are seeing these errors.

I think you need to figure out how to give the compiler the -fpermissive flag when compiling for Teensy.  I just installed the latest TeensyDuino package and it's not obvious to me - the platform.txt looks totally different to the one I have for my hardware.  Sorry!

Simulatedbog545, 2017-04-20:
Benlye
No problem mate, I realize I've got a pretty unique setup here. I'm trying to find it myself, I'll come back and tell you if I find it
:)

benlye, 2017-04-20:
Looking at these lines in the teensy platform.txt:
```
## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{build.toolchain}{build.command.g++}" -c {build.flags.optimize} {build.flags.common} {build.flags.dep} {build.flags.cpp} {build.flags.cpu} {build.flags.defs} -DARDUINO={runtime.ide.version} -DF_CPU={build.fcpu} -D{build.usbtype} -DLAYOUT_{build.keylayout} {includes} "{source_file}" -o "{object_file}"
```
It *might* be as simple as changing it to this:
```## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{build.toolchain}{build.command.g++}" -c {build.flags.optimize} {build.flags.common} {build.flags.dep} {build.flags.cpp} -fpermissive {build.flags.cpu} {build.flags.defs} -DARDUINO={runtime.ide.version} -DF_CPU={build.fcpu} -D{build.usbtype} -DLAYOUT_{build.keylayout} {includes} "{source_file}" -o "{object_file}"
```

I added `-fpermissive` after `{build.flags.cpp}`.

Good luck!

Simulatedbog545, 2017-04-20:
Benlye
Hey, I did a very similar thing, and it seems to work now. I added the "-fpermissive" tag in lines 
Compile c++ files
Compile c files
Compile S files
The location didn't seem to matter, and now the firmware is properly compiling. I will finish putting in my settings and the try and upload it to my printer. Thanks!

Simulatedbog545, 2017-04-20:
Hehey! Everything compiles! Now I just have to trim down the file size, as at the moment, it's way too big to fit on my printrboard. RIP

Simulatedbog545, 2017-04-21:
You guys are the best. Finished trimming file down, uploaded to my simple metal, and printing a benchy right now. Thanks a lot! :+1:

closed 2017-04-21 by Simulatedbog545