MarlinFirmware/Marlin #6398: HOMING Z BEFORE X and Y ?

Benditbeuh, 2017-04-20:
Hello,

Would it be possible to add a parameter allowing HOME_Z BEFORE X and Y on G28?

Thank you for your reply,
Ben

Tannoo, 2017-04-20:
Normally, that's not what's wanted due to the Z might crash into parts if G28 was issued after a print.

Benditbeuh, 2017-04-20:
Thanks Tannoo !

Indeed, there may be many cases where this setting would cause problems.

In some other specific cases, and if this potential parameter is only possible in advanced mode, could we consider adding it to MARLIN?

In practical terms, it would be very useful for me. Indeed, I installed a BLTOUCH on a GREG WADES Prusa i3 to the right of the extruder. The BLTOUCH does the HOMING Z. But, since it is placed outside the heated bed, it does not properly work.

Tannoo, 2017-04-20:
Then wouldn't it be prudent to hook up a z_min limit and swap the following settings?
```
//#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
```

Benditbeuh, 2017-04-20:
This is actually what was set up in MARLIN for my machine.

    +--  BACK ---+
    |                       |
  L |           (+)      | 
  E |                      | 
  F |    **P** (-) N (+) | G <-- nozzle (10,10)
  T |                      | 
    |            (-)       | 
    |                       |
    O-- FRONT --+
  (0,0)

**P** is the BLTOUCH position.

As the X and Y are checked before the Z. The BLTOUCH is outside the heating bed.

Simulatedbog545, 2017-04-20:
I mean, this is a bit of a wacky solution, but would it work to home X to max instead of min? Doing that should leave the BL-touch in the print area, and then it would work as a z-endstop.

Benditbeuh, 2017-04-20:
It is an idea !
But in that case, should I move the ENDSTOP X on the other side?

Tannoo, 2017-04-20:
Yes, then change `X_HOME_DIR` to 1.
```
// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1
```
Then you may have to swap `X_MIN_POS` with `X_MAX_POS`.
```
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 185
#define Z_MAX_POS 190
```

Benditbeuh, 2017-04-20:
Tonight, I roll up my sleeves And I put my laziness in the closet !
Tkanks very much !

Bob-the-Kuhn, 2017-04-20:
Is the real issue that the Z homing is trying to probe off the bed?

If yes then enable Z_SAFE_HOMING & set the coordinates as desired.  The default is the middle of the bed.
```
#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 2)    // Y point for Z homing when homing all axis (G28).
#endif

```

----

You shouldn't swap `X_MIN_POS `with `X_MAX_POS `when homing in the + direction. The X location will be set to `X_MAX_POS ` when homing in the + direction.  

The only downside to homing in the +X direction is now you need to be VERY precise with the  `X_MAX_POS ` value or your 0,0 location will be off.

JustAnother1, 2017-04-21:
You can also add a parameter to G28:

G28 Z1 for example will only home the Z axis.
G28 X1 Y1 will only home X and Y

Benditbeuh, 2017-04-21:
If I understand correctly, your solution would be to HOME at the center of bed ?

Tannoo, 2017-04-21:
Center or at the position you choose:
```
// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0
```

Tannoo, 2017-04-21:
Nevermind...that's not it.

Bob-the-Kuhn, 2017-04-21:
Yes, enable Z_SAFE_HOMING & set the coordinates as desired.  Doesn't have to be at the center of the bed.

Benditbeuh, 2017-04-21:
That's it ! That's the solution!
`#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_MIN_POS + X_MAX_POS) / 8)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_MIN_POS + Y_MAX_POS) / 8)    // Y point for Z homing when homing all axis (G28).
#endif
`

Thank you so much All of you,

closed 2017-04-24 by Benditbeuh