MarlinFirmware/Marlin #6399: auto leveling issue?

d2crafty, 2017-04-20:
I already set up the autoleveling and had successful prints.. however now after it levels and starts printing.. one side of my brim is showing up as "too squished down" and the other side as though it is "normal" -- seems as though the sensor is not working as intended?  is there something I can do other than adjusting the bed screws to accommodate the leveling?

Tannoo, 2017-04-20:
Dual Z steppers? Possible one got out of sync at some point causing the X carriage to be out of kilter?
Is it possible that one or more of the bed adjustments worked loose(r)?

You are not "autoleveling" before every print, are you?
If so, you are not taking advantage of the EEPROM features to store the matrix for multiple uses.

benlye, 2017-04-20:
Check that your Z-probe X and Y offsets are correct - you need to set the offset from the center of the nozzle to the center of the probe (what kind of probe, out of interest) - having this out by 1-2mm can cause the behaviour you are seeing (I had the same problem).

You might find the troubleshooting steps here helpful (it's written for the A8, which is another i3 clone):
https://3dprint.wiki/reprap/anet/a8/improvement/autobedleveling#troubleshooting

Bob-the-Kuhn, 2017-04-20:
If you continue to have problems with the two Z steppers getting out of synch then consider going to Z_DUAL_ENDSTOPS.  

If you're using Bilinear leveling then take a look at this.
[Bilinear Auto Bed Leveling Details 0.4.zip](https://github.com/MarlinFirmware/Marlin/files/944479/Bilinear.Auto.Bed.Leveling.Details.0.4.zip)

sarf2k4, 2017-04-21:
I am also wondering about the configuration since I am using original bltouch classic and here are my configs for it

these are the offset especially x and y, I got them from measuring the model file in the 3d modeling software

```
#define X_PROBE_OFFSET_FROM_EXTRUDER 30  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 11 // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.95   // Z offset: -below +above  [the nozzle]
```

while this one is the probe boundary
``` 
// Set the boundaries for probing (where the probe can reach).
    #define LEFT_PROBE_BED_POSITION ((X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
    #define RIGHT_PROBE_BED_POSITION 195
    #define FRONT_PROBE_BED_POSITION 11
    #define BACK_PROBE_BED_POSITION 192
```

d2crafty, 2017-04-22:
sorry I havn't been on here a bit.. #define X_PROBE_OFFSET_FROM_EXTRUDER 20  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -15  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.70 // Z offset: -below +above  [the nozzle]

// X and Y axis travel speed (mm/m) between probes
#define XY_PROBE_SPEED 3000
// Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
// Speed for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
// Use double touch for probing
#define PROBE_DOUBLE_TOUCH

-- 

I caliper checked my x axis for height discrepancies and now I'm manually rechecking my bed for it being set off.

d2crafty, 2017-04-22:
using http://www.ebay.com/itm/New-3d-printer-Auto-Leveling-Sensor-limit-switch-for-Prusa-I3-3D-printer-/302251659470?hash=item465f9a50ce:g:z18AAOSw4A5Yx2bf --- as a bed leveling sensor

d2crafty, 2017-04-22:
well.. i spent 6 hours trying to get this damn printer to print correctly again.  I still have to "manually" adjust the bed to the print head on one of the points for the print to stick to the bed correctly.  I almost feel like I'm better off going back to the manual end stops :(

Roxy-3D, 2017-04-22:
> well.. i spent 6 hours trying to get this damn printer to print correctly again. I still have to "manually" adjust the bed to the print head on one of the points for the print to stick to the bed correctly. I almost feel like I'm better off going back to the manual end stops :(

Turn on the UBL Bed Leveling...   You will get good adhesion across 100% of your bed every time.  If you do choose to do that...   I'll help you get it set up.    Do these things and report back...
- MAKE SURE your Z_PROBE_OFFSET_FROM_EXTRUDER is set very accurately in Configuration.h
- Turn on BABYSTEPPING and DOUBLE_CLICK in Configuration.h
- Turn on UBL and tell it to do a 10 x 10 Mesh
- Do a M502, M500, M501
- Do a G28
- Do a G29 P1
- Do a G29 S 1
- Do a G29 O  and post the topology map here saying you are ready to continue.  (at this point, you are at the fun part.)

d2crafty, 2017-04-22:
i'll get back tonight.  I've been up from 11pm to now 645am.. thanks for the help!

sarf2k4, 2017-04-22:
@Roxy-3D , i need a bit of help here as well for my configuration above. Was it okay especially for the probe area bounding?

Roxy-3D, 2017-04-22:
> Was it okay especially for the probe area bounding?

I don't understand the question.    But if you have a bizarre setup, we can address it with the Configuration_adv.h settings.  (bizarre like your X-Min is -50 and X-Max is 200).     UBL knows about the probe offset and that is the reason I asked for the topology map as the last step up above.  Depending on what we see...  We may want to do address the un-probed areas differently.

Go ahead and see far you can get with the list up above...

sarf2k4, 2017-04-22:
well, according to bob, give a bit of edge margin so the probed point value are much better

Roxy-3D, 2017-04-22:
> well, according to bob, give a bit of edge margin so the probed point value are much better

There is truth in that....    But really, it is up to you.  Do you want every last mili-meter of printable bed space?   Because if you do, it will be harder (meaning an extra iteration of G26 & G29 P4) to clean up the inaccurate numbers along the edge.  If you want to get setup with minimal problems...  Yeah...  define a 5mm or 10mm inset in your configuration.h file.   And that issue will be automatically addressed.

sarf2k4, 2017-04-22:
okay, i will try with 5-10mm inset, that would mean at about the following right?

```
// Set the boundaries for probing (where the probe can reach).
    #define LEFT_PROBE_BED_POSITION 25
    #define RIGHT_PROBE_BED_POSITION 190
    #define FRONT_PROBE_BED_POSITION 20
    #define BACK_PROBE_BED_POSITION 185
```

the above setting are after the margin inset

Roxy-3D, 2017-04-22:
Those numbers are not used in UBL.    Please ignore them.    Just set UBL_MESH_INSET 10  and we will know as soon as we do a G26 whether you want to shift numbers before you fine tune your mesh.

sarf2k4, 2017-04-22:
I'm unable to find the specified variable, the closest is this `  #define MIN_PROBE_EDGE 10` on line 817

Roxy-3D, 2017-04-22:
You should be using the current RCBugFix, right???   This is the section to fill in your UBL numbers:
```cpp

#elif ENABLED(AUTO_BED_LEVELING_UBL)

  //===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  #define UBL_MESH_INSET 45         // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
  #define UBL_PROBE_PT_1_X 39       // These set the probe locations for when UBL does a 3-Point leveling
  #define UBL_PROBE_PT_1_Y 325      // of the mesh.
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 325
  #define UBL_PROBE_PT_3_Y 200
  #define UBL_G26_MESH_EDITING      // Enable G26 mesh editing
```

sarf2k4, 2017-04-22:
aahhhh.... i see, I am using the initial rc8 release since rcbugfix kind of always updated and i might lose track of which is which to be honest. i will give it a try.

so if I used the previous setting that i currently use, not the one with margin applied, the ubl inset will apply the margin automatically right?

Roxy-3D, 2017-04-22:
Yes.   BUT  please pull up the configuration.h files in a visual diff program and move your settings to the new file.  Do not just try to drop your old file in place.   That will not work.    And probably, you should do that for both Configuration.h and Configuration_adv.h.

sarf2k4, 2017-04-22:
is it possible if i add `#include <customconfig.h>` at the top of the original configuration.h file in hoping to override the original file? sorry a bit of out of topic

Roxy-3D, 2017-04-22:
I don't know...    It is safest if you don't add any variables.   It would be best to just get your settings into Configuration.h

sarf2k4, 2017-04-22:
I'm porting the previous config.h to the rcbugfix's config and reached to the bed leveling section. which do i have to pick?

Roxy-3D, 2017-04-22:
```cpp
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

sarf2k4, 2017-04-22:
okay. I will try with the bilinear first then moving to ubl if problem still persist

d2crafty, 2017-04-22:
tried configuring the new marlin.  seems my printer doesn't like the new ver.  no UBL leveling in my ver. :/

Bob-the-Kuhn, 2017-04-22:
It's only in [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

You'll have to manually transfer your machine specific items over to the new config files.

jasoncross, 2017-04-23:
I have a very similar issue.  Migrated my config and am now on the latest RCBugFix.  Would like to try using UBL, but haven't figured it out yet - any guide out there for setup and testing?

Roxy-3D, 2017-04-23:
> Would like to try using UBL, but haven't figured it out yet - any guide out there for setup and testing?

We need a guide written for it.   But do these steps and report back...  I'll help you.

- MAKE SURE your Z_PROBE_OFFSET_FROM_EXTRUDER is set very accurately in Configuration.h
- Turn on BABYSTEPPING and DOUBLE_CLICK in Configuration_adv.h
- Turn on UBL and tell it to do a 10 x 10 Mesh (Flash the firmware)
- Do a M502, M500, M501
- Do a G28
- Do a G29 P 1
- Do a G29 S 1
- Do a G29 O and post the topology map here saying you are ready to continue. (at this point, you are at the fun part.)

jasoncross, 2017-04-23:
Ok!  Will give it a go in the morning and report back!

fiveangle, 2017-04-23:
> Turn on BABYSTEPPING

So UBL requires LCD ?

Roxy-3D, 2017-04-23:
> So UBL requires LCD ?

Right now... Yes.   But once we get Delta's up and running that constraint can be relaxed.

jasoncross, 2017-04-23:
Do you recommend turning on BABYSTEP_ZPROBE_OFFSET?

Map:

Recv: Bed Topography Report:
Recv: 
Recv: (0,9)                                                                                                    (9,9)
Recv: (20,20)                                                                                                (280,280)
Recv:   0.625    0.715    0.555    0.408    0.750    0.963    0.912    1.112    1.100      .
Recv: 
Recv:   0.345    0.283    0.393    0.628    0.787    0.765    0.807    1.082    1.098      .
Recv: 
Recv:   0.130    0.010    0.423    0.413    0.420    0.918    0.795    0.767    0.985      .
Recv: 
Recv:  -0.048    0.120    0.082    0.462    0.582    0.355    0.435    0.680    0.925      .
Recv: 
Recv:  -0.450   -0.245   -0.077    0.160    0.125    0.357    0.505    0.538    0.818      .
Recv: 
Recv:  -0.620   -0.460   -0.102   -0.187    0.158  [ 0.010]   0.320    0.490    0.548      .
Recv: 
Recv:  -0.780   -0.442   -0.072   -0.293   -0.067   -0.050    0.120    0.340    0.477      .
Recv: 
Recv:  -0.970   -0.638   -0.650   -0.330   -0.232   -0.092    0.187    0.192    0.452      .
Recv: 
Recv:  -0.965   -0.950   -0.795   -0.405   -0.288   -0.170    0.025    0.255    0.305      .
Recv: 
Recv:     .        .        .        .        .        .        .        .        .        .
Recv: (20,20)                                                                                                    (280,20)
Recv: (0,0)                                                                                                       (9,0)

Roxy-3D, 2017-04-23:
@jasoncross  I had to pull your Topology Map into a text editor so I could get the columns to line up.   Your probe has a negative X & Y OFFSET_FROM_EXTRUDER.  Which means it can't probe all the way to the right side or the front (if your origin is left front) of the bed.    

We need to fill in the unprobed areas with 'reasonable' numbers.   Please do the following:

- Reset the printer
- G28 
- G29 L 1     (this loads your mesh)
- G1 X25 Y30
- G29 P3 C -.90 R3 O    (You will see the lower right 3 unprobed numbers filled in)
- G29 P3 C -.3 R3 O  (and the next three numbers)
- G29 P3 C 0.0 R1 O
- G29 P3 C 0.3 R4 O
- G1 X250 Y30
- G29 P3 C .5 R3 
- G29 P3 C .85 R2
- G29 P3 C 1.0 R O   (No need to specify a number... We are filling the rest with 1.0)
- G29 A
- G29 S1
- reset the printer   (not really needed...  but lets start clean)
- G26 P C O3.0    (Put hair spray (or prep) the entire bed.  We are going to see where the mesh needs to be adjusted with G29 P4 R)

jasoncross, 2017-04-23:
I have those commands done.  What's next?  Nothing happened when I ran the G26 P C O3.0

Roxy-3D, 2017-04-23:
> I have those commands done. What's next? Nothing happened when I ran the G26 P C O3.0

That doesn't sound right...  Do you have   #define UBL_G26_MESH_EDITING   enabled?

jasoncross, 2017-04-24:
> That doesn't sound right... Do you have #define UBL_G26_MESH_EDITING enabled? If not, you need to enable it and re-flash the firmware. Then continue on...

That was commented out.  I uncommented, reflashed, then ran  G28 followed by G26 P C O3.0

A pattern printed.  Most (but not all) of the grid looked to be extruded from further away from the heat bed than I would expect.  Should I adjust my offset with M851 and run again?  Any way to test a smaller portion of the grid to see if that helps without waiting for the full 10x10?

Roxy-3D, 2017-04-24:
> A pattern printed. Most (but not all) of the grid looked to be extruded from further away from the heat bed than I would expect. Should I adjust my offset with M851 and run again? Any way to test a smaller portion of the grid to see if that helps without waiting for the full 10x10?

Step one was:

- MAKE SURE your Z_PROBE_OFFSET_FROM_EXTRUDER is set very accurately in Configuration.h

In theory, it doesn't matter because the mesh can adjust for what ever error is present.  But things do go much better if this is accurately set.    And it helps to set the Z-Homing to happen in the center of the bed.   If you want...   It might be good to make sure the Z Probe offset is very accurately set. (along with homing in the center of the bed)

But...   The way the system works is you can now move the nozzle over an area of the bed that is not 'perfect' and issue a G29 P4 R.    You can dial up or down each mesh point.    As a rule of thumb...  If the line is too squished down and too thing...  Bring the mesh point up.  Don't go crazy... but depending on what you see, bring it up .1 or .2 mm.     If the line is just barely adhering to the bed, bring the mesh point down .1mm.  If the line is not adhearing at all, bring it down .25mm.  

And then run another G26 command.   Things will be better.  You will see improvement.  But there still will be flaws.   So, you move the nozzle to each bad area, and do another G29 P4 R.

After a few iterations...   You will have 100% adhesion across 100% of your bed 100% of the time.

But seriously...   It really does help to have your Z-Probe-Offset dialed in very accurately.

jasoncross, 2017-04-24:
Here is how I tested my offset:

G28
Put a piece of paper under the nozzle on the bed
Set Z to 0
Paper had that tiny bit of pull between build surface and nozzle that one likes

Not sure how to get more exact than that.  Plus it's the same offset I've been using with bilinear with good results at least in the middle of my bed. 

Will tweak the grid tomorrow and report back.

Roxy-3D, 2017-04-24:
Yeah...  It is always difficult to get that number right.    But here is another way to do it.    You have a G26 pattern printed on your bed.   Is it pretty much too high?   Or is it pretty much too low?    Is half of it perfect?  And half of it not adhering at all?

The Z_PROBE_OFFSET_FROM_EXTRUDER should be  ---PERFECT--- in the area where the nozzle is when the Z-Axis is homed.   (And homing in the center of the bed helps for a number of reasons!!!!).   It is worth taking a quick detour and getting that number set right such that  the mesh has very small correction numbers around the place where the nozzle homes the Z-Axis.

But that is the whole purpose of the G26 Mesh Validation Pattern.  It lets you see what is happening with regard to your nozzle and how it prints on the bed.    Dial in your Z_PROBE_OFFSET_FROM_EXTRUDER number based on where the nozzle is when your home the Z-Axis.    Make sure that part of the G26 pattern looks nice with no correction.

jasoncross, 2017-04-24:
What's the best way to stop the G26 pattern?  Anything short of the kill switch?

I used m851 to move the z offset .1 and re-ran G26.  The center looks correct, the right has the nozzle too close, as does the top center, and the bottom left is too far.

Can you clarify how I go about making corrections at each point?

Don't worry, it doesnt feel hopeless.  I'm excited to get it all figured out.  i do have a question though - once the mesh is established, does the whole process need to be retuned from scratch every time the bed gets slightly off level left-to right or my glass shifts slightly?

Roxy-3D, 2017-04-24:
> What's the best way to stop the G26 pattern? Anything short of the kill switch?

Press and hold the Encoder Wheel.  There is some latency.  It is only checked once during each circle (and connecting line) cycle.   It might take 2 or 3 seconds to be noticed.   Just keep the encoder wheel pressed...  and it will stop 'eventually'.

> I used m851 to move the z offset .1 and re-ran G26. The center looks correct, the right has the nozzle too close, as does the top center, and the bottom left is too far.

Ok! good on the 'center' part!    Probably, you should start clean with a new G29 P1.   But if you choose not to...  You can just edit each and every mesh point.   But if you adjust the Z probe offset...  You really should start clean with a fresh G29 P1.

> Can you clarify how I go about making corrections at each point?

After you do a G26 command, the mesh validation pattern will be on the bed.   Different areas may be less than 'perfect'.   Move the nozzle to the center of the area that is flawed.   Do a G29 P4 R command.   The nozzle will go to the closest mesh point that has not been edited yet.   You can turn the encoder wheel to adjust what should happen at that mesh point.  If you have bad adheasion at the mesh point, bring the nozzle down a little bit.  If the filament is squished too thin at the mesh point, bring it up 'the right' amount.

When you done editing that area...  Press and Hold the encoder wheel.   This is typically done over a mesh point you don't want to edit.

> Don't worry, it doesnt feel hopeless. I'm excited to get it all figured out. i do have a question though - once the mesh is established, does the whole process need to be retuned from scratch every time the bed gets slightly off level left-to right or my glass shifts slightly?

No...   If an area or edge starts not being perfect...  You can just edit that one area and everything is good for a while.   Typically...  A perfectly tuned mesh lasts at least a month (and I print a lot).  But as the linear bearings wear and different things happen...  Your mesh does need to be 'refreshed'.   But it usually is small areas that are less than perfect and it takes longer to do the G26 command than to actually do the editing.     What can really screw things up is if you drive the nozzle into the bed and one Z stepper motor keeps going down but the other side freezes.   Doing something like that causes you to lose everything and may have to start over.

d2crafty, 2017-04-24:
ok, heres the issue.  
[Marlin_REVISED_NEW.zip](https://github.com/MarlinFirmware/Marlin/files/953363/Marlin_REVISED_NEW.zip)
[marlin_OLD.zip](https://github.com/MarlinFirmware/Marlin/files/953367/marlin_OLD.zip)

The OLD marlin works.  The NEW marlin refuses to work.   I could swear I'm putting all the right things in, maybe I missed something.  Can somebody review it and tell me please?  thank you.
[image]

Roxy-3D, 2017-04-24:
@d2crafty   A visual diff program is very useful.   Try bringing up Notepad++ and turning on the visual comparison plug in.  Then bring in both files and make sure your settings are all crossed over.   This really is something you need to learn how to do.

d2crafty, 2017-04-24:
I know there are differences, however I don't know what is needed and unneeded.  E.g. in the new version a LED color system is enabled which I disabled. Maybe something I have needs to be enabled that was added to t-3 new system I don't know of that is crashing the system?

Roxy-3D, 2017-04-25:
> E.g. in the new version a(n) LED color system is enabled which I disabled. 

It is not enabled by default, right?   No way can that be the way it is when this gets promoted to Golden.

d2crafty, 2017-04-25:
Well, I decided to uninstall repetier and reinstall to try for the prints since I have been trying to level this bed over and over with no avail.  Funny, first print that comes out is 95% perfect.  The offset is about .02 too close to the bed.  But everything seems to be coming put straight.  All this and maybe it was just a software issue on repetier alone?

Roxy-3D, 2017-04-25:
> first print that comes out is 95% perfect. T

The first print with UBL turned on (and the mesh not even edited....)?   Or do you mean with no bed leveling turned on...  You can now get a 95% perfect print?

d2crafty, 2017-04-25:
ok, I fixed it, its coming out 99.9% there.  In my book, its "perfect" enough compared to what I was getting 17 hours ago.  The bed leveling is turned on, I have no clue why nothing was registering.  But whatever.  I'm still curious about the UBL, but right now, I just need to get some prints out.  Roxy-3D commented 12 minutes ago "You can now get a 95% perfect print?"  -- give me crap

Roxy-3D, 2017-04-25:
Yeah...   This stuff is more complicated than it needs to be!

d2crafty, 2017-04-25:
well, I spent alittle over a week on this, 7-8hours a day, not including my weekends which went even longer.  like today is technically my Sunday.  To say it was something simple as, uninstall and reinstall sort of pisses me off

Roxy-3D, 2017-04-25:
> To say it was something simple as, uninstall and reinstall sort of pisses me off

This isn't quite what you are talking about...   But Slic3r and PronterFace and others use Java.   And I'm not 100% that Java subsystem has the user's best interest in mind.   There is all kinds of flakiness and just installing or uninstalling one of those Java apps affects all of the other Java apps you have installed.

This stuff all needs to work better with other app's in the 3D-Printer ecosystem.

d2crafty, 2017-04-25:
just curious, I get the msg OpenGL: Out of Memory.  I have 3gb ddr5 memory on a 16GB of ram.  I don't see how this is possible?

jasoncross, 2017-04-25:
> You can turn the encoder wheel to adjust what should happen at that mesh point. If you have bad adheasion at the mesh point, bring the nozzle down a little bit. If the filament is squished too thin at the mesh point, bring it up 'the right' amount.

If the nozzle is too close to the bed at that point in the grid, do I raise or lower the number?

Roxy-3D, 2017-04-25:
The mesh numbers say what to do to the nozzle to make it track the bed.  So if the nozzle is too close to the bed...  What ever the number is currently, make it bigger.  Make it more positive.   Make it less negative.

If the nozzle is too far from the bed...  bring the mesh point number down.  make the mesh point number smaller.  make it more negative.  

If you have any adhesion at all...  don't move the number by more than .15 mm     .15mm will cause a visible difference in how the lines at that point appear.

jasoncross, 2017-04-25:
Once I make all of the adjustments, do I do a G29 S1 to save to EEPROM?  And then another G26?

Roxy-3D, 2017-04-25:
Yes to both questions...    You should see improvement in the mesh validation pattern.  But some areas will still be wrong.   They will be 'less wrong'.   The process converges very quickly.

jasoncross, 2017-04-25:
What is a G29 D vs a G29 P0?  And do either override the fine tuning?

Roxy-3D, 2017-04-25:
Yes, G29 P1 invalidates the mesh and start the auto probing.

All of the bed leveling systems are mutually exclusive.   But if your mesh is bad, You can do a G29 P0 to set the entire mesh to 0.000     And you can do a 3-Point or grid based leveling of your bed.   That is G29 T or G29 J3    You won't have the corrections for a non-flat surface.  But you will have the system correcting for any tilt in the bed.

jasoncross, 2017-04-25:
Ok, that's good to know.

I assume after a G29 P1 I'd also need to refill in where the probe couldn't reach.  Any progress on your SmartFill update?  How will that be invoked (or will it be automatic)?

Ok, so let's say I get my mesh all fine tuned.  After each printer reset do I run a G29 L 1 to reload it, or will it reload automatically out of EEPROM?  And when I go to print, do I just do a G28 (right now with the old systems I have gcode that runs G28 then G29)?  What happens if a G29 command alone gets sent?

jasoncross, 2017-04-25:
Oh wait, I see, G29 P0 sets to mesh to 0, it doesn't just disable.

Does a G29 T or G29 J3 with the mesh enabled compensate for overall tilt along the mesh?  So if for example I have a good mesh on my build surface but the overall tilt gets off, could one of those correct for it?

[154 more replies]

closed 2017-11-03 by thinkyhead