MarlinFirmware/Marlin #6402: Heater suddenly stop heating

Jacques-D, 2017-04-20:
Hello,

When I command the extruder to warm up to 250°C (for testing purposes), the extruder warms up quickly then suddenly and approximately stop to a random temperature (usually between 170°C and 200°C, but sometimes higher). Why "approximately stop" ? Because tension at the board output is not zero at that moment, but around 3V. So the extruder actually cool down slowly because not completely powered off.
Then, because the targetted temperature is not reached, I get the error:heating failed (which is normal).

Here is the log, for example :

22:25:02.921 : N62 M104 S250*86
22:25:02.921 : ok
22:25:04.247 : ok T:65.5 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:07.315 : ok T:66.2 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:10.372 : ok T:70.6 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:13.430 : ok T:77.1 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:16.492 : ok T:84.1 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:19.560 : ok T:91.7 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:22.628 : ok T:98.7 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:25.691 : ok T:106.2 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:28.754 : ok T:113.1 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:31.811 : ok T:120.1 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:34.869 : ok T:127.2 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:37.931 : ok T:133.6 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:40.989 : ok T:140.1 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:44.054 : ok T:146.6 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:47.117 : ok T:153.0 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:50.176 : ok T:158.7 /250.0 B:20.7 /0.0 @:127 B@:0
22:25:53.239 : ok T:164.9 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:56.301 : ok T:170.5 /250.0 B:20.8 /0.0 @:127 B@:0
22:25:59.364 : ok T:176.3 /250.0 B:20.7 /0.0 @:127 B@:0
22:26:02.422 : ok T:180.9 /250.0 B:20.8 /0.0 @:127 B@:0
22:26:05.479 : ok T:183.2 /250.0 B:20.8 /0.0 @:127 B@:0
22:26:08.542 : ok T:183.6 /250.0 B:21.0 /0.0 @:127 B@:0
22:26:11.594 : ok T:183.2 /250.0 B:20.8 /0.0 @:127 B@:0
22:26:14.651 : ok T:182.3 /250.0 B:20.7 /0.0 @:127 B@:0
22:26:17.715 : ok T:181.4 /250.0 B:20.8 /0.0 @:127 B@:0
22:26:20.783 : ok T:180.3 /250.0 B:20.8 /0.0 @:127 B@:0
22:26:23.846 : ok T:179.4 /250.0 B:20.7 /0.0 @:127 B@:0
22:26:26.913 : ok T:178.5 /250.0 B:20.6 /0.0 @:127 B@:0
22:26:29.971 : ok T:177.7 /250.0 B:20.7 /0.0 @:127 B@:0
22:26:33.028 : ok T:176.7 /250.0 B:20.6 /0.0 @:127 B@:0
22:26:33.262 : Error:Heating failed, system stopped! Heater_ID: 0
22:26:33.262 : Error:Printer halted. kill() called!

The failure happened around 22:26:03. As you can see, before the time of the heater stop, the temperature rose quite quickly (5-6°C every 3s).
Additionaly, if I reset quickly the board after the heater stopped, and give quickly a 250°C command, the heater will be able to go higher than the temperature reached before. So it is likely not a faulty heater.

Note that the board do not actually crash when the heater stops heating : I first encountered the problem during a M303 PID tuning. During this test, the error:heating failed **did not** work (maybe something that should be checked/corrected in a future RC?) and I left the temperature cool down slowly for ten minutes (hoping that, for some reason, it would start rising again). Before resetting the board, I tried to play with the LCD display, and move the axes and it worked so Marlin do not completely freeze.

What could be the reason ? 
I have a peculiar setup : I run a rumba board with a 24V power supply. But my heater cartridge and fans (e3D v6) are all 12V components (reusing components). On rumba, it is possible to power the two fans with 12V (even if rumba is powered with 24V) by moving two jumpers. It is also possible to power the heater output HE2 to 12V with the same method. To get the HE2 output working instead of the HE0, I switched the pins from HE0 and HE2 in the pins_RUMBA.h :

// Heaters / Fans
//
//#define HEATER_0_PIN        2   // default value
#define HEATER_0_PIN        6
#define HEATER_1_PIN        3
//#define HEATER_2_PIN        6   // default value
#define HEATER_2_PIN        2
#define HEATER_3_PIN        8
#define HEATER_BED_PIN      9

So I think this trick may be the reason of the bug. But before investing in 24V components, I wanted to expose my problem to see if there is a firmware solution ?

Thank you,
Jacques

Tannoo, 2017-04-21:
Not a faulty heater, but a faulty thermistor?

Jacques-D, 2017-04-21:
Thank you for your reply.
I have e3d V6 heater, and I set the thermistor type that is provided by e3d (#define TEMP_SENSOR_0 5).

Faulty thermistor ? It is possible, I have a spare one so I'll check tonight.
However, if it was a faulty thermistor, the printer would crash with a MIN_TEMP fatal error because signal would drop to zero, no ? Additionaly, the reading in the log seems okay (although it only provides a temperature reading every 3s or so).
I will have a laser thermometer tonight and I'll check that the actual heater temperature is similar to the one provided by the log.

Regarding my rumba trick, there is something else going on : when the heater stops heating (output voltage drops to 3V instead of 12V), the fan0 stops working as well. I'll have to check tonight if voltage also drops to 3V, or if it drops to zero.

Thank you for your support, I appreciate !

Jacques-D, 2017-04-21:
So I just measured the resistance of my thermistors. The one mounted on the e3d v6 had a resistance around 125 kohms, while the spare has a resistance around 108 kohms. They were both sitting in the same room, around 21°C.

Could this explain the problem ?

Thank you,
Jacques

Tannoo, 2017-04-21:
Possibly.

Jacques-D, 2017-04-24:
Hello,

I went back to a 12V power supply for the rumba board, and plugged the resistor to the HE0 pin. Everything is working well now.

So I'm guessing that the board cannot transform 24V to 12V and handle the power required by the resistor. It probably works well for fans, but not for a 30W resistor. 

Thank you for your help,
Jacques

closed 2017-04-24 by Jacques-D