MarlinFirmware/Marlin #6408: X-axis motor disconnects during print

Tailhappy, 2017-04-21:
So the symptom of the problem is that during a print (using Marlin 1.10-RC8), the x-axis motor disconnects inexplicably resulting  in a circular part becoming a vertical line (see attached pic)[image]

My setup is unusual in that I've set extruder E1 to be the **primary** extruder using the "T1" command in my startup gcode. This extruder is positioned at X=-33.0mm from E0. I've used either the slicer or the firmware to specify this X-offset dimension. In either case the result is identical.  Somehow while in the middle of printing (actually about 33% complete), Marlin decides to cutout the X-axis motor and the only way to reconnect is to disconnect and reconnect Octoprint (v1.3.2).

The sliced gcode commands are normal and continue to specify typical X and Y coordinates during the loss of the X-axis. This change is not commanded by gcode, rather it's the Arduino processor (Mega2560) directing the motor driver (on the Rambo controller board) to disconnect. Unfortunately I have no ability to view this motor driver communication link.

emartinez167, 2017-04-21:
Have you noticed if it always happens at the same height? If so, could be a loose cable on the stepper that only manifests itself when the cables bend beyond a certain point.

Tailhappy, 2017-04-21:
While yes, it definitely happens at the same layer height each time, it can't be a hardware (cable) issue  because I reset the connection by disconnecting and then reconnecting Octoprint. Could Octoprint be involved? Yes, but how do I diagnose this? I need some visibility of the Ardiuno comm link like a log file or something, but I'm unfamiliar with Arduino code executing besides the Marlin FW compiling. Again, the serial link shows no oddities in the Gcode stream.

Another quicker way to reproduce this bug is to initialize, and then immediately stop the print. The X-axis motor will disconnect even prior to the first layer being printed.

Bob-the-Kuhn, 2017-04-21:
Could the X stepper or it's driver be over heating?

Tailhappy, 2017-04-22:
The last sentence in my second post above provides the evidence that this isn't a thermal (or hardware) related issue. And the motor is well matched to the driver and cold. It's a bug. I suspected the cause was the negative offset value for the second extruder so I swapped E0 and E1, made the value positive 33.0mm and all is well. Workaround complete.

Bob-the-Kuhn, 2017-04-22:
Thanks for the feedback.  

I've added this to my Marlin Things To Do list.

thinkyhead, 2017-04-26:
> My setup is unusual in that I've set extruder E1 to be the primary extruder using the "T1" command in my startup gcode. This extruder is positioned at X=-33.0mm from E0. I've used either the slicer or the firmware to specify this X-offset dimension.

Please post your `Configuration.h` and `Configuration_adv.h` files to help with testing. (You may need to add `.txt` extension for Github to accept the files being dropped into the textfield.)

closed 2017-11-01 by thinkyhead