MarlinFirmware/Marlin #6411: Marlin Fan Percentage is not showing... Its showing "---" Pls help

ShrikantPai123456, 2017-04-21:
I am using clone of sanguino.. Fan is showing only "---".. I wanted in percentage %... That is 0-255 is 0%-100%.....

Pls tell what shall I add in Marlin V1.0 or uncomment?????
[image]

Tannoo, 2017-04-21:
Is that with the fan off?

ShrikantPai123456, 2017-04-21:
No sir.. thks for replying. When fan is ON, its showing "---" and also when its off, the same message is displaying.

I went to control menu and kept 255 for FAN.. then also same message "---"

Help...

ShrikantPai123456, 2017-04-21:
I kept 50% which nearly 128 value....but same message "---"

What am I missing line in marlin???

Tannoo, 2017-04-21:
Extruder fans?

Tannoo, 2017-04-21:
In `configuration_adv.h`, there are these settings:

I am running RCBugFix. You can upgrade to that and see if that issue is fixed for you.
I only have one extruder, but not controlling extruder fans.
```
/**
 * Extruder cooling fans
 *
 * Extruder auto fans automatically turn on when their extruders'
 * temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
 *
 * Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 *
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
#define E0_AUTO_FAN_PIN -1
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed
```

Tannoo, 2017-04-21:
That says that you'll have to check your board's pin file for fan - pin assignments.

ShrikantPai123456, 2017-04-21:
This is my code

#define EXTRUDER_0_AUTO_FAN_PIN   -1
#define EXTRUDER_1_AUTO_FAN_PIN   -1
#define EXTRUDER_2_AUTO_FAN_PIN   -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

Fan is rotating fine.... Nozzle as well as for mother board sanguino....But only displaying as "---" on LCD....

The FAN symbol working on LCD and FAN is also rotating on sanguino... Only FAN percentage is not displaying sir....

what shall I do????

Tannoo, 2017-04-21:
Time for RCBugFix uprade. That's what this branch is for. Fixing bugs in the RC branch along with new features.

Tannoo, 2017-04-21:
Or at least up to RC8.

ShrikantPai123456, 2017-04-21:
Yeah ok... I will upgrade.

But pls tell me what code shall I add in Marlin???

or comment or uncomment sir???

ShrikantPai123456, 2017-04-21:
Pls tell the coding part in Marlin V1.0...

It important so sir....

Tannoo, 2017-04-21:
Well, you can download RC8 and compare the difference. Make appropriate changes.

It will be easier to upgrade than to try to fix what may already have been fixed.

ShrikantPai123456, 2017-04-21:
Pls see my code    dogm_implementation.h

// Heatbed
 u8g.setFont(FONT_STATUSMENU);
 u8g.setPrintPos(81,6);
 u8g.print(itostr3(int(degTargetBed() + 0.5)));
 lcd_printPGM(PSTR(LCD_STR_DEGREE " "));
 u8g.setPrintPos(81,27);
 u8g.print(itostr3(int(degBed() + 0.5)));
 lcd_printPGM(PSTR(LCD_STR_DEGREE " "));
 if (!isHeatingBed()) u8g.drawBox(88,18,2,2);
	else
		{
		 u8g.setColorIndex(0);	// white on black
		 u8g.drawBox(88,18,2,2);
		 u8g.setColorIndex(1);	// black on white
		}
 
 // Fan
 u8g.setFont(FONT_STATUSMENU);
 u8g.setPrintPos(104,27);
 #if defined(FAN_PIN) && FAN_PIN > -1
 u8g.print(itostr3(int((fanSpeed*100)/255 + 1)));
 u8g.print("%");
 #else
 u8g.print("---");
 #endif

but code is taking only  **u8g.print("---");**

ShrikantPai123456, 2017-04-21:
help somebody.... ASAP

my nozzle and controller FANS are working.... also FAN rotation symbol on LCD is displaying... But only the FAN percentage is not displaying....

Help....

Tannoo, 2017-04-21:
```
#if defined(FAN_PIN) && FAN_PIN > -1
u8g.print(itostr3(int((fanSpeed*100)/255 + 1)));
u8g.print("%");
#else
u8g.print("---");
#endif
```

ShrikantPai123456, 2017-04-21:
@tannoo 

whats that sir???

Tannoo, 2017-04-21:
What you posted..

```
#if defined(FAN_PIN) && FAN_PIN > -1             <--- FAN_PIN AND FAN_PIN >-1
u8g.print(itostr3(int((fanSpeed*100)/255 + 1)));
u8g.print("%");
#else
u8g.print("---");                                                    <--- If not FAN_PIN AND FAN_PIN <-1
#endif
```

That means that your FAN_PIN has to be defined AND FAN_PIN has to be greater than -1.

So... try making that an OR by changing 
`#if defined(FAN_PIN) && FAN_PIN > -1`
to
`#if defined(FAN_PIN) || FAN_PIN > -1`
and see if that works.

stigjoergensen, 2017-04-21:
Hi was trying to tell you, that you allready have written what you needed to know ;)

if you look at it:

#if defined(FAN_PIN) && FAN_PIN > -1
u8g.print(itostr3(int((fanSpeed*100)/255 + 1)));
u8g.print("%");
#else
u8g.print("---");
#endif

This means:
if FAN_PIN is defined somewere and the value is bigger than -1 then
   print fanspeed * 100 / 255 (this formula converts the 0-255 value to 0-100
else
  print ---

that means, if you see --- then either the FAN_PIN isnt defined somewhere or the value is -1 - normally you would look for this in 3 places - config.h config_adv.h or your board.h files

Tannoo, 2017-04-21:
>somew`h`ere and the value is bigger than -1

Search for it. Likely in your pins file.

ShrikantPai123456, 2017-04-21:
yeah I will check with "OR"ing and reply to you sir...

@Tannoo
@ stigjoergensen 

Thnks I will be back in minute....

ShrikantPai123456, 2017-04-21:
this is pins.h

#define LED_PIN            -1

#define FAN_PIN            -1
 #if FAN_PIN == 12 || FAN_PIN ==13
  #define FAN_SOFT_PWM
#endif

#ifdef MELZI
 #define LED_PIN            27 /* On some broken versions of the Sanguino libraries the pin definitions are wrong, which then needs LED_PIN as pin 28. But you better upgrade your Sanguino libraries! See #368. */
 #define FAN_PIN            4 // Works for Panelolu2 too
#endif

#ifdef STB
 #define FAN_PIN            4
	//  Uncomment this if you have the first generation (V1.10) of STBs board
 #define LCD_PIN_BL         17 // LCD backlight LED
#endif

ShrikantPai123456, 2017-04-21:
Sanguino controller board...

ShrikantPai123456, 2017-04-21:
Wow "OR"ing working sir...

Ok now the FAN is automatically getting ON even if its 1% (Almost OFF).

At 1% my FAN is rotating FULL... Pls tell me sir...

ShrikantPai123456, 2017-04-21:
[image]

At this 1% FAN is rotating FULL.. What shall I do sir????

thinkyhead, 2017-04-22:
The `FAN_PIN` fan is for the software-controlled fan which is set with `M106` and used to cool the printed object when printing in PLA. This is the fan status that will be printed on the graphical display. The "extruder auto-fans" are not displayed.

>
```
#if defined(FAN_PIN) && FAN_PIN > -1
```

In Marlin we can use `#if HAS_FAN0` in place of that. (And `PIN_EXISTS(FAN)` would be the literal equivalent.)

ShrikantPai123456, 2017-04-23:
ya ok let me give a try sir... thx

closed 2017-11-01 by thinkyhead