MarlinFirmware/Marlin #6412: feature request real colour mixing

sebastiangonzalezv, 2017-04-21:
I had a idea for 3d printer that include some change in g code like 
G1 Xnnn Ynnn Ennn Annn Bnnn Cnnn Dnnn Hnnn Innn
1=Annn +Bnnn+Cnnn+ Dnnn+Hnnn+Innn
were 
A cyan color
B magenta color
C yellow color
D black color
H white  color
I support material
the is can use mixer extrude from 3 to 6 mixer like the picture down 
[image]
for 3 color
to 6 like 
[image]
so the community after hat the new G code can real any color print

Roxy-3D, 2017-04-21:
Yes!  I think this is a good idea.   But I'm thinking some kind of Pallet command would be better.  It would do the same thing as what you are suggesting.  But we would set the amount each extruder contributes to the color  just once until there is a further color change.

The reason is very few people would have this system.  But to do the parsing the G1 command would have to check for 3 extra parameters (that are rarely present) on every move command.  That is going to burn up a lot of CPU cycles we do not have spare.   

So...  I like the idea.  I would like to see it move forward.  But I would like to see it done slightly differently.
(And I would like to have one of these nozzles and the associated drive system!!!)

psavva, 2017-04-22:
How will you mix the colours whist is in the chamber? 

It you push red and blue though the nozzle at the same time, the filament on the other end might come out half red and half blue, rather than an expected violet/purple colour.

You also have to consider the filament's ability to mix and blend once it's melted.  If you melt two strands  of pla for example, they may not mix as you might expect due to the different pigments used for different brands of pla...

sebastiangonzalezv, 2017-04-22:
its is possible her is a demonstration using the nozzle
http://www.reprap.org/wiki/Repetier_Color_Mixing#Setting_up_colors_in_Repetier-Host

thinkyhead, 2017-04-22:
This type of extruder is supported in Marlin through the `MIXING_EXTRUDER` feature.
```cpp
/**
 * "Mixing Extruder"
 *   - Adds a new code, M165, to set the current mix factors.
 *   - Extends the stepping routines to move multiple steppers in proportion to the mix.
 *   - Optional support for Repetier Host M163, M164, and virtual extruder.
 *   - This implementation supports only a single extruder.
 *   - Enable DIRECT_MIXING_IN_G1 for Pia Taubert's reference implementation
 */
//#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif
```

psavva, 2017-04-26:
@sebastiangonzalezv, if you agree, please close this ticket

closed 2017-04-26 by thinkyhead