MarlinFirmware/Marlin #6414: Upgrade Configuration File

jasoncross, 2017-04-21:
Is there a tool or method to update the old version configuration files to the new version short of a full variable-by-variable compare and rebuild?

Bob-the-Kuhn, 2017-04-21:
Most of ease the pain by using an editor like Notepad++ that has a nice side by side compare.

I've seen a mention of a tool (Marlin3DprinterTool) that's supposed to help with the migration.  It's still in beta & you might have to pay for it.  
Marlin3DprinterTool - http://www.marlin3dprintertool.se/LatestVersion/Marlin3DprinterTool.msi
Website -  http://www.marlin3dprintertool.se

Kaibob2, 2017-04-21:
My favorite Tool is [winmerge] (http://winmerge.org/downloads/?lang=en) as it is only made for comparing two files. 
Notepad++ is a great tool, but, in my opinion, a little bit too bloated for only comparison.

Bob-the-Kuhn, 2017-04-21:
I just tried winmerge.  The difference displays are very similar but it's a LOT better at not getting lost (loosing synch between the two files).

fiveangle, 2017-04-21:
Don't try [Beyond Compare](http://www.scootersoftware.com/index.php) or you'll be forced to eventually whip our your check book ;)

closed 2017-04-22 by thinkyhead