MarlinFirmware/Marlin #6415: At 1% of FAN percentage on LCD, My controller FAN is getting ON at FULL speed.....Help

ShrikantPai123456, 2017-04-21:
Hi all...

I am using 128*64 GLCD. On the LCD its 1% FAN PERCENTAGE but my controller FAN is FULL rotating (up to 255)... Pls tell me what can I do???

**Here is pins.h**
#define LED_PIN            -1

#define FAN_PIN            -1
 #if FAN_PIN == 12 || FAN_PIN ==13
  #define FAN_SOFT_PWM
#endif

#ifdef MELZI
 #define LED_PIN            27 /* On some broken versions of the Sanguino libraries the pin definitions are wrong, which then needs LED_PIN as pin 28. But you better upgrade your Sanguino libraries! See #368. */
 #define FAN_PIN            4 // Works for Panelolu2 too
#endif

#ifdef STB
 #define FAN_PIN            4
	//  Uncomment this if you have the first generation (V1.10) of STBs board
 #define LCD_PIN_BL         17 // LCD backlight LED
#endif

Using sanguinololu clone...

Here is **config_adv.h**

#define CONTROLLERFAN_PIN -1 //Pin used for the fan to cool controller (-1 to disable)
#define CONTROLLERFAN_SECS 60 //How many seconds, after all motors were disabled, the fan should run
#define CONTROLLERFAN_SPEED 255  // == full speed

// When first starting the main fan, run it at full speed for the
// given number of milliseconds.  This gets the fan spinning reliably
// before setting a PWM value. (Does not work with software PWM for fan on Sanguinololu)
//#define FAN_KICKSTART_TIME 100

// Extruder cooling fans
// Configure fan pin outputs to automatically turn on/off when the associated
// extruder temperature is above/below EXTRUDER_AUTO_FAN_TEMPERATURE.
// Multiple extruders can be assigned to the same pin in which case
// the fan will turn on when any selected extruder is above the threshold.
#define EXTRUDER_0_AUTO_FAN_PIN   -1
#define EXTRUDER_1_AUTO_FAN_PIN   -1
#define EXTRUDER_2_AUTO_FAN_PIN   -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

help asap....

Kaibob2, 2017-04-21:
Opening more issues related to the same "issue" won't help much. Keep calm...

Tannoo, 2017-04-22:
Likely your fan is not plugged into the fan output.

ShrikantPai123456, 2017-04-22:
Yeak ok sir. I got what do mean.

But pls tell me my FAN is rotating FULL even at 1%.

What shall I change sir????

Tannoo, 2017-04-22:
Is the fan powered, or aee you spinning by hand?

Tannoo, 2017-04-22:
If it is powered, is it powered by the FAN_PIN or not?
It (the fan) needs to be powered, or switched, ot triggered BY the FAN_PIN.

ShrikantPai123456, 2017-04-22:
Ya FAN pin is 4,,, and is spinning as POWER is ON FULL but displaying 1%

Tannoo, 2017-04-22:
Bad output?

Tannoo, 2017-04-22:
If you set it to 0, does it turn off?

Tannoo, 2017-04-22:
Is pin 4 controlling a FET on your board?

Tannoo, 2017-04-22:
If so, likely bad FET.

ShrikantPai123456, 2017-04-22:
thnks fr replying..

changing to 0 is FAN is getting OFF but then also displaying 1%.

Hardware FAN is working but only 1% its showing sir....

ShrikantPai123456, 2017-04-22:
I wanted that as I change 0-255 value of CONTROL menu---->FAN Speed which varies 0-255 then my controller board should vary.... EXAMPLE: At 128 FAN CONTROL MENU, my controller FAN should spin half..

What shall I do sir?

Tannoo, 2017-04-22:
Unplug the fan, check for voltage throught the range.
If you see nothing but 0 and full system voltage. Eouldn't it then be possible that either the FET is bad, or pin 4 is not driving a FET, and it's also not analog.

thinkyhead, 2017-04-22:
#6411

closed 2017-04-22 by thinkyhead