MarlinFirmware/Marlin #6417: Any idea why #6400 breaks the steppers for my Anet printer fork?

oderwat, 2017-04-21:
When I add the changes from #6400 the steppers just go crazy (sounds like being much to fast an losing nearly all steps). Unfortunately I don't have any clue on what is really done in that commit and why these changes break everything so bad for my fork (https://github.com/SkyNet3D/Marlin/)

benlye, 2017-04-21:
@oderwat, you probably figured this out already, but just reverting stepper.cpp gets the motion back to normal.  Once the kids are in bed I'll try to work out why.

bgort, 2017-04-21:
Same issue here.

bgort, 2017-04-21:
> @oderwat, you probably figured this out already, but just reverting stepper.cpp gets the motion back to normal.

Yeah, same here, again.  Reverting stepper.cpp changes fixes the problem.  Not sure what the issue is without digging into the 2560 datasheet.

benlye, 2017-04-21:
I believe I found it.  There is a typo in fastio.h at line 141.

This:
```
    TCCR##T##Q = (TCCR##T##Q & !(0x3 << COM1##Q##0) | (int(V) & 0x3) << COM1##Q##0); \
```

Should be this:
```
    TCCR##T##Q = (TCCR##T##Q & ~(0x3 << COM1##Q##0) | (int(V) & 0x3) << COM1##Q##0); \
```

Changing that line in the new macro solves the crazy stepper motion for me.

bgort, 2017-04-21:
Ah, that makes sense.  Good catch.  Fixed things here, too.

oderwat, 2017-04-21:
Fixed it for me as well! Thank you for the quick response!

closed 2017-04-21 by oderwat