MarlinFirmware/Marlin #6426: RCBugFix weird LCD screen at start up

cormier8, 2017-04-22:
Hi,

I'm using a VELLEMAN K8400 dual haeds 3d printer with RCBugFix.

Works fine, but I've a strange LCD screen at startup (see picture) :

[image]

temperatures and positions are reported correctly, but temperature tags are replaced by '\', there is a 'i' just before 'Z', and a '\" before SD ...

If I click on the knob, the tags are OK and the bug don't come back anymore.

So I think it's a LCD initialization problem and I'm looking for a command to add in setpup().

I explored ultralcd.h and .cpp, but I did'nt find any suitable command.

Any idea ?

cormier8, 2017-04-22:
got it !

In file ultralcd_impl_HD44780.h, line 533, there is a '/*' wich make lines 533 to 541 commented :

    /*    <- line 533
    lcd.clear();

    lcd_set_custom_characters(
      #if ENABLED(LCD_PROGRESS_BAR)
        false
      #endif
    );
    //*/ <- line 541

so, the lcd_set_custom_characters(...) is not executed and the custom characters are not restored after bootscreen, wich explain wrong tag and weird status screen.

closed 2017-04-22 by cormier8

thinkyhead, 2017-04-26:
I'll have to check and see whether that is intentional or not, and how it got in.

thinkyhead, 2017-04-26:
Actually it was an intentional patch, meant to hide the change of character set on the boot screen, (which looks glitchy). For some reason on my own unit the character set is changing. But I think it's because I have an SD card inserted, and this causes an extra init of the screen.

So, I will reinstate this code, but add a small delay to hide the glitch.

commit 4d96d3ea (2017-04-26): Bootscreen delay after clear, then set chars

Fixes #6426