MarlinFirmware/Marlin #6430: Why isn't '

Roxy-3D, 2017-04-22:


thinkyhead, 2017-04-22:
It isn't always desirable to reduce all commits down to a single commit. But when it is, that would be a helpful option. Currently you must squash commits with `git rebase -i ; git push -f`.

Roxy-3D, 2017-04-22:
OK...   I do understand that.     I won't abuse it.   I'll only use it where it makes sense to do that.

How about we turn that on?

thinkyhead, 2017-04-26:
I've enabled the feature. Hopefully it provides a way to set the commit message on the squashed+merged result.

Roxy-3D, 2017-04-26:
Thank you.    I'm pretty sure it is going to let us set the commit message.   But I've always down that in the Pull Request, so for me it probably doesn't change anything.

closed 2017-11-01 by thinkyhead