MarlinFirmware/Marlin #6433: Disable bed heater during Z-Probe move

thijsk, 2017-04-23:
I was wondering if the magnetic field caused by the heated bed being powered on could influence the accuarcy of my Z-probe (a hall effect sensor). So I modified the run_z_probe() function to disable the bed right at the start and turn it back on when the probe is completed.

This way the bed will turn be off during each probe with (bi)linear auto bed leveling turned on, while the temperature does not drop significantly during the whole process.

Any thoughts on this? 
Would it be usefull to submit a pull request for this?

```
#if ENABLED(DISABLE_HEATING_DURING_PROBE)
	float temp = thermalManager.degTargetBed();
	thermalManager.setTargetBed(0);
#endif 
```    
```
#if ENABLED(DISABLE_HEATING_DURING_PROBE)
	thermalManager.setTargetBed(temp);
#endif
 ```

thijsk, 2017-04-23:
By browsing other issues I just found a similar discussion here: https://github.com/MarlinFirmware/Marlin/issues/6394

Grogyan, 2017-04-23:
It's the same issue.
EM Radiation from the heat bed, hot end, fan, motors etc

While the probe is usually in close proximity to the hot end and bed, these
are two areas to be looked at after Marlin goes 1.1

As any modifications like this could have serious consequences.

The Marlin Dev team have spent months getting the UBL working and, and work
reliably, I have suggested in the other issue, and i'll repeat here.

Best to leave it till after Marlin goes 1.1.
Work around is to manually preheat the heat bed and hot end till they are
working temp, then turn them off.  Then run the auto bed levelling.

closed 2017-11-01 by thinkyhead