MarlinFirmware/Marlin #6435: Switching extruders with two steppers

chromvis, 2017-04-24:
Currently SWITCHING_EXTRUDERS only work with one stepper. How to modify (I believe Marlin_Main.cpp) soft to allow two stepper operation? It is very simple to make swinging extruder system from Dual Nozzle MK8 Extruder kit. Switching allow to set hotend offsets and there is no need for very careful and tedious hotend alignment, but how to control second stepper? 
It would be good to include option for TWO STEPPERS switching extruder.

Tannoo, 2017-04-24:
>Currently SWITCHING_EXTRUDERS only work with one stepper.

..at a time, I believe.

thinkyhead, 2017-04-26:
This is a duplicate of #6260

thinkyhead, 2017-04-26:
> ..at a time, I believe.

No, @chromvis is correct. It uses a single stepper motor, but two nozzles.

closed 2017-04-26 by thinkyhead