MarlinFirmware/Marlin #6438: Printer resets when homing (starting a new print) or executing G28/G29 command

lucianstefan, 2017-04-24:
Hey.

My printer started to restart after an incident  (i switched the power off when i was testing my Z calibration, and didn't want it to crush into the bed) 
Doing so, now when i try to print anything it restarts/resets when homing.
If i input into the command line the g-code G28/G29 it does the same thing, it restarts/resets when the Z axis goes down a few steps. ( I don't know if i am calling it right ? mybe it is a firmware crash)

The fact is that i can't use my printer anymore. And i don't know what to do.

I mounted a inductive probe a few days ago and had to recalibrate my Z axis and set the offset for it.

I am using 1.1.0 RC7 with ramps 1.4 and an original Arduino Mega.
My endstops are working fine, my probe is working fine when testing it with m119. Motors cheked and work fine individually.

I don't know what to do :|

Roxy-3D, 2017-04-24:
What type of bed leveling do you have active?

lucianstefan, 2017-04-25:
Hello.

I am using Auto bed leveling - "AUTO_BED_LEVELING_FEATURE" - grid with 4 points. But it doesn't even get to probe a point because at the first one when going down it restarts/resets.

thinkyhead, 2017-04-26:
First, upgrade to `RCBugFix` so that we are on the same page. We might have fixed this already. If the problem persists with `RCBugFix` then I would ask, what kind of microstepping do you have on your Z axis? What is your Z axis feed-rate? If you move the Z axis at top speed does the board reset?

lucianstefan, 2017-04-26:
Thanks for trying to help!

Oki, i am running now on RC8. I assume this is what RCBugFix means? going to the latest version !?

The problem persists, but now in a different kind of way. My resets have gone away.

- these are my steps per unit: (X,Y,Z,E0)  81.2311, 79.00, 8036.00, 105.00
- feedrate: 300, 300, 4, 30

My new problem. 
Sending G28 - it's doing it well. 
Sending G29 - it's doing it well but not completely. After a fiew points that are probed at almost half of the way it freezez, and stays like that a while (30sec - 1 min) then in continues but it looks like it forgot his steps per unit... and when Z axis rises, he now rises like 50mm insted of 15mm and goes down slowley not fast and slow at the end, just slowley all the way.

Another example:
And if i send M114 and lets say we are at 10mm hight, and want to go to G0Z5 inputting this command doesn't do anything, it dosen't move. But if i then type M114 it says he is with Z axis at 5mm if then i type G0Z10 it goes at (what he thinks it's 10mm) so he actualy moves upper. So this is wired. Maybe Arduino board or RAMPS or a Z driver is malfunctioning?

Bob-the-Kuhn, 2017-04-26:
RCBugFix is newer then RC8.  You can download it [here](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

You'll need to manually move your machine specific items over to the new config files.

lucianstefan, 2017-07-08:
Hello guys.... well it's me again.
I haven't powerd my printer from april. I just finished my Arhitecture exams.

And now i am back. Back with the same problem.

I've updated to version 1.1.x

I can issue g28 for homing, but when i try G29 for bed leveling, it just stays in the center and doesn't do enything. After a while it starts probes 4 points, and at the fifth stais again just... "thinking". After a minute it starts to probe point 5 but it doesn't know his Z hight anymore and probes in air the point 5 and 6.

closed 2017-07-10 by lucianstefan