MarlinFirmware/Marlin #6439: After G28 Probe begins G29 at the center of bed

Misailer85, 2017-04-24:
Hello
Yesterday I updated to RC8, after the configuration I got a big problem.

I start G28 (safe z homing), X and Y leveling, Probe moves to the center and level the z axis. 

OK, works perfect!

Now I type G29, the Probe doesn't move and Beginn the Auto leveling at this place (center) and get over the printbed. My normally start point for G29 is 0/0 at the left front corner!

Please help me

Roxy-3D, 2017-04-24:
Which bed leveling scheme are you using?

Misailer85, 2017-04-24:
Actually 3-point.

Roxy-3D, 2017-04-24:
In 3-Point you specify in the Configuration.h file the location of where you want the probe locations to be.   Can you check that you have those defined correctly?    And...   You may want to verify that you have the names correct by comparing things to the very latest RCBugFix configuration.h file.   I believe the names for the probe points might have changed.  If you just dropped an old configuration.h file on top of the new code base, it probably won't work.

thinkyhead, 2017-04-26:
If the problem persists, please post your `Configuration.h` file. You may need to add a `.txt` extension for Github to accept the file drop onto the textfield.