MarlinFirmware/Marlin #6440: [Support] Bed heater not reaching 100C - What is the best setup?

mperdue, 2017-04-24:
Until yesterday I only really printed with PLA and a bed temperature of 50c to 60c. I now have a part that need to be printed with ABS so I set the bed temperature to 100c it errors out at about 90c. I'm using an NTC 3950 thermistor and two 300mm x 200mm beds in parallel. Each bed is about 1.4 ohms at 12V that should be about 9.25 amps for a total draw of 18.5 amps or 222 watts. My power supply is a Thermaltake 850 that can supply 70 amps @ 12V or 840 watts so there shouldn't be an issue with the power.

What's the best way to setup the heated bed, bang-bang or PID?

What thermistor should be selected in the configuration for the bed? I'm currently using 11.

Any help would be appreciated.

Roxy-3D, 2017-04-24:
What electronics are you using?    For example...  The RAMPS boards probably can't provide enough current to drive two beds.   The current it can supply would get split between the two beds and probably neither would get to full temperature which is kind of what you are seeing.

You could use the electronics to drive a 12 volt (output) SSR (Solid State Relay) and that would let enough current get to both boards.  Something like this: https://www.amazon.com/Uxcell-Temperature-Controller-3V-32V-12V-100V/dp/B019MO50UY/ref=sr_1_1?ie=UTF8&qid=1493050707&sr=8-1&keywords=12v+ssr

mperdue, 2017-04-24:
I'm using a Panucatt AZTEG X3 Pro. It has two circuits for heated beds and says it can handle up to 30 amps.

mperdue, 2017-04-25:
I'm going to try messing with the settings tonight to see if I can make it work better. If anyone else has any ideas I'd love to hear them. Otherwise I'll be closing this tomorrow. I may just go with the SSR and a 110V silicone heater and be done with it.

Bob-the-Kuhn, 2017-04-25:
Exactly what do you mean by errors out?

Is the bed not getting to the desired temperature?  Is there an error message?

mperdue, 2017-04-25:
I get:

Heating failed
PRINTER HALTED
Please reset

Bob-the-Kuhn, 2017-04-25:
I take it the bed never gets to the desired temperature.

If you have any cooling fans I suggest trying to divert air away from the bed.

I'm running a 300x300 bed with a 400 W heater.  It's all it can do to get up to ABS temperatures.

I suggest adjusting your power supply higher.  Going from 12V to 17V will double your power.  14.5 V will get you 50% more power.  

If you increase the voltage then feel how warm the connectors are getting.  They're the weak links in the system. Warm is OK.  Hot isn't.

mperdue, 2017-04-25:
I suspect that the power supply I'm using isn't putting out anywhere near the rated current. According to the specs, it should be putting out 850W of 12v power which should be more than enough to power everything. I don't have an ammeter to test it. I do have two identical printers and both of them are doing the same thing. I don't know that the power supply is easily adjustable as it's a PC supply. I've got a Chinese 500W 12v supply that I may try tomorrow. 

I'm using 14 AWG wiring to the heated beds so that should be good for about 3 times the current they will draw at 12 volts. The wires don't really heat much at all, just barely above ambient temperature after hours of printing.

Roxy-3D, 2017-04-25:
> Heating failed
> PRINTER HALTED
> Please reset

Personally...  I think we have the parameters much too tight.   I end up relaxing them on all of my printers.   But I think we promote RCBugFix to Golden before we have any discussion about relaxing heater and safety constraints.     People's thoughts are very welcome here!

mperdue, 2017-04-25:
How much do you relax yours? I don't think it would help in my case because when I plot the temperature curve it seems to flatline at about 91C.

Roxy-3D, 2017-04-25:
[Configuration_adv-h.txt](https://github.com/MarlinFirmware/Marlin/files/953818/Configuration_adv-h.txt)
[Configuration-h.txt](https://github.com/MarlinFirmware/Marlin/files/953819/Configuration-h.txt)

mperdue, 2017-04-25:
I'm thinking that a 110V silicon heat pad with an SSR is going to be the setup I need to go to. Do you attach the pad directly to you glass or do you have it attached to a bed with the glass sitting on top of it?

Roxy-3D, 2017-04-25:
I'll try to remember to post a picture in the morning.   What I did was super simple but it works out perfectly for me.  I have a very thin sheet of plywood with the generic 900 watt 110 VAC heating pad with a generic type 75 NTC something or other thermistor switch by a Fortec 60 Amp SSR bought on eBay for $3.96 including shipping from Asia somewhere.  I have metal corner guides that my 400mm x 400mm sheet of borosilicate glass drops into.

lrpirlet, 2017-04-25:
@mperdue 
I sort of had the same kind of problem with my printer at about 85°C... Disabling the security in Marlin, I could go up to 90°C after 35 minutes... Since I was to print with ABS, I was cornered to find something...

Before to increase the power (and blow away the FET AGAIN :-( ), I did try to increase the efficiency of the "system" and try to REDUCE the heat loss... I finally glued some aluminium foil on a cardboard and using  some spacer in cork, I attached this with capton tape so that it hangs about 8 mm under my MK2A making a nearly closed chamber (I still needed some apertures for my clips) with infra red reflective bottom... I could go to 110°C before the curve become asymptotic again.... 

My next plan, if the above had not answered my need would have been to use cork protected by capton tape in direct contact with the MK2A so as to press its other surface to the glass (except that I was not sure the stress induced in the glass would have been too much for my glass AND the weight would have been excessive in my eyes)...

closed 2017-04-26 by mperdue