MarlinFirmware/Marlin #6441: Add extruder load/unload  in LCD Menu Prepare

garaguido, 2017-04-24:
Is this possible to ad and updated the program lines ( original from Glenn West for RC1) in ultralcd.cpp for loading and unloading filament to bowden extruder ?

1. Menu Prepare - Preheat ABS (Or PLA)
2. Menu Prepare - Unload Filament
3. Menu Prepare - Load Filament
4. Menu Prepare - Feed Filament

With1 meter bowden tube, adjust your code for your bowden size.

Change this in ultralcd.cpp

**static void lcd_prepare_menu()
{
    START_MENU();
    MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
#ifdef SDSUPPORT
    //MENU_ITEM(function, MSG_AUTOSTART, lcd_autostart_sd);
#endif
    MENU_ITEM(gcode, MSG_DISABLE_STEPPERS, PSTR("M84"));
    MENU_ITEM(gcode, MSG_AUTO_HOME, PSTR("G28"));
    MENU_ITEM(gcode, MSG_LOAD_FILAMENT, PSTR("G21 G F200 E10"));
    MENU_ITEM(gcode, MSG_FEED_FILAMENT, PSTR("G F1000 E1000"));
    MENU_ITEM(gcode, MSG_UNLOAD_FILAMENT, PSTR("G F1000 E-1000"));
    //MENU_ITEM(gcode, MSG_SET_ORIGIN, PSTR("G92 X0 Y0 Z0"));
    MENU_ITEM(function, MSG_PREHEAT_PLA, lcd_preheat_pla);
    MENU_ITEM(function, MSG_PREHEAT_ABS, lcd_preheat_abs);
    MENU_ITEM(function, MSG_COOLDOWN, lcd_cooldown);
#if PS_ON_PIN > -1
    if (powersupply)
    {
        MENU_ITEM(gcode, MSG_SWITCH_PS_OFF, PSTR("M81"));
    }else{
        MENU_ITEM(gcode, MSG_SWITCH_PS_ON, PSTR("M80"));
    }
#endif
    MENU_ITEM(submenu, MSG_MOVE_AXIS, lcd_move_menu);
    END_MENU();
}**

Roxy-3D, 2017-04-24:
Yes..   it is possible...   But the M600 Filament Change command is available from the LCD Panel.  And the user does get full control over those types of things.  Can you give it a try and see if it handles what you need?
```cpp
/**
 * Filament Change
 * Experimental filament change support.
 * Adds the GCode M600 for initiating filament change.
 *
 * Requires an LCD display.
 * This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 */
#define FILAMENT_CHANGE_FEATURE
#if ENABLED(FILAMENT_CHANGE_FEATURE)
  #define FILAMENT_CHANGE_X_POS 3             // X position of hotend
  #define FILAMENT_CHANGE_Y_POS 3             // Y position of hotend
  #define FILAMENT_CHANGE_Z_ADD 10            // Z addition of hotend (lift)
  #define FILAMENT_CHANGE_XY_FEEDRATE 100     // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
  #define FILAMENT_CHANGE_Z_FEEDRATE 5        // Z axis feedrate in mm/s (not used for delta printers)
  #define FILAMENT_CHANGE_RETRACT_FEEDRATE 60 // Initial retract feedrate in mm/s
  #define FILAMENT_CHANGE_RETRACT_LENGTH 2    // Initial retract in mm
                                              // It is a short retract used immediately after print interrupt before move to filament exchange position
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 10  // Unload filament feedrate in mm/s - filament unloading can be fast
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 100   // Unload filament length from hotend in mm
                                              // Longer length for bowden printers to unload filament from whole bowden tube,
                                              // shorter length for printers without bowden to unload filament from extruder only,
                                              // 0 to disable unloading for manual unloading
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 6     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 0       // Load filament length over hotend in mm
                                              // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                              // Short or zero length for printers without bowden where loading is not used
  #define FILAMENT_CHANGE_EXTRUDE_FEEDRATE 3  // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define FILAMENT_CHANGE_EXTRUDE_LENGTH 50   // Extrude filament length in mm after filament is loaded over the hotend,
                                              // 0 to disable for manual extrusion
                                              // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                              // or until outcoming filament color is not clear for filament color change
  #define FILAMENT_CHANGE_NOZZLE_TIMEOUT 45   // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS 5 // Number of alert beeps before printer goes quiet
  #define FILAMENT_CHANGE_NO_STEPPER_TIMEOUT  // Enable to have stepper motors hold position during filament change
                                              // even if it takes longer than DEFAULT_STEPPER_DEACTIVE_TIME.
  #define PARK_HEAD_ON_PAUSE                  // Go to filament change position on pause, return to print position on resume
#endif
```

garaguido, 2017-04-24:
From witch Marlin version is this aplyed RC8 , and enable it  in configuration.h ?

Jinxycob, 2017-04-25:
you found it yet?. have a look in the Configuration_adv.h folder

garaguido, 2017-04-25:
I found it , but is it only for the ULTIPANEL ? 
I use the RepRapDiscount FULL GRAPHIC Smart Controller.

Jinxycob, 2017-04-25:
still works I use it all the time cracking feature just do it! :P 
only becomes visible under controls setting once you start printing

garaguido, 2017-04-25:
Nop dos not working by me, even by start printing .
What MarlinRC are you using ?

garaguido, 2017-04-25:
It is working now thanks fore your help , but that is not my idee of using it 
It was verry simpler to integrate this in de LCD menu , my use of it is for loading / unloading the filament in the PTE tube at PREPARE , before printing , not for changing a other filament color .
The code I get (see higher up)  dos not working , iI get errors in Arduino IDE
_

Jinxycob, 2017-04-26:
its a funny one a loader feature  seen it on a couple of printers out there
have it,but I think the wise folks behind marlin  considered it waste of
time.  i think!! I dont speak for them. because manually its just quicker
to do it by hand  before printing starts and right after the warm up.

thinkyhead, 2017-04-26:
Grr... If you're going to post code, format it correctly.
https://guides.github.com/features/mastering-markdown/

thinkyhead, 2017-04-26:
> a loader feature

Pre-loading filament has been implemented as part of the `M600` feature. The only requirement is that you heat the nozzle above minimum extrusion temperature. Then the option becomes available in the LCD menu. This is only the first implementation. In the future it may be available up-front, but pause on a "wait" screen while it heats up the nozzle.

garaguido, 2017-04-26:
Yes , that is what I am talking about .... for the future ... please ?

CoderSquirrel, 2017-04-26:
this is similar to Manual Filament Change: A New Hope #3605 dont?

thinkyhead, 2017-11-01:
Now included as part of `ADVANCED_PAUSE_FEATURE`. The only thing currently missing is an option to skip unload and go straight to load, but that will be added soon.

closed 2017-11-01 by thinkyhead