MarlinFirmware/Marlin #6442: Head crashing for no reason after working fine for days...

NAP-Richard, 2017-04-24:
Situation...

When I attempt to home my printer it deploys the probe and retracts on the first touch but then the z axis doesn't raise and continues to drive the head into the build plate.  I am using the latest Marlin RC Bug Fix and my hardware is as follows: 
MKS GEN V1.4
DRV_8825 Steppers
T2 timing belt with 16 tooth pulley
T8 Lead Screw Dia 8MM Pitch 2mm Lead 8mm
BLtouch

I have my DEFALT_AXIS_STEPS_PER_UNIT @(200, 200, 800, Extruder)

Where am I going wrong?  It was working just fine.

psavva, 2017-04-25:
Was your BLTouch obstructed from deploying the probe when z was homed?  Ie: it was too close to the bed?

NAP-Richard, 2017-04-25:
Here is a video and a link to my config.h file.  Where am I going wrong?

[https://www.dropbox.com/s/pehod4k7i2wjz5m/20170425_120129.mp4?dl=0](url)
[https://www.dropbox.com/s/74hs9plad12fg6u/Configuration.h?dl=0](url)

boelle, 2017-04-26:
psavva commented a day ago
Was your BLTouch obstructed from deploying the probe when z was homed? Ie: it was too close to the bed?

cbusillo, 2017-04-27:
Do you have ABL_BILINEAR_SUBDIVISION enabled?

closed 2017-11-01 by thinkyhead