MarlinFirmware/Marlin #6444: Typo in docs

cgcardona, 2017-04-25:
**Given** I'm a new user hoping to configure marlin
**When** I'm reading [the configuring and compilation docs ](http://reprap.org/wiki/Marlin#Configuring_and_compilation)
**Then** I expect to see
> Example: If you're using a RAMPS 1.4 with a extruder, fan (optional) and heated bed, then you'd need RAMPS_14_EFB.

**Actual** I see
> Example: If you're using a RAMPS 1.4 with a extruder, fan (optional) and heated bed, then you'd need RAMPS_13_EFB.

It incorrectly says `RAMPS_13_EFB.` instead of `RAMPS_14_EFB.`

Tannoo, 2017-04-25:
>RAMPS 1.4 with a extruder

Should also be `an extruder`

thinkyhead, 2017-04-26:
If you've edited the Wiki at reprap.org to make the corrections, then you have our undying gratitude and you can close this issue.

closed 2017-11-01 by thinkyhead