MarlinFirmware/Marlin #6445: Compile error with RCBugFix and BOARD_SANGUINOLOLU_12

drdelaney, 2017-04-25:
This seems to be replicated when using the example\configurations\delta\generic\ config files as well as my own. Changed board to BOARD_SANGUINOLOLU_12 and I am getting the error below, which seems to be a direct result of #6443.
I manually reverted the fastio.h and stepper.cpp files to the previous revisions and it compiles fine.
This happens with arduino IDE 1.6.13 as well as arduino IDE 1.8.1.

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\stepper.cpp:47:

sketch\stepper.cpp: In static member function 'static void Stepper::init()':

fastio.h:189: error: 'COM1C0' was not declared in this scope

 #define _SET_COM(T,Q,V) (TCCR##T##Q = (TCCR##T##Q & ~(0x3 << COM##T##Q##0)) | (int(V) << COM##T##Q##0))

                                                              ^

sketch\fastio.h:194:24: note: in expansion of macro '_SET_COM'

 #define SET_COM(T,Q,V) _SET_COM(T,Q,COM_##V)

                        ^

sketch\fastio.h:197:23: note: in expansion of macro 'SET_COM'

 #define SET_COMC(T,V) SET_COM(T,C,V)

                       ^

sketch\fastio.h:198:66: note: in expansion of macro 'SET_COMC'

 #define SET_COMS(T,V1,V2,V3) do{ SET_COMA(T,V1); SET_COMB(T,V2); SET_COMC(T,V3); }while(0)

                                                                  ^

sketch\stepper.cpp:1087:3: note: in expansion of macro 'SET_COMS'

   SET_COMS(1, NORMAL, NORMAL, NORMAL);

   ^

exit status 1
'COM1C0' was not declared in this scope

drdelaney, 2017-04-25:
I reverted to the following versions and it compiles:
https://raw.githubusercontent.com/MarlinFirmware/Marlin/10da175be515fdf9512809b7d08b41a065bc5a23/Marlin/fastio.h
https://raw.githubusercontent.com/MarlinFirmware/Marlin/ee46ce17436cc28cdf1cd41d53c1c09e00ed55e4/Marlin/stepper.cpp

benlye, 2017-04-25:
Same issue for me.

thinkyhead, 2017-04-26:
Sorry about that. Some timers don't have `COMnCx` as I'm learning! Change:
```cpp
SET_COMS(1, NORMAL, NORMAL, NORMAL);
```
…to…
```cpp
SET_COMA(1, NORMAL);
```

commit 5512130d (2017-04-26): Only set Compare Mode for channel A

Fixes #6445
closed 2017-04-26 by thinkyhead