MarlinFirmware/Marlin #6447: RC8 Extruder1 steps per mm are ignored

kr15uk, 2017-04-25:
I have RC8 running on the machine with 2x different extruders.
1st is equipped with MK8 type gear and and 2nd one with MK7 type so I need different steps per unit for each extruder.
So I did simple comma separated addition it completely ignores that value.
`/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 148.15, 84 }

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 20, 50, 50 }

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000, 10000 }
`

I've tried to do some funny business with [ and ] as per example but it fails to compile.
*Example of my weird code...
`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 148.15[, 84] }`

Is there a glitch or I'm doing something wrong.

Thank you ladies and gents.

Kris

boelle, 2017-04-26:
needs to be done in this format:

 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]

so it seems you got it right

thinkyhead, 2017-04-26:
And you enabled `DISTINCT_E_FACTORS`, yes?

kr15uk, 2017-04-26:
Thanks for the reply.

Just to clarify.
Shall I do this?
`#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 148.15, 84 }`
Or this?
`#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 148.15[, 84] }`

If the 1st one then my mistake for not commenting out `DISTINCT_E_FACTORS`
Will test this later on when machine is done printing.

If 2nd one then my Arduino IDE throws out an errors.

> Configuration.h:489: error: expected primary-expression before ',' token
 #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 148.15 [, 84] }
                                                              ^

Thanks again.

Bob-the-Kuhn, 2017-04-26:
> #define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 400, 148.15, 84 }

This is the correct format.

cbusillo, 2017-04-27:
Do you have EEPROM enabled?

kr15uk, 2017-04-27:
Why EEPROM stuff is relevant here?!

Anyway.

I can confirm that's working fine now.
I'd suggest to remove those [ and ] from the example, quite confusing.

Also to move `DISTINCT_E_FACTORS` up in the code next to related config fields.

Thanks for your help.

closed 2017-04-27 by kr15uk

thinkyhead, 2017-04-29:
> Also to move DISTINCT_E_FACTORS up in the code next to related config fields.

I've added that change to #6491

thinkyhead, 2017-04-29:
> Why EEPROM stuff is relevant here?!

Because the settings stored in EEPROM override those set in the configuration.

thinkyhead, 2017-04-29:
> I'd suggest to remove those [ and ] from the example, quite confusing.

Sorry about that, but it's a standard format used to indicate optional elements.