MarlinFirmware/Marlin #6448: Problem retracting extruding material

batata004, 2017-04-25:
Hi,

Everytime I start print lots of material falls from the nozzle (melted) because the nozzle took some time to get hot. So I decided to do this: after every print finishes I execute the code below to retract filament:

G1 E-3 F50

And right before start printing I execute the code below to extrude a little bit more material

G1 E4 F50

This looks to work perfectly but when my print really starts (printing the skirt) the extruder does not look to move for the first seconds (despite the printer is moving it's X, Y and Z axis correctly). Maybe the command above changed the offset of the extruder and so it waits till it consume 4 mm of material? If this is the case, how do I tell my printer to extrude filament before any print WITHOUT messing aroudn with the offsets?

ALSO a friend of mine told me I should report this in github because he said "every decent printer should understand G10 and G11" commands. I think that what I am trying to do above could be easily solved with G10 and G11 but I tried sending those in the PRONTERFACE but nothing happens.

ANYWAY I would not like to use G10 and G11 cause I want to retract/extrude a very specific amount of material after every print finish and start, so I think I should be better of using G1 command.

Could you please tell me how can I use G1 command (with latest marlin firmware) before everyprint without affecting it's internal "extruder offset"?

tohara, 2017-04-25:
Use G92 E0 after your G1 E* command to reset E axis to 0.

batata004, 2017-04-25:
@tohara  thanks! I tried it and it worked. But I see something strange, is this command really necessary? Weirdly sometimes the extruder does not extrude material in the first seconds of the print but sometimes not.

Could anyone here tell me if the G92 E0 is really necessary after using G1 E4 F50 so the print starts correctly?

tohara, 2017-04-26:
That depends on your slicer. For S3D or SLic3r you can select Relative Extrusion Distance. Then M83 is added to the beginning of you gcode file and you can extrude/retract in your start script without resetting E.

batata004, 2017-04-26:
I think this depends on the firmware not on the slicer.

closed 2017-04-26 by batata004

thinkyhead, 2017-04-29:
> Could anyone here tell me if the G92 E0 is really necessary after using G1 E4 F50 so the print starts correctly?
I think this depends on the firmware not on the slicer.

The printer might have E at any arbitrary value when you start your GCode, so your slicer must properly set E at the start of the GCode with `G92 E0`. If the "Start GCode" moves E to 4.0 then of course it must reset it to 0.0 before proceeding, otherwise the GCode will move the extruder backwards 4mm with the first `G1` command.