MarlinFirmware/Marlin #6451: Allow setting progress bar with g-code command

schwabe, 2017-04-26:
The full graphics display displays a nice progress bar when printing from SD. I would like to extend Octoprint (the status plugin) to show the progress of the print on the progress. Unfortunately, there is no gcode to set this progress.

Is that a good idea to do so? If yes, I can contribute the implementation. (How to I get a gcode assigned for that?)

Roxy-3D, 2017-04-26:
> (How to I get a gcode assigned for that?)

Just grab one.   But try to stay away from important ones because we may want to extend capability of those with GCodes that are similar in number.    But it really doesn't matter...  We will change the GCode number if we think it should be located some place else in the list.

thinkyhead, 2017-04-26:
@schwabe Use `M73`, as it is already set aside for that exact use (in Makerbot firmware).

tcm0116, 2017-05-04:
@schwabe does the [DisplayProgress](https://github.com/OctoPrint/OctoPrint-DisplayProgress) plugin for OctoPrint not meet your needs?

schwabe, 2017-05-05:
@tcm0116 It would be nice if the progress bar that is already on the display could be used by said plugin.

schwabe, 2017-05-05:
Implementing is also not a quick half hour patch for someone how does not know much about Marlin, I don't I will implement this anytime soon, feel free to close the issue

SydneyMakers, 2017-09-02:
I believe I have implemented this function but I have no clue how to use github so:
`#if ENABLED(LCD_PROGRESS_BAR)
        #include "ultralcd_impl_HD44780.h"
        case 73: // M73: Set Progress Bar Percentage
          gcode_M73(); break;
      #endif`
goes into Marlin_main.cpp at around line 1105** between case 49's section and case 75's section.

`/**
 * M73: Set LCD Progress Bar Percentage
 */
inline void gcode_M73() {
  #if ENABLED(LCD_PROGRESS_BAR)
    #include "ultralcd_impl_HD44780.h"
    ultralcd_impl_HD44780::lcd_draw_progress_bar(parser.seen('P') ? (parser.has_value() ? parser.intval('P') : 100) : 0);
  #endif
}`

goes around line 7206 between M49's gcode function, and M75's gcode function.

Grogyan, 2017-10-12:
M73 is already used in CNC to save Modal State
[LinuxCNC](http://linuxcnc.org/docs/2.7/pdf/LinuxCNC_Documentation.pdf)

There is already disparity between the NIST GCODE standard and what we use.

SydneyMakers, 2017-10-12:
Well the code is implemented, only the function name, and case number need to be changed to fit the new gcode command

SydneyMakers, 2017-10-12:
A further proposal is M74 returns the current build percentage as an integer from 0 to 99

boelle, 2019-02-18:
@schwabe did you try latest bugfix 2.0? it should have the option

maybe we can close this one?

closed 2019-02-18 by thinkyhead

schwabe, 2019-03-11:
@boelle sorry for responding so late. I made a quick hack to the octoprint plugin and it nows displays nicely the percentage graph.

https://github.com/dattas/OctoPrint-DetailedProgress/issues/37