MarlinFirmware/Marlin #6454: Need for an house elf?

boelle, 2017-04-26:
Hi again

Since the issue list has grown to 21 pages i wonder if the project is in need for an label house elf ?

ie go through those that dont have an label yet and put the correct one on
maybe expand that to close those issues where the OP no longer repsonds

thinkyhead, 2017-04-26:
What would ever make you think that? ;-)
[image]

thinkyhead, 2017-04-26:
Indeed, there are many! From my POV it seems a great mountain, but we had to restore the repo last month and all previously-read issues became "unread." My unread issue count has yet to drop below 200.

So… yes, we certainly do need all the elves we can get.

Especially as Marlin 1.1.0 is due to be released on Sunday.

boelle, 2017-04-26:
i'm in for the tasks mentioned

i will start with the oldest issues and work my way to the top of that mountain

the first many pages is just to make sure that labels are correct and that the issues are still somewhat active

boelle, 2017-04-26:
@thinkyhead btw i have added you to hangouts

i dont have skype etc. hope it will make do

boelle, 2017-04-26:
@thinkyhead and just so i dont hammer you for an answer at the wrong times are you located on the west or east cost? i see that east cost is barely out of bed yet.

and in my end of the world we have just passed noon

Roxy-3D, 2017-04-26:
Bo...  PLEASE STOP MAKING ONE LINE POSTS ASKING "Can we close this one?"   That is not helping anything.  It is causing every thread to be recently updated and it is making it impossible to be responsive to anybody that has an issue because I can't even tell what is current and what is not.

boelle, 2017-04-26:
oki... will stop

if you had told me on hangsouts i would have stopped before

boelle, 2017-04-26:
anyway dinner time

@Roxy-3D what is your take on my idea of getting labels on everything and clean up the issue list a bit?

will be on hangouts if that is a better medium to discuss that

markshancock, 2017-04-26:
I just got a Tronxy X3 and am a professional developer and I am interested in contributing to the code base.
I have been programming since 1971 and have written in almost everything.  
I have done a lot of embedded stuff; but, mostly now I do C#.
I also do a lot of CM and SW project management.
I need about another couple of weeks to finish get my printer fully up (an to recover from Robotics season).
Is there something I could help with?

markshancock, 2017-04-26:
Note: I am in the implantable medical devices field; so, I am not afraid of documentation either. ;)

thinkyhead, 2017-04-26:
> thread to be recently updated

Ah yes, forgot that is a caveat.

thinkyhead, 2017-04-26:
@markshancock Hi Mark. Thanks for stepping forward!

My suggestion is to grab the latest Marlin (`RCBugFix` branch), then configure and install it on your machine. As you use it, find the things that you most want to change or fix. Then submit patches. This is a good way to get in practice and familiarize yourself with the various coding units. There are a few knowledgable volunteers here who can help to guide you on the code-base. Be sure to adhere to the existing coding style. You can pick up a lot of that from scanning the code, and we also have a [coding style guide](https://github.com/MarlinFirmware/Marlin/wiki/DNE-Coding-Standards).

There are several items on our to-do list already, but we are putting most of them on hold until after the 1.1.0 release (this Sunday). Once 1.1.0 is released we will immediately start the 1.2.x branch — which will be 1.1.0 exactly, except with a proper file hierarchy. Then we may get more formal, setting up "task forces" to deal with those aspects of Marlin needing attention.

The biggest task ahead will be to add support for 32-bit boards like the Due and Re-ARM. (Much-needed with so many users starting to use kinematic systems and 32x micro-stepping.)

teemuatlut, 2017-04-26:
All the 32bit platforms will at least need testers if you want to help out on that area.

Is there a public todo list or some sort of future goals list?

What's the reasoning behind banning c++ templates? I'd need to duplicate a lot of TMC specific functions if I need to change the implementations.

Roxy-3D, 2017-04-27:
>> Note: I am in the implantable medical devices field; so, I am not afraid of documentation either. ;)

> My suggestion is to grab the latest Marlin (RCBugFix branch), then configure and install it on your machine. As you use it, find the things that you most want to change or fix. Then submit patches. This is a good way to get in practice and familiarize yourself with the various coding units. 

I know the UBL Auto Bed Leveling System could use a 'Quick Start Guide' and maybe a follow up 'Advanced User Guide'.    If you do Thinky's suggestion of bringing up the current RCBugFix, please turn on UBL Bed Leveling and keep notes on things that are not obvious to you and where you needed to ask questions.   I will help you bring it up and get it doing its magic on your printer.  Especially if a 'Quick Start Guide' might fall out of the effort!

thinkyhead, 2017-04-27:
> What's the reasoning behind banning c++ templates?

You can use templates. I think that our past concern was that templates would produce less-efficient code, or that the compiler would bark. There is something called "embedded C++" and I think that we were trying to stay within the bounds of that variant.

markshancock, 2017-04-27:
> I know the UBL Auto Bed Leveling System could use a 'Quick Start Guide' 

 @Roxy-3D  Unfortunately my X3 does not have a bed sensor (yet); but, that is on my list.

markshancock, 2017-04-27:
> My suggestion is to grab the latest Marlin (RCBugFix branch), then configure and install it on your machine.  ...

@thinkyhead Will do, great place to start.  I have played Arduino a little but, I have a Mega 2560 that I have used to drive some addressable LEDs and also to drive a servo.  Seemed easy to use for the basic setup/loop configuration.  I haven't tried anything like interrupts yet.

BTW, are you still in the PDX area?  I live in Tigard.

boelle, 2017-04-28:
I take that my idea was not that welcomed as i thought.

No hard feelings from here :)

markshancock, 2017-04-29:
> I take that my idea was not that welcomed as i thought.
@boelle I looks more like timing.  It sounds like @thinkyhead is focused on the imminent release and wants to focus on that right now and work the issues list afterwards.

markshancock, 2017-04-29:
@boelle, @thinkyhead 

  Thanks so much for the hard work and advice; but, it looks like I won't be helping with Marlin (at last not for now).  
  I thought that Marlin was the firmware that I had; but, it appears, my printer came with Repetier firmware.
 I am hesitant to load a new firmware at this point and risk bricking my board, messing up my config, or at a minimum not being able to get it the printer back operational.  At some point I will probably get another controller board and them you might see me back.

Thanks
Mark Hancock

closed 2017-04-29 by boelle

thinkyhead, 2017-04-29:
> BTW, are you still in the PDX area? I live in Tigard.

No, I moved to Austin, TX last fall following a Marlin-related gig.
Have attended any of the PDX-3DP Lab meet-ups?

markshancock, 2017-05-01:
@thinkyhead 
>No, I moved to Austin, TX last fall following a Marlin-related gig.

Ah, Austin, the West Coast of Texas.  I lived in Texas 69-99 (mostly Houston), graduated TAMU, friends in Austin area.  Went mountain biking at Zilker Park once with friends.

>Have attended any of the PDX-3DP Lab meet-ups?
No, hadn't heard of them.  Found them on meetup, I'll look into it.  Not too hard to find people, seems like everyone has a 3D printer now.  There is even one at the UPS store. :)

Roxy-3D, 2017-05-01:
> graduated TAMU

Whoop!

markshancock, 2017-05-01:
@Roxy-3D 
> Whoop!
BSEE '79
Just was at TAMU a week ago.  We took 12 students from Oregon to Houston for the First Robotics World Championship and when we got eliminated we decided to drive up to Aggieland to show them around.

Roxy-3D, 2017-05-01:
> BSEE '79
> Just was at TAMU a week ago. We took 12 students from Oregon to Houston for the First Robotics World Championship and when we got eliminated we decided to drive up to Aggieland to show them around.

Cool!  I'm BME!  But they made us do enough programming that I can make the changes I want to Marlin and annoy the real Computer Science types.   

You should be showing your support for a fellow Aggie and be running Marlin!

thinkyhead, 2017-05-01:
I did a year and a half of community college once. Arts and humanities.

markshancock, 2017-05-04:
> You should be showing your support for a fellow Aggie and be running Marlin!

@Roxy-3D , I want to; but, I am not confident enough with the code that is on the Melzi Board now that I am sure I could get it working again.  I know at least one person that when he loaded new code onto the Melzi Board he totally bricked the board (he thinks it is a processor config fuse).  He is waiting on a new processor and is still not back up again.  

My main goal for the printer is to print.  The secondary goal is to develop.  I think the safest way to accomplish that is likely to get a 2nd controller (either another Melzi or a RAMPS).

Roxy-3D, 2017-05-04:
>> You should be showing your support for a fellow Aggie and be running Marlin!

> I want to; but, I am not confident enough with the code that is on the Melzi Board now that I am sure I could get it working again if I tried to load new code. 

Yeah!  I understand.   And of course you know I meant that as a joke!   And then I deleted it because it really wasn't appropriate!

> My main goal for the printer is to print. The secondary goal is to develop. I think the safest way to accomplish that is likely to get a 2nd controller (either another Melzi or a RAMPS).

Yeah...   Getting a second setup would make sense.  That way you can easily recover.  You can get the entire electronics setup for $32 including shipping:  http://www.ebay.com/itm/281537747879   That even includes the step-sticks to drive the stepper motors.

In theory...  You would just unplug everything...   And start plugging things into the new electronics.   And at that point...  You start bringing up your printer with changes to the Configuration.h file.

markshancock, 2017-05-04:
> Yeah! I understand. And of course you know I meant that as a joke! And then I deleted it because it really wasn't appropriate!
@Roxy-3D I liked your joke, I am glad I noticed it in the email notice.
Regarding the new electronics, you think RAMPS is a better choice over Melzi?  Conceptually I like the modularity of the design better.  Any preference regarding which display/UI to get?

Roxy-3D, 2017-05-04:
For me...  The answer is "Yes!"   for a couple of reasons...   Probably over half of the RepRap printers are now running Atmega2560 boards with RAMPS v1.4 shields plugged into them.   For sure...  You will have no problem getting good answers to any question you have.   But also...  The Atmega2560 boards have 256KB of program space, so you can run as big of a firmware image as you want.    And lastly...   They are cheap.   If you blow up anything...   you can replace any board easily and cheaply.

But for you...    The easiest thing would be to move the wires from the dead Melzi to a new Melzi one for one.    So...   I guess that has to be factored in.       But a Melzi doesn't let you have an LCD-Panel, does it?   Those really are nice to have!!!

markshancock, 2017-05-04:
> But for you... The easiest thing would be to move the wires from the dead Melzi to a new Melzi one for one. So... I guess that has to be factored in. But a Melzi doesn't let you have an LCD-Panel, does it? Those really are nice to have!!!

Good in concept; but, unscrewing the board to put the new board in probably makes it a wash.
What I would like to do is have a separate controller box to swap between.
Looks like I need to plan out my connection approach so I have the connectors/adapters I need.

> It would have a 20x4 LCD in it. And those are $9 on eBay including shipping!

Hmm... since they are cheap, maybe I should get both.  That would give me some flexibility and if I don't use it now, I can use it later.  Actually, I already have a MEGA2560 and an LCD controller in my parts bin that I use for playing around, maybe I already have the 20x4.

markshancock, 2017-05-05:
Bought RAMPS set + 2004 LCD + 360W P/S last night for $45. From Texas so I will get it next week.
Looks like I am going to play with Marlin after all.

boelle, 2017-05-07:
@thinkyhead & @Roxy-3D 

Why is it that you both have gone cold in responding on hangouts?

You could at least have the courtesy to tell me to piss off, would also make things a lot more easy at my end as it would never have to think about offering help.